openrazer/openrazer #279: Does changing Razer Mouse Keybinds not work then?

Swabbo, 2017-05-14:
I noticed after I ditched the Issue I made about Razer Naga 2014 (and Left-handed version) someone saved my skin and took over with your help, which is really appreciated thanks guys :)

But I was just wondering, I'm left-handed and when I play games I use the Numpad with my right hand, which means the 1-12 keys on my Razer Naga need to be remapped or they are just redundant, is there any way to map them to other keys like with Razer Synapse? I've tried xmodmap but that changes my keyboard's buttons too :(

For example when i type xev into the console and press 1 on my Razer Naga 2014, it shows this:

KeyPress event, serial 37, synthetic NO, window 0x4a00001,
root 0x2a4, subw 0x0, time 20734953, (11,-9), root:(1214,654),
state 0x10, keycode 10 (keysym 0x31, 1), same_screen YES,
XLookupString gives 1 bytes: (31) "1"
XmbLookupString gives 1 bytes: (31) "1"
XFilterEvent returns: False

KeyRelease event, serial 38, synthetic NO, window 0x4a00001,
root 0x2a4, subw 0x0, time 20735046, (11,-9), root:(1214,654),
state 0x10, keycode 10 (keysym 0x31, 1), same_screen YES,
XLookupString gives 1 bytes: (31) "1"
XFilterEvent returns: False

The guide that I used (https://askubuntu.com/questions/24916/how-do-i-remap-certain-keys-or-devices)
states the 3rd row is the imortant one, and that "Keycode 10" shows the button's code that is being pressed, so I continued by typing:

xmodmap -e "keycode 10 = Q"

Which did make the 1 button on my Naga type Q instead, but it also changed the 1 button on my keyboard to Q too and the Q button now typed 1, so it basically swapped them around

kived, 2017-06-06:
You can remap keys for just the mouse using XKB: https://superuser.com/questions/760602/how-to-remap-keys-under-linux-for-a-specific-keyboard-only/869064#869064

The script I use to remap the side buttons to F1-F12 is below. Seems to be working - the buttons on my mouse are triggering F1-F12 while the keyboard is unaffected. xev reports the new codes from the buttons on the mouse.

My device shows up twice under keyboards (and twice under mice), so I had to make a slight modification to the `xinput list | ...` line, choosing to have it affect the first listed device, which seems to work. Check the output of `xinput list` yourself to see what shows up. If you have other Razer devices you might need to modify the regular expression in the sed call to select the right device (replace `Razer` in the `xinput list | ...` line with the full name of the device as listed by `xinput list`). Do not just hard code the number you see in `xinput list`, as it can and will change on reboot, unplug/replug, etc.

```
#!/bin/bash -e

remote_id=$(
    xinput list | sed -n 's/.*Razer.*id=\([0-9]*\).*keyboard.*/\1/p' | head -n1
)
[ "$remote_id" ] || exit

mkdir -p /tmp/xkb/symbols
cat >/tmp/xkb/symbols/custom <<\EOF
xkb_symbols "remote" {
    key <AE01> {
        type= "CTRL+ALT",
        symbols[Group1]= [              F1,              F1,              F1,              F1, XF86Switch_VT_1 ]
    };
    key <AE02> {
        type= "CTRL+ALT",
        symbols[Group1]= [              F2,              F2,              F2,              F2, XF86Switch_VT_2 ]
    };
    key <AE03> {
        type= "CTRL+ALT",
        symbols[Group1]= [              F3,              F3,              F3,              F3, XF86Switch_VT_3 ]
    };
    key <AE04> {
        type= "CTRL+ALT",
        symbols[Group1]= [              F4,              F4,              F4,              F4, XF86Switch_VT_4 ]
    };
    key <AE05> {
        type= "CTRL+ALT",
        symbols[Group1]= [              F5,              F5,              F5,              F5, XF86Switch_VT_5 ]
    };
    key <AE06> {
        type= "CTRL+ALT",
        symbols[Group1]= [              F6,              F6,              F6,              F6, XF86Switch_VT_6 ]
    };
    key <AE07> {
        type= "CTRL+ALT",
        symbols[Group1]= [              F7,              F7,              F7,              F7, XF86Switch_VT_7 ]
    };
    key <AE08> {
        type= "CTRL+ALT",
        symbols[Group1]= [              F8,              F8,              F8,              F8, XF86Switch_VT_8 ]
    };
    key <AE09> {
        type= "CTRL+ALT",
        symbols[Group1]= [              F9,              F9,              F9,              F9, XF86Switch_VT_9 ]
    };
    key <AE10> {
        type= "CTRL+ALT",
        symbols[Group1]= [             F10,             F10,             F10,             F10, XF86Switch_VT_10 ]
    };
    key <AE11> {
        type= "CTRL+ALT",
        symbols[Group1]= [             F11,             F11,             F11,             F11, XF86Switch_VT_11 ]
    };
    key <AE12> {
        type= "CTRL+ALT",
        symbols[Group1]= [             F12,             F12,             F12,             F12, XF86Switch_VT_12 ]
    };
};
EOF

setxkbmap -device $remote_id -print | sed 's/\(xkb_symbols.*\)"/\1+custom(remote)"/' | xkbcomp -I/tmp/xkb -i $remote_id -synch - $DISPLAY 2>/dev/null
```

terrycain, 2017-06-07:
@JaydenW I've been real busy as of late but it is being worked on

closed 2018-02-28 by stale[bot]