openrazer/openrazer #280: Daemon fails to start if notification daemon is not running

hobarrera, 2017-05-15:
I'm currently running `razer-service` as a systemd user service, but it seems to fail to run if the notification daemon isn't running. This can happen in a few situations:

1. It's simply misconfigured.
1. There isn't one.
1. It hasn't started yet.

I actually have my reasons to start the daemon automatically without depending on Xorg (one of which is: I'm fiddling around with wayland/sway, so I don't always even run Xorg).

```
May 14 23:40:12 hyperion.barrera.io razer-service[875]: Traceback (most recent call last):
May 14 23:40:12 hyperion.barrera.io razer-service[875]:   File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 175, in activate_name_owner
May 14 23:40:12 hyperion.barrera.io razer-service[875]:     return self.get_name_owner(bus_name)
May 14 23:40:12 hyperion.barrera.io razer-service[875]:   File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 361, in get_name_owner
May 14 23:40:12 hyperion.barrera.io razer-service[875]:     's', (bus_name,), **keywords)
May 14 23:40:12 hyperion.barrera.io razer-service[875]:   File "/usr/lib/python3.6/site-packages/dbus/connection.py", line 651, in call_blocking
May 14 23:40:12 hyperion.barrera.io razer-service[875]:     message, timeout)
May 14 23:40:12 hyperion.barrera.io razer-service[875]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.freedesktop.Notifications': no such name
May 14 23:40:12 hyperion.barrera.io razer-service[875]: During handling of the above exception, another exception occurred:
May 14 23:40:12 hyperion.barrera.io razer-service[875]: Traceback (most recent call last):
May 14 23:40:12 hyperion.barrera.io razer-service[875]:   File "/usr/bin/razer-service", line 102, in <module>
May 14 23:40:12 hyperion.barrera.io razer-service[875]:     run()
May 14 23:40:12 hyperion.barrera.io razer-service[875]:   File "/usr/bin/razer-service", line 96, in run
May 14 23:40:12 hyperion.barrera.io razer-service[875]:     daemonize(**daemon_args)
May 14 23:40:12 hyperion.barrera.io razer-service[875]:   File "/usr/lib/python3.6/site-packages/razer_daemon/daemon.py", line 116, in daemonize
May 14 23:40:12 hyperion.barrera.io razer-service[875]:     daemon = RazerDaemon(verbose, log_dir, console_log, run_dir, config_file, test_dir=test_dir)
May 14 23:40:12 hyperion.barrera.io razer-service[875]:   File "/usr/lib/python3.6/site-packages/razer_daemon/daemon.py", line 229, in __init__
May 14 23:40:12 hyperion.barrera.io razer-service[875]:     self._load_devices(first_run=True)
May 14 23:40:12 hyperion.barrera.io razer-service[875]:   File "/usr/lib/python3.6/site-packages/razer_daemon/daemon.py", line 452, in _load_devices
May 14 23:40:12 hyperion.barrera.io razer-service[875]:     razer_device = device_class(sys_path, device_number, self._config, testing=self._test_dir is not None, additional_interfaces=sorted(additional_interfaces))
May 14 23:40:12 hyperion.barrera.io razer-service[875]:   File "/usr/lib/python3.6/site-packages/razer_daemon/hardware/mouse.py", line 41, in __init__
May 14 23:40:12 hyperion.barrera.io razer-service[875]:     self._battery_manager = _BatteryManager(self, self._device_number, 'Razer Mamba')
May 14 23:40:12 hyperion.barrera.io razer-service[875]:   File "/usr/lib/python3.6/site-packages/razer_daemon/misc/battery_notifier.py", line 112, in __init__
May 14 23:40:12 hyperion.barrera.io razer-service[875]:     self._battery_thread = BatteryNotifier(parent, device_number, device_name)
May 14 23:40:12 hyperion.barrera.io razer-service[875]:   File "/usr/lib/python3.6/site-packages/razer_daemon/misc/battery_notifier.py", line 31, in __init__
May 14 23:40:12 hyperion.barrera.io razer-service[875]:     notify2.init('razer_daemon')
May 14 23:40:12 hyperion.barrera.io razer-service[875]:   File "/usr/lib/python3.6/site-packages/notify2.py", line 96, in init
May 14 23:40:12 hyperion.barrera.io razer-service[875]:     '/org/freedesktop/Notifications')
May 14 23:40:12 hyperion.barrera.io razer-service[875]:   File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 241, in get_object
May 14 23:40:12 hyperion.barrera.io razer-service[875]:     follow_name_owner_changes=follow_name_owner_changes)
May 14 23:40:12 hyperion.barrera.io razer-service[875]:   File "/usr/lib/python3.6/site-packages/dbus/proxies.py", line 248, in __init__
May 14 23:40:12 hyperion.barrera.io razer-service[875]:     self._named_service = conn.activate_name_owner(bus_name)
May 14 23:40:12 hyperion.barrera.io razer-service[875]:   File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 180, in activate_name_owner
May 14 23:40:12 hyperion.barrera.io razer-service[875]:     self.start_service_by_name(bus_name)
May 14 23:40:12 hyperion.barrera.io razer-service[875]:   File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 278, in start_service_by_name
May 14 23:40:12 hyperion.barrera.io razer-service[875]:     'su', (bus_name, flags)))
May 14 23:40:12 hyperion.barrera.io razer-service[875]:   File "/usr/lib/python3.6/site-packages/dbus/connection.py", line 651, in call_blocking
May 14 23:40:12 hyperion.barrera.io razer-service[875]:     message, timeout)
May 14 23:40:12 hyperion.barrera.io razer-service[875]: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
May 14 23:40:12 hyperion.barrera.io systemd[860]: razer.service: Main process exited, code=exited, status=1/FAILURE
May 14 23:40:12 hyperion.barrera.io systemd[860]: razer.service: Unit entered failed state.
```

I really think that `razer-service` should be perfectly capable of running without a notification daemon (soft-failing is probably the best choice) here.

terrycain, 2017-05-15:
Course, yeah that functionality was just a quick hack and then never visited again. It was my original intention to try import then fail silently, possbily spit some stuff into log file.

terrycain, 2017-07-01:
Version 1.1.14 is out on launchpad, non Ubuntu packages will be out shortly.

closed 2017-07-01 by terrycain