openrazer/openrazer #281: xdg basedir spec

hobarrera, 2017-05-15:
Currently the daemon looks for the config file in `$HOME`. This is really ugly, and if every app expected this, I'd have my home littered with stuff (there's lot of other reasons why apps don't do this any more, like making backups easier, etc).

The [XDG basedir spec](https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html) specifies the appropriate places where applications should look for their configurations (and caches, etc).

Ideally, the daemon should look in `$HOME/.config/razer/daemon.conf` (or something like that) for its configuration file.

terrycain, 2017-05-15:
Feel free to make a pull request :wink:

terrycain, 2017-05-15:
Though I dont disagree with you, but for now I have bigger things to work on.

z3ntu, 2017-05-19:
I'll reopen because I am working on it 😉

z3ntu, 2017-05-19:
PR in #286

terrycain, 2017-07-01:
Version 1.1.14 is out on launchpad, non Ubuntu packages will be out shortly.

closed 2017-07-01 by terrycain