openrazer/openrazer #282: gnome-shell issues

borgdylan, 2017-05-15:
I recently switched to gnome-shell ahead of unity's impending doom. When I lock my screen, the keyboard trurns all leds off. When I am logging in I have to turn the backlight on manually. If I do that, it turns off by itself on successful log in. If I do not turn it on myself, it turns on log in. Is tehre something that can be done?

z3ntu, 2017-05-15:
It's a feature, that the backlight turns off when the screensaver is active. Unfortunately we are not aware of any D-Bus events that signal the actual monitor sleep and wakeup so we are using the screensaver events that are supported by all major desktop environments (i.e. screensaver lock and screensaver unlock => `org.freedesktop.ScreenSaver`).

If you know a solution how we can do it better, please feel free to let us know. Closing for now because it's working as intended. 🙂

closed 2017-05-15 by z3ntu

borgdylan, 2017-05-15:
It seems that you are executing a flag toggle. For unlock it should be a turning on of the backlight rather than toggle. Like that it will not turn off if I turned it on manually.

borgdylan, 2017-05-15:
also gnome implements a WakeupScreen method on org.gnome.Screensaver but I'm sure its non-standard

terrycain, 2017-05-15:
No the screensaver DBus handler is acting upon the state value passed when an activeChanged event is emitted. I cant reproduce the issue but I too will soon need to move to gnome shell so it'll probably appear soon.

borgdylan, 2017-05-15:
Most of the issues actually arise from the move to gdm3 from lightdm. That move is mandatory when moving to gnome-shell if you want your lock screen and locking in general to work.

terrycain, 2017-05-15:
Is a shame, never had any issues with lightdm

borgdylan, 2017-05-15:
Lightdm has been the center of a CVE recently.

terrycain, 2017-05-15:
Ha, but then again its the same with anything. You look hard enough and your bound to find something.