openrazer/openrazer #283: BlackWidow Chroma V1 - Numpad not working

IhToN, 2017-05-17:
Hello,

I'm currently using BlackWidow Chroma V1 and Polychromatic. It works without problems but the numpad is not working (macro and function keys works like a charm). Whenever I press a Key in the numpad it does other actions (the + button, for example, works like a Button2-Mouse)

terrycain, 2017-05-17:
@IhToN Does the same thing happen if the drivers arnt installed?

IhToN, 2017-05-17:
No, before installing the drivers the keyboard was working fine (just with the default rainbow wave).

More information: Num Lock button works, the '1' indicator turn on and off.

terrycain, 2017-05-17:
@IhToN Have you tried uninstalling the drivers since the issue. I too have a chromav1 and haven't seen the issue at all

IhToN, 2017-05-17:
I've trying using rmmod but the only thing I got was disabling the whole keyboard.
After rebooting the system all was working as before the rmmod.

I must say that I only bound the 

lrwxrwxrwx 1 root root 0 may 17 22:14 0003:1532:0203.0001 -> ../../../devices/pci0000:00/0000:00:1c.5/0000:06:00.0/usb5/5-2/5-2:1.0/0003:1532:0203.0001
lrwxrwxrwx 1 root root 0 may 17 22:14 0003:1532:0203.0002 -> ../../../devices/pci0000:00/0000:00:1c.5/0000:06:00.0/usb5/5-2/5-2:1.1/0003:1532:0203.0002
lrwxrwxrwx 1 root root 0 may 17 22:14 0003:1532:0203.0003 -> ../../../devices/pci0000:00/0000:00:1c.5/0000:06:00.0/usb5/5-2/5-2:1.2/0003:1532:0203.0003

1532:0203.0001 (not 0002 neither 0003)

terrycain, 2017-05-18:
You sure something else isnt messing with the numpad? / was it working before you did any macro stuff?>

IhToN, 2017-05-18:
I'm pretty sure, the macro things came after testing the numpad.

It stopped working once I did the binding to the driver. Also I must say that before the sistem is loaded it works well (before and during the user login).

z3ntu, 2017-05-18:
Could you try killing the daemon by running for example `pkill -e razer-service` or `razer-service -s`?

IhToN, 2017-05-18:
root@ATAlgaba:/home/ihton# razer-service -s
Stopping razer-service... (PID 2868)
No razer-service currently running.
root@ATAlgaba:/home/ihton# pkill -e razer-service
root@ATAlgaba:/home/ihton# 

Still the same poblem u.u

terrycain, 2017-05-18:
@IhToN Now try and bind all 3 keyboard interfaces to hid-generic does that fix

IhToN, 2017-05-18:
Before: http://i.imgur.com/ganFtse.png (drivers on razerkdb, not workin)
I've changed the interfaces to hid-generic and still the same.

I must say that using "onboard" app exists the same error. It seems that once the system is initialized it changes how the numpad works on the whole system.

z3ntu, 2017-05-18:
Could you elaborate what you mean with "onboard app"?

IhToN, 2017-05-18:
It's an application which works as a Screen Keyboard.

z3ntu, 2017-05-18:
Uhm and what does this have to do with the driver?

z3ntu, 2017-05-18:
Or just start explaining what doesn't work from the beginning

terrycain, 2017-05-18:
So if that's exibititing the same issue something else is messing with the keys not the Razer stuff I don't tink

IhToN, 2017-05-18:
z3ntu, the problem started after I've installed the razerkbd drivers.

I've been trying to test some other things to locate what is the problem and it seems that it messed up the keys.

I'll be trying out to locate where this is going and after I find a solution I'll post it here for future problems.
Thanks for your support.

terrycain, 2017-05-18:
If you find any more evidence feel free to reopen

closed 2017-05-18 by terrycain

IhToN, 2017-05-18:
It seems that somehow, after unbinding the keyboard and binding it again to the razerkbd the system updated assistive technologies changing how the keyboard works to act as a mouse.

Thanks for your assistance :D

Assaf14, 2021-05-06:
> It seems that somehow, after unbinding the keyboard and binding it again to the razerkbd the system updated assistive technologies changing how the keyboard works to act as a mouse.
> 
> Thanks for your assistance :D

Hey ihToN, I have the EXACT problem as you had. every thing is peaty much the same. My key-board is recognized as a mouse in my device manager. Also, My keyboard numpad also doesn't work but the num IK does work  and make a sounds when typed. Additionally, the buttons on keyboard work but do weird stuff and don't work as numbers.