openrazer/openrazer #284: Issues with installation

pnelego, 2017-05-18:
So i have bounced around packages for razer products for a while. such as the "chroma" drivers.
```
Unfortunately I keep encountering this error:
dpkg: error processing archive /var/cache/apt/archives/razer-kernel-modules-dkms_1.1.12-0ubuntu1_amd64.deb (--unpack):
 trying to overwrite '/usr/src/razer_chroma_driver-1.0.0/driver/razerfirefly_driver.h', which is also in package razer-chroma-driver 1.0.0

```

is there a way to fix this?

z3ntu, 2017-05-18:
Please run `sudo apt purge razer-chroma-driver`

pnelego, 2017-05-18:
@z3ntu 
returned the following:
`The following packages have unmet dependencies:
 razer-daemon : Depends: razer-kernel-modules-dkms but it is not going to be installed
                Recommends: python3-razer but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
`

z3ntu, 2017-05-18:
```
sudo apt remove python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
sudo apt --fix-broken install
sudo apt remove razer-chroma-driver
```

pnelego, 2017-05-18:
That did the trick! Thank you so much.

closed 2017-05-18 by pnelego

z3ntu, 2017-05-18:
np :)