openrazer/openrazer #291: Daemon doesn't work with new sysfs probing

spstarr, 2017-05-26:
Debugging the daemon it seems with kernel 4.12-rc2 (at this time) it fails to load anything.

It looks like the objects the daemon is trying to find are gone:

Default is now:
/sys/devices/pci0000:00/0000:00:14.0/usb3/3-5/3-5.2/3-5.2:1.0/0003:1532:021A.0022/

```
$ ls -la /sys/bus/hid/devices/0003\:1532\:021A.0022/
total 0
drwxr-xr-x 5 root root    0 May 26 03:07 .
drwxr-xr-x 5 root root    0 May 26 03:07 ..
-r--r--r-- 1 root root 4096 May 26 03:07 country
lrwxrwxrwx 1 root root    0 May 26 03:07 driver -> ../../../../../../../../bus/hid/drivers/hid-generic
drwxr-xr-x 3 root root    0 May 26 03:07 hidraw
drwxr-xr-x 3 root root    0 May 26 03:07 input
-r--r--r-- 1 root root 4096 May 26 03:07 modalias
drwxr-xr-x 2 root root    0 May 26 03:07 power
-r--r--r-- 1 root root 4096 May 26 03:07 report_descriptor
lrwxrwxrwx 1 root root    0 May 26 03:07 subsystem -> ../../../../../../../../bus/hid
-rw-r--r-- 1 root root 4096 May 26 03:07 uevent
```
In device_base.py, the match() method fails because the pattern isn't valid anymore.
```
- ^[0-9A-F]{4}:1532:021A\.[0-9A-F]{4}$ 
+[0-9A-F]{4}:1532:021A\.[0-9A-F]{4}$
```
Any word on when this will be ported to support newer sysfs hid structure?

kernel: 4.12.0-rc2+ #1 SMP Tue May 23 04:09:51 EDT 2017 x86_64 x86_64 x86_64 GNU/Linux

Also wasn't there a plan to not have a kernel driver at all as gregkh mentioned we should not need a kernel driver just userspace?

My attempted startup

```
$ ./run_razer_daemon.py -F -v
['razer_daemon.hardware.mouse_mat', 'razer_daemon.hardware.__pycache__', 'razer_daemon.hardware.keyboards', 'razer_daemon.hardware.mouse', 'razer_daemon.hardware.core', 'razer_daemon.hardware.headsets', 'razer_daemon.hardware.mug']
2017-05-26 03:27:06 | razer                          | INFO     | Initialising Daemon (v1.1.12). Pid: 9505
2017-05-26 03:27:06 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2012 --------------------- (1532:010D)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderChroma ------------------- (1532:0043)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderElite -------------------- (1532:005C)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerImperiator ------------------------- (1532:002F)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
2017-05-26 03:27:06 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wired --------------------- (1532:0024)
[... 21 lines cut ...]
2017-05-26 03:27:06 | razer.device0                  | WARNING  | getting serial: [Errno 2] No such file or directory: '/sys/devices/pci0000:00/0000:00:14.0/usb3/3-5/3-5.2/3-5.2:1.0/0003:1532:021A.0022/device_serial'
2017-05-26 03:27:06 | razer.device0                  | DEBUG    | getting serial:  count:1
2017-05-26 03:27:06 | razer.device0                  | WARNING  | getting serial: [Errno 2] No such file or directory: '/sys/devices/pci0000:00/0000:00:14.0/usb3/3-5/3-5.2/3-5.2:1.0/0003:1532:021A.0022/device_serial'
2017-05-26 03:27:07 | razer.device0                  | DEBUG    | getting serial:  count:2
2017-05-26 03:27:07 | razer.device0                  | WARNING  | getting serial: [Errno 2] No such file or directory: '/sys/devices/pci0000:00/0000:00:14.0/usb3/3-5/3-5.2/3-5.2:1.0/0003:1532:021A.0022/device_serial'
2017-05-26 03:27:07 | razer.device0                  | DEBUG    | getting serial:  count:3
2017-05-26 03:27:07 | razer.device0                  | WARNING  | getting serial: [Errno 2] No such file or directory: '/sys/devices/pci0000:00/0000:00:14.0/usb3/3-5/3-5.2/3-5.2:1.0/0003:1532:021A.0022/device_serial'
2017-05-26 03:27:07 | razer.device0                  | DEBUG    | getting serial:  count:4
2017-05-26 03:27:07 | razer.device0                  | WARNING  | getting serial: [Errno 2] No such file or directory: '/sys/devices/pci0000:00/0000:00:14.0/usb3/3-5/3-5.2/3-5.2:1.0/0003:1532:021A.0022/device_serial'
2017-05-26 03:27:07 | razer.device0                  | DEBUG    | getting serial:  count:5
NEW OBJECT PATH: /org/razer/device/
Exception ignored in: <bound method RazerDevice.__del__ of RazerBlackWidowXChromaTournamentEdition:>
Traceback (most recent call last):
  File "/home/spstarr/tmp/razer-drivers/daemon/razer_daemon/hardware/device_base.py", line 420, in __del__
    self.close()
  File "/home/spstarr/tmp/razer-drivers/daemon/razer_daemon/hardware/device_base.py", line 332, in close
    self._close()
  File "/home/spstarr/tmp/razer-drivers/daemon/razer_daemon/hardware/keyboards.py", line 438, in _close
[... cut ...] urnamentEdition, self)._close()
  File "/home/spstarr/tmp/razer-drivers/daemon/razer_daemon/hardware/keyboards.py", line 34, in _close
    self.set_device_mode(0x00, 0x00)  # Device mode
  File "/home/spstarr/tmp/razer-drivers/daemon/razer_daemon/hardware/device_base.py", line 250, in set_device_mode
    with open(device_mode_path, 'wb') as mode_file:
PermissionError: [Errno 13] Permission denied: '/sys/devices/pci0000:00/0000:00:14.0/usb3/3-5/3-5.2/3-5.2:1.0/0003:1532:021A.0022/device_mode'
```

spstarr, 2017-05-26:
I need to bind driver.. ugh n/m

terrycain, 2017-05-26:
Am not currently interested due to the amount of effort and testing involved.

closed 2017-05-26 by terrycain

madscientist42, 2017-05-26:
If you're having sysfs probe problems...it's a bit of a hint.