openrazer/openrazer #292: Razer Mamba Tournament Edition not recognised on Ubuntu 16.04

ma-sadeghi, 2017-05-28:
Hi,

I have a Razer Mamba Tournament Edition, and I use Ubuntu 16.04 LTS. When I open Polychromatic Controller app, it says device not recognized. I'd be thankful if you could help me with the issue.

Cheers,
Amin

PS. Here's the warning I get after running Polychromatic Controller app.

[image]

z3ntu, 2017-05-29:
* What distro are you running? 
* Please post the output of `sudo dkms status razer_chroma_driver/1.0.0`

terrycain, 2017-05-29:
Also output of `dpkg -l | grep razer`

tildearrow, 2017-05-29:
Are you in the `plugdev` group?
If not, do `sudo usermod -a -G plugdev $USER`, reboot and try again.

(or does the PPA package set you in the plugdev group by default?)

z3ntu, 2017-05-29:
The screenshot implies that the daemon is running which is only possible when the user id in the group afaik.

ma-sadeghi, 2017-05-31:
Thank you all for your help.

@z3ntu
```
razer_chroma_driver, 1.0.0, 4.8.0-52-generic, x86_64: installed
```
@terrycain 
```
ii  python3-razer                               1.1.12-0ubuntu1                               all          Razer Python Library
ii  razer-daemon                                1.1.12-0ubuntu1                               amd64        Razer Device Daemon
ii  razer-doc                                   1.1.12-0ubuntu1                               all          Documentation and examples for the Razer Tools
ii  razer-kernel-modules-dkms                   1.1.12-0ubuntu1                               amd64        Razer Peripheral Drivers
```

z3ntu, 2017-05-31:
Dumb question: Did you already reboot after installing?

ma-sadeghi, 2017-05-31:
@z3ntu I think I did, but I just did and it's still not working.

z3ntu, 2017-05-31:
Can you run `sudo modprobe razermouse; lsmod | grep razer`

ma-sadeghi, 2017-05-31:
```
modprobe: FATAL: Module razermouse not found in directory /lib/modules/4.8.0-53-generic
```

z3ntu, 2017-05-31:
`sudo dkms install razer_chroma_driver/1.0.0`

ma-sadeghi, 2017-05-31:
```
Error! Your kernel headers for kernel 4.8.0-53-generic cannot be found.
Please install the linux-headers-4.8.0-53-generic package,
or use the --kernelsourcedir option to tell DKMS where it's located
```

z3ntu, 2017-05-31:
Then you gotta install your kernel headers. On Ubuntu it's most likely linux-headers-generic.

ma-sadeghi, 2017-05-31:
@z3ntu 
```
$ sudo modprobe razermouse; lsmod | grep razer
razermouse             61440  0
hid                   122880  3 razermouse,hid_generic,usbhid
```

ma-sadeghi, 2017-05-31:
@z3ntu After running `sudo madprobe razermouse`, I rebooted the system and it works now! Thanks a lot for your help/time :)

closed 2017-05-31 by ma-sadeghi