openrazer/openrazer #293: Naga Hex V2 mechanical buttons not working.

dssouza-ti, 2017-05-29:
Hello!

When I plugged my Naga Hex V2 (1532:0050) into my ubuntu 16.04, the 1-7 buttons on the side wheel were mapped to the 1-7 numbers on the keyboard. I was looking up for software to configure this mappings, and found your drivers (Congrats on the development, by the way), but as soon as I install razer-drivers, the 1-7 buttons stopped working entirely. I can't even get a mapping when I run xev and press any of the buttons. I couldn't find a open issue for this matter, so here am I. I'll be glad if you can help me ;)

Thanks in advance.

Diego Souza

terrycain, 2017-05-29:
If the numbers stop working that means the daemon is receiving the keypresses, some macro logic exists to remap them (its pretty rubbish atm) but im working on something better, which is neeearly finished (just need some time to put some finishing touches on it). Closing as have a few other issues open related to it.

closed 2017-05-29 by terrycain

dssouza-ti, 2017-05-29:
Thank you for the quick answer. :)

Since this is closed, how do I know when will this changes be implemented?

terrycain, 2017-05-29:
Its the only major thing im working on at the moment, so it'll be out in the next release or the one after. I assmue you'll know when decent functionality is added to the UIs like polychromatic