openrazer/openrazer #294: Razer Ripsaw! Driver Support!

xImInsanity, 2017-06-01:
I Like To Make YouTube Videos! Which Is Why I Bought The Razer Ripsaw! My OBS Wasn't Working On Windows So I Installed Linux And it Was Working But I Would Like To Use My Capture Card!

z3ntu, 2017-06-01:
Duplicate of #156

closed 2017-06-01 by z3ntu

primumprime, 2019-04-27:
Which windows were you using? The device (razer ripsaw) is very temperamental. I have found in order to get it to work (on windows) you have to make sure your display resolution on PC is the same as the display resolution on the game while also making sure the display resolution you are using for both devices is a supported resolution by razer on the box.   On OBS the device is considered a video capture device so look for the device there. Lastly if no signal is showing up for your xbox one or playstation 4 (if that is what you are streaming) you have to go on the settings on each (either the xbox or ps4) and turn off the pchp feature.  (i think that is what it is called) the feature prevents users from streaming video content like netflix hulu etc. to OBS. When this feature is turned off you can stream your video gameplay. 

For linux the device right now does say it is not compatible with linux. however it is worth a shot check and see if these trouble shooting methods work for linux and the device shows up.