openrazer/openrazer #295: Wrong dependence requirement for Debian 8

Mte90, 2017-06-03:
I use Debian Sid and I cannot install the package but:
```
python3-evdev : Diepends: python3 (< 3.5) but the version 3.5.3-1 was to be installed
```
So for me is not possible to install, it is possible to update the requirement for that dependence?

z3ntu, 2017-06-03:
You followed the instructions on the website, right?

Mte90, 2017-06-03:
yes, seems only that for that dependence for me is impossible to install the daemon.
I think that only update the require will unblock my situation.

z3ntu, 2017-06-03:
Do you use Debian 8 or Sid?

z3ntu, 2017-06-03:
But anyways, if you have `Debian_Stretch` in your `/etc/apt/sources.list.d/openrazer-meta.list` file, please change it to either `Debian_8.0`, `Debian_9.0` or `Debian_Sid` depending on what you are using. Then it should work fine. :) Please report back!

Mte90, 2017-06-03:
With `Debian_Sid` now is working :-)
Usually the debian repo does not support debian sid so this was an happy resolving :-)

closed 2017-06-03 by Mte90