openrazer/openrazer #296: Support for Orweaver Chroma

DmitryBurstein, 2017-06-03:
Please add support for Device ID 1532:0207 (Orbweaver Chroma).
Thanks ahead.

z3ntu, 2017-06-03:
Please test the pull request #188

DmitryBurstein, 2017-06-04:
I probably am having trouble installing it: my original installation is Fedora's rpm. I downloaded the #188, made 'all' and 'install' (as root), and rebooted. The log file (~/.razer-service/logs/razer.log) shows the Orbweaver Chroma 'specification' present, the razerkbd is loaded, but /sys/bus/hid/drivers/razerkbd has no devices in it. Moreover, 'echo -n "0003:1532:0207.0005" > bind' (this is the mouse part) gives me '-bash: echo: write error: No such device', while I'm root, and the file is writable. Checking the /sys/bus/hid/drivers/hid-generic reveals no Razer devices in it either. Any further instructions would be welcome.

z3ntu, 2017-06-06:
I'll write you instructions tomorrow hopefully.

z3ntu, 2017-06-09:
Sorry for the delay. 

But I got something even better for you: a prebuilt [rpms.zip](https://github.com/terrycain/razer-drivers/files/1064744/rpms.zip) you can install with (download the rpms.zip first and go into the directory you have the zip):

```
mkdir rpms
unzip -d rpms rpms.zip
sudo dnf install rpms/*.rpm
```
Then add your user to the plugdev group if you haven't already (`sudo gpasswd -a $USER plugdev`) and reboot. Then please test everything as we can merge it then after it's tested and working.

DmitryBurstein, 2017-06-11:
Great, thanks! This one works - kind of. Now I can see the device in Polychromatic, change the lightning effects and brightness, and it works. Unfortunately, everything else has stopped working: no keypress or joystick movement did generate any xev at all.

terrycain, 2017-06-15:
ok keypresses missing is the daemon grabbing them all, so that means its working. Macro stuff is rubbish atm but am working on a better implementation

lighting sounds like its setting brightness to 0 and then setting it back to the original value, normally its a bug in the code that handles screensavers

DmitryBurstein, 2017-06-15:
Thanks. Mostly, the lighting is restored to its initial value and effect (spectrum in my case), but from time to time it just remains solid zero.
As to the key/joystick presses - wouldn't it be better to implement a default scheme working - with no need to reassign everything manually? Just a suggestion.
Is anything else in particular you'd like me to test?

DmitryBurstein, 2017-06-21:
Would you like me to test something else - in order to release it?

z3ntu, 2017-06-21:
The PR is already merged so it should be included in the next release. So if you say, everything works, then it's great 😉

sKreevA4, 2017-06-30:
Hi, I'm new to all of this. (Both having razer peripherals, and this is actually my first comment on github). May I ask if/when this will come to "released"? Also, without having the particular distros with dnf, is there another way to install said driver? Thanks in advance

terrycain, 2017-07-01:
Version 1.1.14 is out on launchpad, non Ubuntu packages will be out shortly.

closed 2017-07-01 by terrycain