openrazer/openrazer #297: Support for Kraken 7.1 Classic

Team1721Zach, 2017-06-04:
Please add support for Device ID 1532:0501 (Kraken 7.1 Classic).
Thanks ahead.

z3ntu, 2017-06-04:
* Please post the output of
`lsusb -d 1532:0501 -v`
and
`ls -lah /dev/input/by-id/`
* Read [this](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol) because we probably need packet captures.
* Screenshots of all available options on Razer Synapse are needed.
* And lastly, the product & model number of the device for documentation reasons please.

Team1721Zach, 2017-06-20:
[Razer Driver.zip](https://github.com/terrycain/razer-drivers/files/1087191/Razer.Driver.zip)

Team1721Zach, 2017-06-20:
Hopefully that's everything!

terrycain, 2017-06-25:
What OS you on, I have a prototype that needs testing

Team1721Zach, 2017-06-28:
https://gist.github.com/anonymous/82dab565786d7cd24b8a0602a45a5953

I got the snapshots from a fresh installation of Windows 10 running in VirtualBox

Team1721Zach, 2017-06-28:
How does one test your prototype? I tried installing the 'feature_kraken71_classic_0501' branch but polychromatic wouldn't recognize it.

z3ntu, 2017-06-28:
Please run the following commands and the post the output of them.
```
lsmod | grep razer
sudo modprobe razerkraken
pgrep razer-daemon
```

Team1721Zach, 2017-07-05:
lsmod | grep razer

    razermouse             61440  0
    hid                   118784  5 i2c_hid,hid_multitouch,razermouse,hid_generic,usbhid
'sudo modprobe razerkraken' yelded no output

pgrep razer-daemon

    2752

ps aux | grep poly && ps aux | grep razer

    zachary   2584  0.5  0.6 2638512 84864 ?       Sl   23:47   0:01 polychromatic-controller
    zachary   3251  0.0  0.0  14224   976 pts/0    S+   23:52   0:00 grep --color=auto poly
    zachary   2752  0.0  0.2 369628 34912 ?        Sl   23:47   0:00 razer-daemon
    zachary   3253  0.0  0.0  14224  1008 pts/0    S+   23:52   0:00 grep --color=auto razer

Team1721Zach, 2017-07-05:
[image]

New poly error

Team1721Zach, 2017-07-05:
I did

     ~ $ make
     ~ $ sudo make install_i_know_what_i_am_doing
to install it

I'm running Linux Mint 18.1 (upgraded)

z3ntu, 2017-07-05:
It's no wonder there are errors because you should never run the install* target unless **you know what you are doing**. I'll send you debs to install later today. If you can't wait you can run from the root of the repo (correct branch please) `scripts/make_debs.sh` yourself and then run `sudo dpkg -i dist/*.deb`.

terrycain, 2017-07-05:
Install debs from here - https://drive.google.com/open?id=0BydYgD2LzDUCeHR4SnJFN25BeFU
Reboot

Team1721Zach, 2017-07-06:
[image]

z3ntu, 2017-07-06:
Please run `systemctl --user status razer-daemon.service` and post the output here

Team1721Zach, 2017-07-06:
I rebooted before I saw your comment (sorry) and polychromatic seems to be working now

z3ntu, 2017-07-06:
Well than it's fine 😄

Team1721Zach, 2017-07-06:
[image]

Team1721Zach, 2017-07-06:
I installed the *.deb from https://drive.google.com/open?id=0BydYgD2LzDUCeHR4SnJFN25BeFU

z3ntu, 2017-07-06:
So the Kraken doesn't get recognized? Please run `ls -al /sys/bus/hid/drivers/razerkraken/0*/` and post the output

Team1721Zach, 2017-07-06:
ls: cannot access '/sys/bus/hid/drivers/razerkraken/0*/': No such file or directory

Team1721Zach, 2017-07-06:
```
zachary@zachary-GL702VM ~ $ xinput
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ Razer Razer DeathAdder Chroma           	id=9	[slave  pointer  (2)]
⎜   ↳ Razer Razer DeathAdder Chroma           	id=10	[slave  pointer  (2)]
⎜   ↳ ASASTeK COMPUTER INC. ROG MacroKey      	id=14	[slave  pointer  (2)]
⎜   ↳ ASASTeK COMPUTER INC. ROG MacroKey      	id=15	[slave  pointer  (2)]
⎜   ↳ ELAN1200:00 04F3:303C Touchpad          	id=16	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Video Bus                               	id=7	[slave  keyboard (3)]
    ↳ Sleep Button                            	id=8	[slave  keyboard (3)]
    ↳ Razer Razer DeathAdder Chroma           	id=11	[slave  keyboard (3)]
    ↳ Yubico Yubikey 4 OTP+U2F+CCID           	id=12	[slave  keyboard (3)]
    ↳ ASASTeK COMPUTER INC. ROG MacroKey      	id=13	[slave  keyboard (3)]
    ↳ Asus WMI hotkeys                        	id=17	[slave  keyboard (3)]
    ↳ AT Translated Set 2 keyboard            	id=18	[slave  keyboard (3)]
```

z3ntu, 2017-07-06:
`sudo journalctl -b | grep -i razer_mount`

Team1721Zach, 2017-07-06:
I unplugged and replugged it back in and now I get this as an output from 'ls -al /sys/bus/hid/drivers/razerkraken/0*/'
```
total 0
drwxr-xr-x 3 root plugdev    0 Jul  6 16:02 .
drwxr-xr-x 5 root root       0 Jul  6 16:02 ..
-rw-rw---- 1 root plugdev 4096 Jul  6 16:02 device_mode
-r--r----- 1 root plugdev 4096 Jul  6 16:02 device_serial
-r--r----- 1 root plugdev 4096 Jul  6 16:02 device_type
lrwxrwxrwx 1 root plugdev    0 Jul  6 16:02 driver -> ../../../../../../../bus/hid/drivers/razerkraken
-r--r----- 1 root plugdev 4096 Jul  6 16:02 firmware_version
-r--r----- 1 root plugdev 4096 Jul  6 16:02 matrix_current_effect
--w--w---- 1 root plugdev 4096 Jul  6 16:02 matrix_effect_none
-rw-rw---- 1 root plugdev 4096 Jul  6 16:02 matrix_effect_static
-r--r--r-- 1 root plugdev 4096 Jul  6 16:02 modalias
drwxr-xr-x 2 root plugdev    0 Jul  6 16:02 power
lrwxrwxrwx 1 root plugdev    0 Jul  6 16:02 subsystem -> ../../../../../../../bus/hid
-rw-rw---- 1 root plugdev 4096 Jul  6 16:02 test
-rw-r--r-- 1 root plugdev 4096 Jul  6 16:02 uevent
-r--r----- 1 root plugdev 4096 Jul  6 16:02 version
```

Team1721Zach, 2017-07-06:
```
Jul 06 15:56:28 zachary-GL702VM razer_mount[722]: Driver razermouse
Jul 06 15:56:28 zachary-GL702VM razer_mount[720]: Driver razermouse
Jul 06 15:56:28 zachary-GL702VM razer_mount[721]: Driver razermouse
Jul 06 15:56:28 zachary-GL702VM razer_mount[723]: Device_ID 0003:1532:0043.0003
Jul 06 15:56:28 zachary-GL702VM razer_mount[724]: Device_ID 0003:1532:0043.0002
Jul 06 15:56:28 zachary-GL702VM razer_mount[725]: Device_ID 0003:1532:0043.0001
Jul 06 15:56:28 zachary-GL702VM razer_mount[727]: Unbinding 0003:1532:0043.0002 from hid-generic
Jul 06 15:56:28 zachary-GL702VM razer_mount[726]: Unbinding 0003:1532:0043.0003 from hid-generic
Jul 06 15:56:28 zachary-GL702VM razer_mount[728]: Unbinding 0003:1532:0043.0001 from hid-generic
Jul 06 15:56:28 zachary-GL702VM razer_mount[729]: Binding 0003:1532:0043.0002 to razermouse
Jul 06 15:56:28 zachary-GL702VM razer_mount[730]: Binding 0003:1532:0043.0003 to razermouse
Jul 06 15:56:28 zachary-GL702VM razer_mount[731]: Binding 0003:1532:0043.0001 to razermouse
Jul 06 15:56:28 zachary-GL702VM razer_mount[737]: Finished binding 0003:1532:0043.0001
Jul 06 15:56:28 zachary-GL702VM razer_mount[738]: Finished binding 0003:1532:0043.0002
Jul 06 15:56:28 zachary-GL702VM razer_mount[739]: Finished binding 0003:1532:0043.0003
Jul 06 15:56:28 zachary-GL702VM razer_mount[741]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0043.0002/
Jul 06 15:56:28 zachary-GL702VM razer_mount[740]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0043.0001/
Jul 06 15:56:28 zachary-GL702VM razer_mount[742]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0043.0003/
Jul 06 15:56:29 zachary-GL702VM razer_mount[746]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0043.0003/
Jul 06 15:56:29 zachary-GL702VM razer_mount[748]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0043.0001/
Jul 06 15:56:29 zachary-GL702VM razer_mount[747]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0043.0002/
Jul 06 16:02:36 zachary-GL702VM razer_mount[4862]: Driver razerkraken
Jul 06 16:02:36 zachary-GL702VM razer_mount[4863]: Device_ID 0003:1532:0501.000A
Jul 06 16:02:36 zachary-GL702VM razer_mount[4864]: Unbinding 0003:1532:0501.000A from hid-generic
Jul 06 16:02:36 zachary-GL702VM razer_mount[4865]: Binding 0003:1532:0501.000A to razerkraken
Jul 06 16:02:36 zachary-GL702VM kernel: CPU: 2 PID: 4861 Comm: razer_mount Tainted: P           OE   4.4.0-79-generic #100-Ubuntu
Jul 06 16:02:36 zachary-GL702VM razer_mount[4871]: Finished binding 0003:1532:0501.000A
Jul 06 16:02:36 zachary-GL702VM razer_mount[4872]: Changing group /sys/bus/hid/drivers/razerkraken/0003:1532:0501.000A/
Jul 06 16:02:36 zachary-GL702VM razer_mount[4874]: Changed group /sys/bus/hid/drivers/razerkraken/0003:1532:0501.000A/

```

Team1721Zach, 2017-07-06:
[image]
still nothing here tho

z3ntu, 2017-07-06:
`python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"`

Team1721Zach, 2017-07-06:
```
zachary@zachary-GL702VM ~ $ python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
[<RazerMouse PM1513121709918>]
```

z3ntu, 2017-07-06:
`cat /sys/bus/hid/drivers/razerkraken/0*/device_type`

Team1721Zach, 2017-07-06:
```
zachary@zachary-GL702VM ~ $ cat /sys/bus/hid/drivers/razerkraken/0*/device_type
Killed

```

z3ntu, 2017-07-06:
`dmesg | tail -n20`

Team1721Zach, 2017-07-06:
```
zachary@zachary-GL702VM ~ $ dmesg | tail -n20
[  715.832658]  ffff8803667bbdd0 ffffffff8155dd70 ffffffff8183e8c2 ffff8803aff68780
[  715.832663]  ffff8803667bbdf0 ffffffff8128ffcc 0000000000000000 ffff8803667bbf18
[  715.832668]  ffff8803667bbe00 ffffffff8128e7e3 ffff8803667bbe70 ffffffff81233cec
[  715.832673] Call Trace:
[  715.832682]  [<ffffffff8155dd70>] dev_attr_show+0x20/0x50
[  715.832688]  [<ffffffff8183e8c2>] ? mutex_lock+0x12/0x30
[  715.832696]  [<ffffffff8128ffcc>] sysfs_kf_seq_show+0xbc/0x130
[  715.832702]  [<ffffffff8128e7e3>] kernfs_seq_show+0x23/0x30
[  715.832708]  [<ffffffff81233cec>] seq_read+0xec/0x390
[  715.832715]  [<ffffffff8128f0da>] kernfs_fop_read+0x10a/0x160
[  715.832720]  [<ffffffff81347d50>] ? security_file_permission+0xa0/0xc0
[  715.832727]  [<ffffffff8120ecc8>] __vfs_read+0x18/0x40
[  715.832732]  [<ffffffff8120f296>] vfs_read+0x86/0x130
[  715.832738]  [<ffffffff8120ffe5>] SyS_read+0x55/0xc0
[  715.832744]  [<ffffffff81840a72>] entry_SYSCALL_64_fastpath+0x16/0x71
[  715.832747] Code: ce 48 c7 c7 20 f1 87 c0 48 89 e5 e8 db 07 91 c0 31 c0 5d c3 0f 1f 44 00 00 55 48 8b 87 a0 00 00 00 48 c7 c6 88 f5 87 c0 48 89 e5 <0f> b7 40 32 66 2d 01 05 66 83 f8 0f 77 0b 0f b7 c0 48 8b 34 c5 
[  715.832804] RIP  [<ffffffffc087d0b7>] razer_attr_read_device_type+0x17/0x40 [razerkraken]
[  715.832810]  RSP <ffff8803667bbdb0>
[  715.832812] CR2: 0000000000000032
[  715.832816] ---[ end trace ba80eec7390124da ]---

```

z3ntu, 2017-07-06:
`dmesg | tail -n100` (need more lines)

Team1721Zach, 2017-07-06:
```
[  378.952166]  [<ffffffff812901ea>] sysfs_create_file_ns+0x2a/0x30
[  378.952172]  [<ffffffff8155d4e2>] device_create_file+0x42/0x90
[  378.952178]  [<ffffffffc087e35c>] razer_kraken_probe+0x18c/0x280 [razerkraken]
[  378.952188]  [<ffffffffc00424f3>] hid_device_probe+0xd3/0x140 [hid]
[  378.952195]  [<ffffffff81562d32>] driver_probe_device+0x222/0x4a0
[  378.952201]  [<ffffffff81561214>] bind_store+0x114/0x160
[  378.952206]  [<ffffffff81560525>] drv_attr_store+0x25/0x30
[  378.952212]  [<ffffffff8128fb97>] sysfs_kf_write+0x37/0x40
[  378.952217]  [<ffffffff8128ef7d>] kernfs_fop_write+0x11d/0x170
[  378.952224]  [<ffffffff8120ea58>] __vfs_write+0x18/0x40
[  378.952229]  [<ffffffff8120f3e9>] vfs_write+0xa9/0x1a0
[  378.952234]  [<ffffffff8122dcb5>] ? set_close_on_exec+0x35/0x70
[  378.952240]  [<ffffffff812100a5>] SyS_write+0x55/0xc0
[  378.952247]  [<ffffffff81840a72>] entry_SYSCALL_64_fastpath+0x16/0x71
[  378.952251] ---[ end trace ba80eec7390124d8 ]---
[  378.957062] usbcore: registered new interface driver snd-usb-audio
[  379.066397] BUG: unable to handle kernel NULL pointer dereference at 0000000000000040
[  379.066407] IP: [<ffffffffc087d5fb>] razer_attr_read_get_serial+0x4b/0x110 [razerkraken]
[  379.066420] PGD 3b274f067 PUD 3b2755067 PMD 0 
[  379.066429] Oops: 0000 [#1] SMP 
[  379.066434] Modules linked in: snd_usb_audio snd_usbmidi_lib razerkraken(OE) rfcomm drbg ansi_cprng ctr ccm pci_stub vboxpci(OE) vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) bnep binfmt_misc nvidia_uvm(POE) nvidia_drm(POE) nvidia_modeset(POE) nvidia(POE) arc4 iwlmvm mac80211 snd_hda_codec_realtek snd_hda_codec_generic i915_bpo razermouse(OE) intel_ips i2c_algo_bit snd_hda_intel snd_hda_codec intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel snd_hda_core snd_hwdep kvm iwlwifi drm_kms_helper snd_pcm joydev drm cfg80211 hid_multitouch snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq snd_seq_device snd_timer btusb btrtl snd fb_sys_fops syscopyarea sysfillrect sysimgblt i2c_designware_platform i2c_designware_core asus_nb_wmi asus_wmi sparse_keymap mei_me mei idma64 virt_dma processor_thermal_device
[  379.066529]  shpchp intel_soc_dts_iosf intel_lpss_pci soundcore irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel aesni_intel hci_uart btbcm btqca btintel bluetooth aes_x86_64 lrw gf128mul glue_helper ablk_helper cryptd input_leds serio_raw wmi int3403_thermal tpm_crb intel_lpss_acpi intel_lpss int3402_thermal int340x_thermal_zone mac_hid int3400_thermal acpi_pad acpi_thermal_rel parport_pc ppdev lp parport autofs4 btrfs hid_generic usbhid raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear dm_mirror dm_region_hash dm_log r8169 ahci mii libahci i2c_hid hid pinctrl_sunrisepoint video pinctrl_intel fjes
[  379.066626] CPU: 1 PID: 2401 Comm: razer-daemon Tainted: P        W  OE   4.4.0-79-generic #100-Ubuntu
[  379.066633] Hardware name: ASUSTeK COMPUTER INC. GL702VM/GL702VM, BIOS GL702VM.303 10/06/2016
[  379.066638] task: ffff8803b198ad00 ti: ffff88038b788000 task.ti: ffff88038b788000
[  379.066643] RIP: 0010:[<ffffffffc087d5fb>]  [<ffffffffc087d5fb>] razer_attr_read_get_serial+0x4b/0x110 [razerkraken]
[  379.066655] RSP: 0018:ffff88038b78bd60  EFLAGS: 00010246
[  379.066660] RAX: ffff88038b78bd63 RBX: 0000000000000000 RCX: 0000000000000000
[  379.066664] RDX: 0000000000000000 RSI: 0000000000000004 RDI: ffff88038b78bd48
[  379.066668] RBP: ffff88038b78bdb0 R08: 0000000000007f00 R09: ffff88038b78bd63
[  379.066672] R10: 0000000000000016 R11: ffff8803b0706100 R12: ffffffff81a7f9c0
[  379.066676] R13: ffff8803580aa000 R14: ffff8803af868600 R15: 0000000000000001
[  379.066681] FS:  00007f4ad4c52700(0000) GS:ffff8803c6440000(0000) knlGS:0000000000000000
[  379.066685] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  379.066688] CR2: 0000000000000040 CR3: 00000003b26e8000 CR4: 00000000003406e0
[  379.066693] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[  379.066696] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[  379.066699] Stack:
[  379.066703]  007f162004000003 0000000000000000 0000000000000000 0000000000000000
[  379.066710]  0000000000000000 00000000ab92218a ffffffffc0880220 ffffffff81a7f9c0
[  379.066717]  ffff88038db5dd00 ffff8803af868600 ffff88038b78bdd0 ffffffff8155dd70
[  379.066724] Call Trace:
[  379.066738]  [<ffffffff8155dd70>] dev_attr_show+0x20/0x50
[  379.066747]  [<ffffffff8183e8c2>] ? mutex_lock+0x12/0x30
[  379.066757]  [<ffffffff8128ffcc>] sysfs_kf_seq_show+0xbc/0x130
[  379.066765]  [<ffffffff8128e7e3>] kernfs_seq_show+0x23/0x30
[  379.066773]  [<ffffffff81233cec>] seq_read+0xec/0x390
[  379.066781]  [<ffffffff8128f0da>] kernfs_fop_read+0x10a/0x160
[  379.066789]  [<ffffffff81347d50>] ? security_file_permission+0xa0/0xc0
[  379.066797]  [<ffffffff8120ecc8>] __vfs_read+0x18/0x40
[  379.066804]  [<ffffffff8120f296>] vfs_read+0x86/0x130
[  379.066812]  [<ffffffff8120ffe5>] SyS_read+0x55/0xc0
[  379.066821]  [<ffffffff81840a72>] entry_SYSCALL_64_fastpath+0x16/0x71
[  379.066824] Code: 49 89 d5 ba 20 00 00 00 48 83 ec 30 48 8b 9f a0 00 00 00 48 8d 7d b3 65 48 8b 04 25 28 00 00 00 48 89 45 d8 31 c0 e8 25 fe ff ff <80> 7b 40 00 4c 8d 63 40 74 30 4c 89 e2 48 c7 c6 9f f5 87 c0 4c 
[  379.066881] RIP  [<ffffffffc087d5fb>] razer_attr_read_get_serial+0x4b/0x110 [razerkraken]
[  379.066889]  RSP <ffff88038b78bd60>
[  379.066891] CR2: 0000000000000040
[... cut ...]
[  715.832744]  [<ffffffff81840a72>] entry_SYSCALL_64_fastpath+0x16/0x71
[  715.832747] Code: ce 48 c7 c7 20 f1 87 c0 48 89 e5 e8 db 07 91 c0 31 c0 5d c3 0f 1f 44 00 00 55 48 8b 87 a0 00 00 00 48 c7 c6 88 f5 87 c0 48 89 e5 <0f> b7 40 32 66 2d 01 05 66 83 f8 0f 77 0b 0f b7 c0 48 8b 34 c5 
[  715.832804] RIP  [<ffffffffc087d0b7>] razer_attr_read_device_type+0x17/0x40 [razerkraken]
[  715.832810]  RSP <ffff8803667bbdb0>
[  715.832812] CR2: 0000000000000032
[  715.832816] ---[ end trace ba80eec7390124da ]---
```

z3ntu, 2017-07-06:
Okay just copy your whole dmesg to pastebin or similar please. There's a bug our kernel driver...

Team1721Zach, 2017-07-06:
https://gist.githubusercontent.com/Team1721Zach/5bf1fac22166ce5ef0d21c574b501b78/raw/8f0cda819eda84d086ecbd1ce79a9722f7867b01/dmsg.txt

z3ntu, 2017-07-06:
We'll continue tomorrow but thanks already for the quick responses. That makes this process much easier :)

Team1721Zach, 2017-07-06:
Thank you for your work in the Linux & Open Source community! Your efforts do not go unnoticed! :smile:

z3ntu, 2017-07-07:
[feature_kraken71_classic_0501.zip](https://github.com/terrycain/razer-drivers/files/1130268/feature_kraken71_classic_0501.zip)
Please install these debs and reboot. Then just open polychromatic and upload the last 500 lines of your dmesg (`dmesg | tail -n500`)

Team1721Zach, 2017-07-07:
https://gist.githubusercontent.com/Team1721Zach/095b8d21d305efaadfad010d66dd446e/raw/fe7fcff174d13e5f1973398e2af23bfd9356e60b/dmesg.txt

Team1721Zach, 2017-07-08:
zachary@zachary-GL702VM ~ $ cat /sys/bus/hid/drivers/razerkraken/0*/device_type
   Razer Kraken 7.1 Classic (??)

z3ntu, 2017-07-08:
Great. Now run 
`python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"` and `systemctl --user status razer-daemon.service` :)

Team1721Zach, 2017-07-08:
```
zachary@zachary-GL702VM ~ $ python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
[<RazerMouse PM1513121709918>]

zachary@zachary-GL702VM ~ $ systemctl --user status razer-daemon.service
● razer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/razer-daemon.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
     Docs: man:razer-daemon(8)
```

Team1721Zach, 2017-07-08:
I restarted the daemon via polychromatic and the output changed

```
zachary@zachary-GL702VM ~ $ python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
[<RazerMouse PM1513121709918>, <RazerDevice HN000003039161088>]

zachary@zachary-GL702VM ~ $ systemctl --user status razer-daemon.service
● razer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/razer-daemon.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
     Docs: man:razer-daemon(8)
```

Team1721Zach, 2017-07-08:
[image]
also, yay!

z3ntu, 2017-07-08:
Can you test all effects?

z3ntu, 2017-07-08:
"all" are not much unfortunately 😄

Team1721Zach, 2017-07-08:
The static effects have color options, the headset has one static color though
[image]

z3ntu, 2017-07-08:
Yeah we know that. Does the one color work though?

Team1721Zach, 2017-07-08:
yes I can now turn my headset light on and off!

z3ntu, 2017-07-08:
`cat /sys/bus/hid/drivers/razerkraken/0*/device_serial`

Team1721Zach, 2017-07-08:
```
zachary@zachary-GL702VM ~ $ cat /sys/bus/hid/drivers/razerkraken/0*/device_serial
HN000003039161088
```

Team1721Zach, 2017-07-24:
Something happened and I can't see the headset in Polychromatic or Razer Genie anymore..

Team1721Zach, 2017-07-24:
Here is all my post start-up dmesg: https://pastebin.com/raw/ypW6tGTP
(sorry I used Pastebin, Gist Hub seems borked)

Team1721Zach, 2017-07-24:
```
zachary@zachary-GL702VM ~ $ cat /sys/bus/hid/drivers/razerkraken/0*/device_type
cat: '/sys/bus/hid/drivers/razerkraken/0*/device_type': No such file or directory

```

Team1721Zach, 2017-07-24:
```
zachary@zachary-GL702VM ~ $ python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
[<RazerMouse PM1513121709918>, <RazerKeyboard IO1445038423241>]

```

z3ntu, 2017-07-24:
It's possible that apt updated the driver to a newer version which isn't from the special support branch.

Team1721Zach, 2017-07-24:
Oh, darn.

terrycain, 2017-07-30:
Shoudl be released with newest version 1.1.15 once its out, update reboot and then check, if its all good, close the issue :smile:

closed 2018-01-28 by z3ntu