openrazer/openrazer #299: Razer Naga Epic Chroma support (003e/003f)

kived, 2017-06-06:
There's some talk about the Naga Epic Chroma mixed in with #59, but it's not the same device (and using the associated branch doesn't work). It does work if treated like a Naga Hex V2 in some ways. DPI can be changed (most important part for me), and while the lighting effects cannot be changed, the brightness controls do work. Logo settings have no effect at all (I don't think the logo glows on this mouse).

https://gist.github.com/kived/086b0ae08a6a9cf7a30f1e068a87ef7e
(I started with the feature_nagachroma_0x0053 branch and rebased off master)

z3ntu, 2017-06-06:
* Please post the output of
`lsusb -d 1532:003e -v`
and
`ls -lah /dev/input/by-id/`
* Read [this](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol) because we probably need packet captures.
* Screenshots of all available options on Razer Synapse are needed.
* And lastly, the product & model number of the device for documentation reasons please.

kived, 2017-06-06:
Here's the output for 003f. This is the dock when using the mouse in wireless mode. The mouse can also be wired directly via USB in which case the id is 003e. There is no difference in the output of lsusb other than the device id.

```
$ lsusb -d 1532:003f -v

Bus 001 Device 003: ID 1532:003f Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x003f 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
[... 62 lines cut ...]
lrwxrwxrwx 1 root root   9 Jun  6 13:09 usb-Logitech_USB_Receiver-if01-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Jun  6 13:09 usb-Logitech_USB_Receiver-if01-mouse -> ../mouse1
lrwxrwxrwx 1 root root  10 Jun  6 13:38 usb-Razer_Razer_Naga_Epic_Chroma_Dock-event-mouse -> ../event15
lrwxrwxrwx 1 root root   9 Jun  6 13:38 usb-Razer_Razer_Naga_Epic_Chroma_Dock-if01-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 Jun  6 13:38 usb-Razer_Razer_Naga_Epic_Chroma_Dock-if02-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 Jun  6 13:38 usb-Razer_Razer_Naga_Epic_Chroma_Dock-mouse -> ../mouse0
```

Direct USB (003e):
```
$ ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 180 Jun  6 13:42 .
drwxr-xr-x 4 root root 560 Jun  6 13:42 ..
lrwxrwxrwx 1 root root   9 Jun  6 13:09 usb-Logitech_USB_Receiver-event-kbd -> ../event6
lrwxrwxrwx 1 root root   9 Jun  6 13:09 usb-Logitech_USB_Receiver-if01-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 Jun  6 13:09 usb-Logitech_USB_Receiver-if01-mouse -> ../mouse1
lrwxrwxrwx 1 root root  10 Jun  6 13:42 usb-Razer_Razer_Naga_Epic_Chroma-event-mouse -> ../event15
lrwxrwxrwx 1 root root   9 Jun  6 13:42 usb-Razer_Razer_Naga_Epic_Chroma-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 Jun  6 13:42 usb-Razer_Razer_Naga_Epic_Chroma-if02-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 Jun  6 13:42 usb-Razer_Razer_Naga_Epic_Chroma-mouse -> ../mouse0
```

Model No: RC30-012301
Product No: RZ01-01230100

kived, 2017-06-06:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

terrycain, 2017-06-15:
Need captures: (repeat for wired and wireless)
* 1 capture of slide dpi from min to max to min
* 1 capture of poll rate from 125, 250, 500
* a capture of each and every lighting option
* 1 capture of brightness off, dim, normal bright
* 1 capture of sleep mode idel min to max to min
* 1 capture of blink when low power min to max to min
* 1 capture of leaving the mouse not moving for ~10 min

drewhoener, 2017-09-03:
Any progress on this / anything I can do to help speed this along? It'd be nice to be able to configure my epic :)

terrycain, 2017-09-03:
^^

Fyro-Ing, 2017-09-07:
could you explain me how to take this capture ? 

with wireshark ? other ?

z3ntu, 2017-09-07:
https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

Fyro-Ing, 2017-11-26:
i've installed virtualbox, but mouse is not recognized as a Razer mouse.
is there a specific configuration to do this ?

z3ntu, 2017-11-26:
@Fyro-Ing Did you pass through the USB device to the VM? And please be aware that the mouse is then exclusively available for the VM as long as it's running so I recommend plugging in a second mouse to use for the "outside OS"/host.

Fyro-Ing, 2017-12-30:
all is good now with the [virtualbox documentation](https://forums.virtualbox.org/viewtopic.php?f=35&t=82639) (my problems : not in virtualbox user group and a virtualbox distro (not sure -- no try with only group user -- for that but in the doubt ^^ ) => no device)

now, i go to see samples on other issues to do the job correctly, i hope the next week, thanks

nyejyh, 2018-01-16:
I would like to contribute as I have this mouse too.  I understand we need the captures as @terrycain was asking for earlier.  Currently I'm stuck figuring out what channel I need to look at on Wireshark.  I already have the information such as Bus/Device numbers and applied the filters accordingly:
```
**Razer Naga Epic Chroma (Wired Mode)**
$ lsusb
Bus 003 Device 010: ID 1532:003e Razer USA, Ltd

**Razer Naga Epic Chroma (Wireless Mode)**
$ lsusb
Bus 003 Device 009: ID 1532:003f Razer USA, Ltd
```
```
**Razer Naga Epic Chroma (Wired Mode)**
Filter: usb.bus_id == 3 && usb.device_address == 10

**Razer Naga Epic Chroma (Wireless Mode)**
Filter: usb.bus_id == 3 && usb.device_address == 9
```
Please see what channels I have so far on Wireshark.  Which channel should I focus on?
[image]

nyejyh, 2018-01-17:
I have a zip file of the Razer Naga Epic Chroma captures in both wired and wireless modes as requested by @terrycain in earlier post.  Please note that the poll rate for this mouse is actually 125, 500, 1000.
[Razer Naga Epic Chroma Captures.zip](https://github.com/openrazer/openrazer/files/1638176/Razer.Naga.Epic.Chroma.Captures.zip)

nyejyh, 2018-01-19:
Any updates?  I can retake captures if needed.

z3ntu, 2018-01-19:
No, I don't think yours are good (and recognized by our scripts). I am just currently very busy learning PL/SQL 😉 (until tomorrow)

nyejyh, 2018-01-19:
Thanks for letting me know! :)  What can I do to make sure the captures are good and recognized by your scripts?  Do I need to throw the data into the colorizer?  I was following another example of posting the captures in which they were accepted, but maybe I did something wrong.

Also, good luck on your learning, it sounds like a lot of fun!

z3ntu, 2018-01-19:
"Fun".
I think it's getting recognized (@terrycain wrote the scripts btw) because it uses an older version of the protocol which is known by the scripts.

z3ntu, 2018-01-20:
I've pushed a commit making the first steps to being supported d35f8259514c13693ca04ffe0653ba3b6220302a
It compiles but needs more changes to work properly.

nyejyh, 2018-01-20:
I can't tell you how happy I am to hear about this! :smile:

z3ntu, 2018-01-21:
Okay I have a state that might work on the branch `feature_nagaepicchroma_0x003e`. Which distro do you use so I can give you instructions on how to compile the package from that branch?

nyejyh, 2018-01-21:
I used Debian 9 Stretch.

z3ntu, 2018-01-21:
Then
```
git clone https://github.com/openrazer/openrazer -b feature_nagaepicchroma_0x003e
cd openrazer/
./scripts/build_debs.sh
```
If you get an error, look into the log file that's printed and install the missing build dependencies listed there.

z3ntu, 2018-01-21:
I fixed the problem and rebased the `feature_nagaepicchroma_0x003e` branch on master. It's probably easiest if you delete the folder you might have cloned before and run the three commands again.

nyejyh, 2018-01-21:
Thanks, your fix solved a different issue I was running into.  I am now stuck at the part where I already updated linux-headers-generic to the latest version (4.9.65-3+deb9u2) but the error still throws a "Failed to generate deb files."

z3ntu, 2018-01-21:
Can you give me the contents of the log file?

nyejyh, 2018-01-21:
```
nyejyh@BananaHarvester:/tmp$ cat tmp.nmh2cAjHCM_deb_build_tmp/build.log 
dpkg-buildpackage: info: source package openrazer
dpkg-buildpackage: info: source version 2.2.0
dpkg-buildpackage: info: source distribution xenial
dpkg-buildpackage: info: source changed by Luca Weiss <luca@z3ntu.xyz>
 dpkg-source --before-build razer
dpkg-buildpackage: info: host architecture amd64
dpkg-source: warning: unknown information field 'Version' in input data in package's section of control info file
dpkg-checkbuilddeps: error: Unmet build dependencies: linux-headers-generic
dpkg-buildpackage: warning: build dependencies/conflicts unsatisfied; aborting
dpkg-buildpackage: warning: (Use -d flag to override.)
```

z3ntu, 2018-01-21:
Run `sed -i '/linux-headers-generic/d' debian/control` and try again. That package is not available on Debian as far as I know.

nyejyh, 2018-01-21:
Different error now, pasted below:
```
nyejyh@BananaHarvester:/tmp$ cat tmp.MJ91kfcxWY_deb_build_tmp/build.log
dpkg-buildpackage: info: source package openrazer
dpkg-buildpackage: info: source version 2.2.0
dpkg-buildpackage: info: source distribution xenial
dpkg-buildpackage: info: source changed by Luca Weiss <luca@z3ntu.xyz>
 dpkg-source --before-build razer
dpkg-buildpackage: info: host architecture amd64
dpkg-source: warning: unknown information field 'Version' in input data in package's section of control info file
 fakeroot debian/rules clean
dh clean --with-python3
   dh_testdir -O--with-python3
   debian/rules override_dh_auto_clean
make[1]: Entering directory '/tmp/tmp.MJ91kfcxWY_deb_build_tmp/razer'
/usr/bin/make lp_driver_clean
make[2]: Entering directory '/tmp/tmp.MJ91kfcxWY_deb_build_tmp/razer'
-e 
:: Cleaning OpenRazer kernel modules (Launchpad)
========================================
make -C "/lib/modules/4.9.0-5-amd64/build" SUBDIRS="/tmp/tmp.MJ91kfcxWY_deb_build_tmp/razer/driver" clean
make[3]: Entering directory '/usr/src/linux-headers-4.9.0-5-amd64'
make[3]: Leaving directory '/usr/src/linux-headers-4.9.0-5-amd64'
make[2]: Leaving directory '/tmp/tmp.MJ91kfcxWY_deb_build_tmp/razer'
make[1]: Leaving directory '/tmp/tmp.MJ91kfcxWY_deb_build_tmp/razer'
   dh_clean -O--with-python3
 dpkg-source -b razer
dpkg-source: warning: unknown information field 'Version' in input data in package's section of control info file
dpkg-source: error: can't build with source format '3.0 (quilt)': non-native package version does not contain a revision
dpkg-buildpackage: error: dpkg-source -b razer gave error exit status 255
```

z3ntu, 2018-01-22:
Sorry for that. Building the debs now works in my Debian Testing VM.

z3ntu, 2018-01-22:
Run the sed command in the git repository. The linux-headers-generic dependency is only valid for Ubuntu and that sed commands removes that dependency

nyejyh, 2018-01-22:
It works!  Does this mean you can integrate this into Open Razer?
```
nyejyh@BananaHarvester:~/openrazer$ ./scripts/build_debs.sh
Temp DIR: /tmp/tmp.0ghG4twamC_deb_build_tmp
Copying source to temporary build directory
Building deb packages

Deb packages build successfully. Copied to ./dist/
```

z3ntu, 2018-01-23:
You have to install the deb packages that are in the `dist/` directory (e.g. with `sudo dpkg -i ./dist/*.deb && sudo apt install -f`)

Fyro-Ing, 2018-01-23:
hello, how to build for ubuntu xenial ?

nyejyh, 2018-01-23:
Building all the deb packages succeed, however there were some error messages related to dependency problems:
```
nyejyh@BananaHarvester:~/openrazer$ sudo dpkg -i ./dist/*.deb && sudo apt install -f
Selecting previously unselected package openrazer-daemon.
(Reading database ... 163989 files and directories currently installed.)
Preparing to unpack .../openrazer-daemon_2.2.0-1_amd64.deb ...
Unpacking openrazer-daemon (2.2.0-1) ...
Selecting previously unselected package openrazer-doc.
Preparing to unpack .../openrazer-doc_2.2.0-1_all.deb ...
Unpacking openrazer-doc (2.2.0-1) ...
Selecting previously unselected package openrazer-kernel-modules-dkms.
Preparing to unpack .../openrazer-kernel-modules-dkms_2.2.0-1_amd64.deb ...
Unpacking openrazer-kernel-modules-dkms (2.2.0-1) ...
Selecting previously unselected package openrazer-meta.
Preparing to unpack .../openrazer-meta_2.2.0-1_all.deb ...
Unpacking openrazer-meta (2.2.0-1) ...
Selecting previously unselected package python3-openrazer.
Preparing to unpack .../python3-openrazer_2.2.0-1_all.deb ...
Unpacking python3-openrazer (2.2.0-1) ...
dpkg: dependency problems prevent configuration of openrazer-daemon:
 openrazer-daemon depends on python3-pyudev; however:
  Package python3-pyudev is not installed.
 openrazer-daemon depends on python3-evdev; however:
  Package python3-evdev is not installed.
 openrazer-daemon depends on python3-setproctitle; however:
  Package python3-setproctitle is not installed.
 openrazer-daemon depends on python3-notify2; however:
  Package python3-notify2 is not installed.
 openrazer-daemon depends on python3-daemonize; however:
  Package python3-daemonize is not installed.
 openrazer-daemon depends on xautomation; however:
  Package xautomation is not installed.
 openrazer-daemon depends on xdotool; however:
  Package xdotool is not installed.

dpkg: error processing package openrazer-daemon (--install):
 dependency problems - leaving unconfigured
Setting up openrazer-doc (2.2.0-1) ...
Setting up openrazer-kernel-modules-dkms (2.2.0-1) ...
Loading new openrazer-driver-1.0.0 DKMS files...
Building for 4.9.0-5-amd64
Building initial module for 4.9.0-5-amd64
Done.

razerkbd:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.9.0-5-amd64/updates/dkms/

razermouse.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.9.0-5-amd64/updates/dkms/

razerfirefly.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.9.0-5-amd64/updates/dkms/

razerkraken.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.9.0-5-amd64/updates/dkms/

razermug.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.9.0-5-amd64/updates/dkms/

razercore.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.9.0-5-amd64/updates/dkms/

depmod...

DKMS: install completed.
-e 
ADD YOUR USER TO THE PLUGDEV GROUP 'sudo gpasswd -a <USERNAME> plugdev'
dpkg: dependency problems prevent configuration of openrazer-meta:
 openrazer-meta depends on openrazer-daemon; however:
  Package openrazer-daemon is not configured yet.

dpkg: error processing package openrazer-meta (--install):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of python3-openrazer:
 python3-openrazer depends on python3-numpy; however:
  Package python3-numpy is not installed.

dpkg: error processing package python3-openrazer (--install):
 dependency problems - leaving unconfigured
Processing triggers for man-db (2.7.6.1-2) ...
Errors were encountered while processing:
 openrazer-daemon
 openrazer-meta
 python3-openrazer
```

Fyro-Ing, 2018-01-23:
you need to install dependencies : 
python3-pyudev python3-evdev python3-setproctitle python3-notify2 python3-daemonize xautomation xdotool

For ubuntu => ./scripts/build_debs.sh is OK

but daemon start with 2 similar errors : 

*** on session started (same as command line /usr/bin/openrazer-daemon &) : 
  File "/usr/bin/openrazer-daemon", line 178, in <module>
    run()
  File "/usr/bin/openrazer-daemon", line 173, in run
    logger=logger)
TypeError: __init__() got an unexpected keyword argument 'chdir'

*** with command line /usr/bin/openrazer-daemon

Traceback (most recent call last):
  File "/usr/bin/openrazer-daemon", line 178, in <module>
    run()
  File "/usr/bin/openrazer-daemon", line 173, in run
    logger=logger)
TypeError: __init__() got an unexpected keyword argument 'foreground'

nyejyh, 2018-01-23:
@Fyro-Ing Thanks for pointing that out, my bad.  Installed the packages as you mentioned with no errors.  Is this where I can install applications like Polychromatic to test the drivers?

Fyro-Ing, 2018-01-23:
After restart computer, yes if openrazer-daemon is running

z3ntu, 2018-01-23:
Some dependencies are only available in the ppa so adding that should help with these problems.

nyejyh, 2018-01-23:
Currently testing RazerGenie but running into errors.  Is the stable version of OpenRazer supposed to be installed before all this?

z3ntu, 2018-01-23:
What issues? And no, you need the 3 or 4 packages built from the support branch as it's not supported in master.

nyejyh, 2018-01-23:
All the dependencies have been installed without issues, so building deb packages have succeeded.  Currently seeing this when running RazerGenie:
```
● openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Tue 2018-01-23 11:50:30 PST; 24s ago
     Docs: man:openrazer-daemon(8)
  Process: 2353 ExecStart=/usr/bin/openrazer-daemon -Fv (code=exited, status=1/FAILURE)
 Main PID: 2353 (code=exited, status=1/FAILURE)
```

z3ntu, 2018-01-23:
Run "openrazer-daemon -Fv" in your terminal and check the output. I guess it's that your user is not a member of the plugdev group

nyejyh, 2018-01-23:
I'm pretty sure I added my user to the group:
```
nyejyh@BananaHarvester:~/openrazer$ history
...
  456  cd openrazer/
  457  sudo dpkg -i ./dist/*.deb && sudo apt install -f
  458  sudo gpasswd -a nyejyh plugdev

```
Here is the output of the command:
```
nyejyh@BananaHarvester:~/openrazer$ openrazer-daemon -Fv
Traceback (most recent call last):
  File "/usr/bin/openrazer-daemon", line 12, in <module>
    from openrazer_daemon.daemon import RazerDaemon, __version__
ModuleNotFoundError: No module named 'openrazer_daemon'
```

z3ntu, 2018-01-23:
What does `python3 -c "from distutils.sysconfig import get_python_lib; print(get_python_lib())"`
output and what does `ls -ald /usr/lib/python3/*/openrazer*` print?

nyejyh, 2018-01-23:
First output:
```
nyejyh@BananaHarvester:~/openrazer$ python3 -c "from distutils.sysconfig import get_python_lib; print(get_python_lib())"
/home/nyejyh/anaconda3/lib/python3.6/site-packages
```
Second output:
```
nyejyh@BananaHarvester:~/openrazer$ ls -ald /usr/lib/python3/*/openrazer*
drwxr-xr-x 4 root root 4096 Jan 23 11:46 /usr/lib/python3/dist-packages/openrazer
drwxr-xr-x 5 root root 4096 Jan 23 11:46 /usr/lib/python3/dist-packages/openrazer_daemon
```

z3ntu, 2018-01-23:
Did you install a custom version of python 3? Because the path from the first command should be /usr/lib/python3/dist-packages

z3ntu, 2018-01-23:
#503 was about another installation that pointed to the home directory for the lib path, maybe you can find something in there.

Fyro-Ing, 2018-01-23:
@z3ntu any ideas for my problem ? https://github.com/openrazer/openrazer/issues/299#issuecomment-359892093

z3ntu, 2018-01-23:
@fyro-ing seems you have an old version of python3-daemonize installed. So add the ppa (see our website) and do a system upgrade.

Fyro-Ing, 2018-01-23:
it's good but razergenie or polychromatic not detect device

if daemon log, seems to be ok 
2018-01-23 22:41:10 | razer                          | DEBUG    | Loaded device specification: RazerNagaEpicChromaWired ---------------- (1532:003E)

z3ntu, 2018-01-23:
Did you try to reboot?

Fyro-Ing, 2018-01-23:
yes, but same result

nyejyh, 2018-01-23:
Attempt to reinstall Python 3 resulted in a fresh copy of Debian 9.  Redid all the steps with successful building of debs.  Below is different error:
```
nyejyh@BananaHarvester:~$ openrazer-daemon -Fv
Traceback (most recent call last):
  File "/usr/bin/openrazer-daemon", line 178, in <module>
    run()
  File "/usr/bin/openrazer-daemon", line 173, in run
    logger=logger)
TypeError: __init__() got an unexpected keyword argument 'foreground'

```

z3ntu, 2018-01-24:
You can probably follow something similar to https://stackoverflow.com/a/7901603/3527128 (of course with adjusted paths) to achieve this; or you use the path Debian uses for adjusting that path.

evaue, 2018-01-24:
@z3ntu will this actually compile if I modify the PKGBUILD slightly for openrazer-git so it points to this branch?

z3ntu, 2018-01-24:
@evaue Yes, that will work (so `source=("$pkgbase::git+https://github.com/openrazer/openrazer.git#branch=feature_nagaepicchroma_0x003e")`)

nyejyh, 2018-01-25:
Sorry if this is a bit of a noob question.  How do I make the PYTHONPATH work?  I need to have Anaconda's version of Python to work on projects too, so would having both lines not commented in the file still work?  This is what I have in the bashrc:
```
# added by Anaconda2 installer
#export PATH="/home/nyejyh/anaconda2/bin:$PATH"
export PYTHONPATH=$PYTHONPATH:/usr/lib/python3
```
Running a search on what Python I have gave me:
```
nyejyh@BananaHarvester:~$ ls /usr/lib/python*
/usr/lib/python2.7
/usr/lib/python3
/usr/lib/python3.5
```

z3ntu, 2018-01-25:
I think adding `/usr/lib/python3/dist-packages` is what you want.

Fyro-Ing, 2018-01-25:
@z3ntu For my problem, this is the razer daemon log https://pastebin.com/8ErjGR08

we see "DBus called get_serial_list" whith polychromatic and razergenie
mouse connected : Bus 003 Device 004: ID 1532:003e Razer USA, Ltd

so i suppose that it's not a device bug

z3ntu, 2018-01-25:
@Fyro-Ing Run `sudo dkms install openrazer-driver/1.0.0`

[44 more replies]

PR #1695 (2022-04-07): Add support for Razer Naga Epic Chroma
Important to note for testers:
* The commit adds a few new functions which aren't supported in polychromatic or RazerGenie yet - you can test those via dbus directly, e.g. via d-feet (edit: it might work with RazerGenie from git master)

How to build from source: https://github.com/openrazer/openrazer/wiki/Building-a-package

Fixes #299
closed 2022-04-07 by z3ntu