openrazer/openrazer #300: Orochi: button remapping / bluetooth

daritter, 2017-06-06:
Hi,

thanks for the great work.

I was wondering if there's any plan to support remapping of mouse buttons. Specifically I would like (as a lefty) to get rid of the "change DPI" side buttons on my Orochi (1532:0048) and replace them with standard button 4/5 presses (disabling would be fine as well). And preferably over bluetooth :D 

I was hoping Synapse would write the button mapping into the mouse but no such luck, apparently only colors are important :/

terrycain, 2017-06-07:
@daritter I've been real busy as of late but it is being worked on

daritter, 2017-06-07:
Great, thanks for the info.

If there's anything you need (traffic capture, testing) just let me know.

Cheers, 

Martin

z3ntu, 2017-07-27:
@daritter Can you give me the product and model number and the full name of the Orochi (should be on a sticker on the bottom of the mouse)?

daritter, 2017-07-27:
It's inside the battery case: 

RAZER Orochi 8200
Model No: RZ01-0155
Product No: RZ01-01550100

closed 2018-02-28 by stale[bot]