openrazer/openrazer #301: Indicator not shown in Ubuntu Zesty

rumplstielz, 2017-06-07:
Hi, 
unfortunately the indicator icon in Ubunt 17.04 (Zesty) is not shown, resp. the service seems not to start correctly:
kurt@Yak:~$ ls -ltra /etc/init.d/^C
kurt@Yak:~$ ps -eaf | grep -i razer
kurt      2949  2597  0 14:37 ?        00:00:01 razer-service
kurt      5839  3449  0 14:49 pts/0    00:00:00 grep --color=auto -i razer
kurt@Yak:~$ date
Mit Jun  7 14:49:26 CEST 2017
kurt@Yak:~$ 

I even do not find the "razer-service" with systemctl. 

How can I achieve to have the service started correctly resp. how do you start the service?

br,
rumpl

terrycain, 2017-06-07:
@rumplstielz Indicator icon? Are you talking about polychromatic? If so, post on their repo. Otherwise logs should be in ~/.razer-service/logs

z3ntu, 2017-06-07:
1. What "indicator" icon do you mean?
2. The daemon is currently started by xdg autostart. (`/etc/xdg/autostart/razer-service.desktop`) but it'll be moved to systemd soon probably.

rumplstielz, 2017-06-07:
@terrycain yeah, you are right, stupid me ;-) Is the driver supposed to have any GUI then? 
@z3ntu I ment the indicator icon right at the indicator menu in Unity.

z3ntu, 2017-06-07:
No the driver doesn't include a GUI :)

terrycain, 2017-06-07:
@rumplstielz No the driver/daemon is purely an inteface between the hardware and a user interface

closed 2017-06-07 by z3ntu