openrazer/openrazer #302: Install Error

magnum93, 2017-06-09:
Hi there thanks for your programming on Razor drivers, I may have installed the wrong way first now in Linux Mint 18 I get the following error

sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
Reading package lists... Done
Building dependency tree       
Reading state information... Done
razer-doc is already the newest version (1.1.12-0ubuntu1).
razer-daemon is already the newest version (1.1.12-0ubuntu1).
The following NEW packages will be installed:
  python3-razer razer-kernel-modules-dkms
0 upgraded, 2 newly installed, 0 to remove and 10 not upgraded.
1 not fully installed or removed.
Need to get 0 B/58,2 kB of archives.
After this operation, 499 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
(Reading database ... 209588 files and directories currently installed.)
Preparing to unpack .../razer-kernel-modules-dkms_1.1.12-0ubuntu1_amd64.deb ...
Unpacking razer-kernel-modules-dkms (1.1.12-0ubuntu1) ...
dpkg: error processing archive /var/cache/apt/archives/razer-kernel-modules-dkms_1.1.12-0ubuntu1_amd64.deb (--unpack):
 trying to overwrite '/usr/src/razer_chroma_driver-1.0.0/driver/razerfirefly_driver.h', which is also in package razer-chroma-driver 1.0.0
Preparing to unpack .../python3-razer_1.1.12-0ubuntu1_all.deb ...
Unpacking python3-razer (1.1.12-0ubuntu1) ...
dpkg: error processing archive /var/cache/apt/archives/python3-razer_1.1.12-0ubuntu1_all.deb (--unpack):
 trying to overwrite '/usr/lib/python3/dist-packages/razer/__init__.py', which is also in package razer-chroma-driver 1.0.0
Errors were encountered while processing:
 /var/cache/apt/archives/razer-kernel-modules-dkms_1.1.12-0ubuntu1_amd64.deb
 /var/cache/apt/archives/python3-razer_1.1.12-0ubuntu1_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
gary@gary-All-Series ~ $

z3ntu, 2017-06-09:
```
sudo dkms remove razer_chroma_driver/1.0.0 --all
sudo apt remove python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
sudo apt --fix-broken install
sudo apt remove razer-chroma-driver
```

magnum93, 2017-06-09:
Hi there I type in these commands sudo apt --fix-broken install just returned a command menu, thanks for that now I get the following output from terminal when trying to re install.

Setting up razer-kernel-modules-dkms (1.1.12-0ubuntu1) ...
Removing old razer_chroma_driver-1.0.0 DKMS files...
Loading new razer_chroma_driver-1.0.0 DKMS files...
Error! DKMS tree already contains: razer_chroma_driver-1.0.0
You cannot add the same module/version combo more than once.
dpkg: error processing package razer-kernel-modules-dkms (--configure):
 subprocess installed post-installation script returned error exit status 3
dpkg: dependency problems prevent configuration of razer-daemon:
 razer-daemon depends on razer-kernel-modules-dkms; however:
  Package razer-kernel-modules-dkms is not configured yet.

dpkg: error processing package razer-daemon (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 razer-kernel-modules-dkms
 razer-daemon
E: Sub-process /usr/bin/dpkg returned an error code (1)

z3ntu, 2017-06-09:
```
sudo rm -r /var/lib/dkms/razer_chroma_driver
sudo apt remove python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
sudo apt remove razer-chroma-driver
```

magnum93, 2017-06-09:
Many thanks, that fixed it. where can we donate towards your work?

z3ntu, 2017-06-09:
@terrycain's PayPal is terry@terrys-home.co.uk
and mine is https://www.paypal.me/z3ntu

Thanks :)

closed 2017-06-09 by z3ntu