openrazer/openrazer #303: tartarus failed with exit code 1

RecursiveProgrammer, 2017-06-10:
I need a bit of direction. 

When I try to start polychromatic controller, I get a message saying that the device is not recognized.

I eventually saw this error:

Jun 10 16:11:43 Desk systemd-udevd[293]: Process 'razer_mount razermouse 0003:1532:0201.0001' failed with exit code 1.

Am I reporting this error in the correct place?

Here is more information:

$ dmesg

[  477.743996] usb 4-1.4: USB disconnect, device number 3
[  500.938421] usb 4-1.3: new full-speed USB device number 4 using ehci-pci
[  501.038692] input: Razer Razer Tartarus as /devices/pci0000:00/0000:00:1d.0/usb4/4-1/4-1.3/4-1.3:1.0/0003:1532:0201.0008/input/input20
[  501.095444] hid-generic 0003:1532:0201.0008: input,hidraw0: USB HID v1.11 Keyboard [Razer Razer Tartarus] on usb-0000:00:1d.0-1.3/input0
[  501.098273] input: Razer Razer Tartarus as /devices/pci0000:00/0000:00:1d.0/usb4/4-1/4-1.3/4-1.3:1.1/0003:1532:0201.0009/input/input21
[  501.151994] hid-generic 0003:1532:0201.0009: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer Tartarus] on usb-0000:00:1d.0-1.3/input1
[  501.153650] input: Razer Razer Tartarus as /devices/pci0000:00/0000:00:1d.0/usb4/4-1/4-1.3/4-1.3:1.2/0003:1532:0201.000A/input/input22
[  501.154000] hid-generic 0003:1532:0201.000A: input,hidraw2: USB HID v1.11 Mouse [Razer Razer Tartarus] on usb-0000:00:1d.0-1.3/input2

$ sudo journalctl -b | grep -i razer_mount

Jun 10 16:06:26 Desk razer_mount[2320]: Driver razermouse
Jun 10 16:06:26 Desk razer_mount[2324]: Driver razermouse
Jun 10 16:06:26 Desk razer_mount[2322]: Driver razermouse
Jun 10 16:06:26 Desk razer_mount[2325]: Device_ID 0003:1532:0201.0009
Jun 10 16:06:26 Desk razer_mount[2326]: Device_ID 0003:1532:0201.000A
Jun 10 16:06:26 Desk razer_mount[2327]: Device_ID 0003:1532:0201.0008
Jun 10 16:06:26 Desk razer_mount[2328]: Modprobing razermouse
Jun 10 16:06:26 Desk razer_mount[2330]: Modprobing razermouse
Jun 10 16:06:26 Desk systemd-udevd[2313]: Process 'razer_mount razermouse 0003:1532:0201.0009' failed with exit code 1.
Jun 10 16:06:26 Desk razer_mount[2332]: Modprobing razermouse
Jun 10 16:06:26 Desk systemd-udevd[2314]: Process 'razer_mount razermouse 0003:1532:0201.000A' failed with exit code 1.
Jun 10 16:06:26 Desk systemd-udevd[2298]: Process 'razer_mount razermouse 0003:1532:0201.0008' failed with exit code 1.

$ uname -a
Linux Desk 4.11.3-1-ARCH #1 SMP PREEMPT Sun May 28 10:40:17 CEST 2017 x86_64 GNU/Linux

Cheers!

z3ntu, 2017-06-10:
more or less known issue with the udev rules... we are not sure what the problem is...

z3ntu, 2017-06-10:
> 0211 (BlackWidow Chroma Overwatch) and 003e (Naga Chroma) are known to have this issue...

This device 0201 (Tartarus) now too... 

Thank you for the logs btw, that made things way easier so I don't have to remotely debug to find out it's this issue ^^

z3ntu, 2017-06-10:
Also for real support of the device (it's not supported yet but gets matched by udev due to that issue) you can look at #235 :) 

(and sorry for the three messages)

z3ntu, 2017-06-11:
Please give me the output of `udevadm info -a /sys/bus/hid/devices/0003:1532:0201.0*`

closed 2017-06-11 by z3ntu

RecursiveProgrammer, 2017-06-11:
* Here is the relevant part:
```
$ udevadm info -a /sys/bus/hid/devices/0003:1532:0201.0* | grep -i "Razer Tartarus" -A 6 -B 25
  looking at parent device '/devices/pci0000:00/0000:00:1d.0/usb4/4-1/4-1.3':
    KERNELS=="4-1.3"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{authorized}=="1"
    ATTRS{avoid_reset_quirk}=="0"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{bDeviceClass}=="00"
    ATTRS{bDeviceProtocol}=="00"
    ATTRS{bDeviceSubClass}=="00"
    ATTRS{bMaxPacketSize0}=="64"
    ATTRS{bMaxPower}=="500mA"
    ATTRS{bNumConfigurations}=="1"
    ATTRS{bNumInterfaces}==" 3"
    ATTRS{bcdDevice}=="0200"
    ATTRS{bmAttributes}=="a0"
    ATTRS{busnum}=="4"
    ATTRS{configuration}==""
    ATTRS{devnum}=="3"
    ATTRS{devpath}=="1.3"
    ATTRS{idProduct}=="0201"
    ATTRS{idVendor}=="1532"
    ATTRS{ltm_capable}=="no"
    ATTRS{manufacturer}=="Razer"
    ATTRS{maxchild}=="0"
    ATTRS{product}=="Razer Tartarus"
    ATTRS{quirks}=="0x0"
    ATTRS{removable}=="removable"
    ATTRS{speed}=="12"
    ATTRS{urbnum}=="28"
    ATTRS{version}==" 2.00"
```

* Here is the full output:

```
$ udevadm info -a /sys/bus/hid/devices/0003:1532:0201.0*

Udevadm info starts with the device specified by the devpath and then
walks up the chain of parent devices. It prints for every device
found, all possible attributes in the udev rules key format.
A rule to match, can be composed by the attributes of the device
and the attributes from one single parent device.

  looking at device '/devices/pci0000:00/0000:00:1d.0/usb4/4-1/4-1.3/4-1.3:1.0/0003:1532:0201.0001':
    KERNEL=="0003:1532:0201.0001"
    SUBSYSTEM=="hid"
    DRIVER==""
    ATTR{report_descriptor}==""

  looking at parent device '/devices/pci0000:00/0000:00:1d.0/usb4/4-1/4-1.3/4-1.3:1.0':
    KERNELS=="4-1.3:1.0"
    SUBSYSTEMS=="usb"
    DRIVERS=="usbhid"
    ATTRS{authorized}=="1"
    ATTRS{bAlternateSetting}==" 0"
    ATTRS{bInterfaceClass}=="03"
    ATTRS{bInterfaceNumber}=="00"
    ATTRS{bInterfaceProtocol}=="01"
    ATTRS{bInterfaceSubClass}=="01"
    ATTRS{bNumEndpoints}=="01"
    ATTRS{supports_autosuspend}=="1"

  looking at parent device '/devices/pci0000:00/0000:00:1d.0/usb4/4-1/4-1.3':
    KERNELS=="4-1.3"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{authorized}=="1"
    ATTRS{avoid_reset_quirk}=="0"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{bDeviceClass}=="00"
    ATTRS{bDeviceProtocol}=="00"
    ATTRS{bDeviceSubClass}=="00"
    ATTRS{bMaxPacketSize0}=="64"
    ATTRS{bMaxPower}=="500mA"
    ATTRS{bNumConfigurations}=="1"
    ATTRS{bNumInterfaces}==" 3"
    ATTRS{bcdDevice}=="0200"
    ATTRS{bmAttributes}=="a0"
    ATTRS{busnum}=="4"
    ATTRS{configuration}==""
    ATTRS{devnum}=="3"
    ATTRS{devpath}=="1.3"
    ATTRS{idProduct}=="0201"
    ATTRS{idVendor}=="1532"
    ATTRS{ltm_capable}=="no"
    ATTRS{manufacturer}=="Razer"
    ATTRS{maxchild}=="0"
    ATTRS{product}=="Razer Tartarus"
    ATTRS{quirks}=="0x0"
    ATTRS{removable}=="removable"
    ATTRS{speed}=="12"
    ATTRS{urbnum}=="28"
    ATTRS{version}==" 2.00"

  looking at parent device '/devices/pci0000:00/0000:00:1d.0/usb4/4-1':
    KERNELS=="4-1"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{authorized}=="1"
    ATTRS{avoid_reset_quirk}=="0"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{bDeviceClass}=="09"
    ATTRS{bDeviceProtocol}=="01"
    ATTRS{bDeviceSubClass}=="00"
    ATTRS{bMaxPacketSize0}=="64"
    ATTRS{bMaxPower}=="0mA"
    ATTRS{bNumConfigurations}=="1"
    ATTRS{bNumInterfaces}==" 1"
    ATTRS{bcdDevice}=="0000"
    ATTRS{bmAttributes}=="e0"
    ATTRS{busnum}=="4"
    ATTRS{configuration}==""
    ATTRS{devnum}=="2"
    ATTRS{devpath}=="1"
    ATTRS{idProduct}=="0024"
    ATTRS{idVendor}=="8087"
    ATTRS{ltm_capable}=="no"
    ATTRS{maxchild}=="6"
    ATTRS{quirks}=="0x0"
    ATTRS{removable}=="fixed"
    ATTRS{speed}=="480"
    ATTRS{urbnum}=="38"
    ATTRS{version}==" 2.00"

  looking at parent device '/devices/pci0000:00/0000:00:1d.0/usb4':
    KERNELS=="usb4"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{authorized}=="1"
    ATTRS{authorized_default}=="1"
    ATTRS{avoid_reset_quirk}=="0"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{bDeviceClass}=="09"
    ATTRS{bDeviceProtocol}=="00"
    ATTRS{bDeviceSubClass}=="00"
    ATTRS{bMaxPacketSize0}=="64"
    ATTRS{bMaxPower}=="0mA"
    ATTRS{bNumConfigurations}=="1"
    ATTRS{bNumInterfaces}==" 1"
    ATTRS{bcdDevice}=="0411"
    ATTRS{bmAttributes}=="e0"
    ATTRS{busnum}=="4"
    ATTRS{configuration}==""
    ATTRS{devnum}=="1"
    ATTRS{devpath}=="0"
    ATTRS{idProduct}=="0002"
    ATTRS{idVendor}=="1d6b"
    ATTRS{interface_authorized_default}=="1"
    ATTRS{ltm_capable}=="no"
    ATTRS{manufacturer}=="Linux 4.11.3-1-ARCH ehci_hcd"
    ATTRS{maxchild}=="2"
    ATTRS{product}=="EHCI Host Controller"
    ATTRS{quirks}=="0x0"
    ATTRS{removable}=="unknown"
    ATTRS{serial}=="0000:00:1d.0"
    ATTRS{speed}=="480"
    ATTRS{urbnum}=="24"
    ATTRS{version}==" 2.00"

  looking at parent device '/devices/pci0000:00/0000:00:1d.0':
    KERNELS=="0000:00:1d.0"
    SUBSYSTEMS=="pci"
    DRIVERS=="ehci-pci"
    ATTRS{broken_parity_status}=="0"
    ATTRS{class}=="0x0c0320"
    ATTRS{companion}==""
    ATTRS{consistent_dma_mask_bits}=="32"
    ATTRS{d3cold_allowed}=="1"
    ATTRS{device}=="0x1c26"
    ATTRS{dma_mask_bits}=="32"
    ATTRS{driver_override}=="(null)"
    ATTRS{enable}=="1"
    ATTRS{irq}=="23"
    ATTRS{local_cpulist}=="0-3"
    ATTRS{local_cpus}=="f"
    ATTRS{msi_bus}=="1"
    ATTRS{numa_node}=="-1"
    ATTRS{revision}=="0x05"
    ATTRS{subsystem_device}=="0x844d"
    ATTRS{subsystem_vendor}=="0x1043"
    ATTRS{uframe_periodic_max}=="100"
    ATTRS{vendor}=="0x8086"

  looking at parent device '/devices/pci0000:00':
    KERNELS=="pci0000:00"
    SUBSYSTEMS==""
    DRIVERS==""

```

z3ntu, 2017-06-11:
It's actually already fixed (in an unreleased version). Thanks :)

RecursiveProgrammer, 2017-06-11:
Sweet, is there an ETA on release date, by any chance!

Thanks for the quick feed back, by the way.

z3ntu, 2017-06-11:
FYI the Razer Tartarus isn't supported yet, but that issue you were experiencing with udev is fixed. See #235 for the thread. 
Now (after that fix) udev won't try to bind the device to a driver but instead ignore it (until it's supported).

terrycain, 2017-06-11:
c13cc1a0225e3002114810a16f8bc9f212de315d (v1.1.13) Solves this.

RecursiveProgrammer, 2017-06-11:
Aha, ok, I think I am up to speed, then.

In that case, [the devices area](https://terrycain.github.io/razer-drivers/) probably needs to have the razar tartarus removed from the supported list.

Is there anything I can do to help get it supported?

Cheerio!

terrycain, 2017-06-11:
@RecursiveProgrammer The tartarus chroma is supported, just not the regular tartarus.

terrycain, 2017-06-11:
go to #235

RecursiveProgrammer, 2017-06-11:
Kk, I will head over there. I did not realize that there was even a difference. Ha! Thanks for the patience!

terrycain, 2017-06-11:
@RecursiveProgrammer np

commit 5c13a3bc (2017-06-11): Fix udev matching

ATTRS{idProduct} not only matches the current device but also parent
devices (like internal usb hubs etc). That sometimes resulted in the
device trying to be bound to a wrong driver. Fix this by explicitly
matching also the idVendor.

See https://github.com/systemd/systemd/issues/6112

Fixes #303