openrazer/openrazer #309: Razer Blade Stealth (Late 2016)

kprinssu, 2017-06-14:
I am trying to run the razer-daemon using `razer-daemon -F`, but it always provides the following error:

See the following gist: https://gist.github.com/kprinssu/cf176e688fa529df4cdd2445d575cfaa

z3ntu, 2017-06-14:
Please post the output of `cat /sys/bus/hid/drivers/razerkbd/0*/device_serial`

kprinssu, 2017-06-14:
It's empty, as in:

`
$ cat /sys/bus/hid/drivers/razerkbd/0*/device_serial
 
`

z3ntu, 2017-06-14:
Could you run `sudo dmidecode -t 1` and post the output here? Either Razer f**** something up as in #192 (where they didn't change the serial so it's "Default string"...) or the driver is doing something wrong.

kprinssu, 2017-06-14:
It's still blank under the serial number.

`# dmidecode 3.0
Getting SMBIOS data from sysfs.
SMBIOS 3.0.0 present.

Handle 0x0001, DMI type 1, 27 bytes
System Information
	Manufacturer: Razer
	Product Name: Blade Stealth
	Version: 2.04
	Serial Number:  
	UUID: Not Settable
	Wake-up Type: Power Switch
	SKU Number:  
	Family: 1A586752

`

z3ntu, 2017-06-14:
Thanks... @terrycain should be able to write a workaround for
* empty serial
* "Default string" serial

kprinssu, 2017-06-14:
Fun times when Razer messes up. Thanks for the help.

terrycain, 2017-06-15:
Possibly fixed in 8b8999a, have a look after the next release (v1.1.14)

kprinssu, 2017-06-16:
I manually installed the latest changes, but it seems that my keyboard is not recognised on Polychromatic.

terrycain, 2017-06-16:
did you reboot?

terrycain, 2017-06-16:
and does `razer-daemon -F` still give errors?

z3ntu, 2017-06-16:
Actually running `cat /sys/bus/hid/drivers/razerkbd/0*/device_serial` should show it 😃

terrycain, 2017-06-16:
^^ @z3ntu nah, patched it in daemon for now

z3ntu, 2017-06-16:
Ah okay - good enough

kprinssu, 2017-06-16:
I did reboot and running `razer-daemon -F` doesn't an error. In fact, the serial issue seems to be fixed, but Polychroamtic says that it doesn't recognise the keyboard. I should also mention that I am on Ubuntu Gnome 17.04.

terrycain, 2017-06-16:
Ahh, I think thats now more an issue with polychromatic than it is with the daemon. @lah7 ^^

terrycain, 2017-06-16:
Should now try raising an issue on the polychromatic repo

z3ntu, 2017-06-16:
What does `python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"` give?

kprinssu, 2017-06-16:
That command outputs an error: https://gist.github.com/kprinssu/718621c63e442cd9528ae1ec4c2fa417

I think it's an underlying issue with the razer-drivers. If you think it's an issue with Polychromatic, I'll go bug the devs over there :).

terrycain, 2017-06-16:
Is the daemon running?

kprinssu, 2017-06-16:
Not when I ran that command, I'll post an updated log.

terrycain, 2017-06-16:
yeah that command tries to talk to the daemon

kprinssu, 2017-06-16:
The command that z3ntu posted just returns an empty array: `[]`

terrycain, 2017-06-16:
Hmm, whats the output of lsusb?

kprinssu, 2017-06-16:
Here is the output of lsusb:

`Bus 004 Device 004: ID 05e3:0617 Genesys Logic, Inc. 
Bus 004 Device 003: ID 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet
Bus 004 Device 002: ID 05e3:0617 Genesys Logic, Inc. 
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 003 Device 004: ID 1532:0215 Razer USA, Ltd 
Bus 003 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 04f3:223c Elan Microelectronics Corp. 
Bus 001 Device 004: ID 1532:0220 Razer USA, Ltd 
Bus 001 Device 003: ID 0bda:579f Realtek Semiconductor Corp. 
Bus 001 Device 002: ID 8087:0a2b Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

terrycain, 2017-06-16:
and whats the output from the daemon?

kprinssu, 2017-06-16:
This is what I get from doing `razer-daemon -F`:

``
$ razer-daemon -F
2017-06-16 09:41:07 | razer                          | INFO     | Initialising Daemon (v1.1.13). Pid: 2597
2017-06-16 09:41:07 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2012 --------------------- (1532:010D)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderChroma ------------------- (1532:0043)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderElite -------------------- (1532:005C)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerImperiator ------------------------- (1532:002F)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wired --------------------- (1532:0024)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wireless ------------------ (1532:0025)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2017-06-16 09:41:07 | razer                          | DEBUG    | Loaded device specification: RazerNaga2014 --------------------------- (1532:0040)
[... cut ...]
2017-06-16 09:41:07 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2017-06-16 09:41:07 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2017-06-16 09:41:07 | razer                          | INFO     | Adding razer.daemon.version method to DBus
2017-06-16 09:41:07 | razer                          | INFO     | Adding razer.daemon.stop method to DBus
2017-06-16 09:41:07 | razer                          | INFO     | Serving DBus
``

kprinssu, 2017-06-16:
Here is the output from the latest changes:

https://gist.github.com/kprinssu/c83ecfba63a4571caf8a162703c98a19

terrycain, 2017-06-16:
contents of /lib/udev/rules.d/99-razer.rules

kprinssu, 2017-06-16:
The file doesn't exist.

terrycain, 2017-06-16:
ok that doesnt help. Does  `sudo modprobe razerkbd` work?

kprinssu, 2017-06-16:
Nope..

terrycain, 2017-06-16:
Wait, have you installed the project using the proper packages?

kprinssu, 2017-06-16:
I installed it via the instructions here:
https://terrycain.github.io/razer-drivers/#other

terrycain, 2017-06-16:
What distro are you running?

kprinssu, 2017-06-16:
Ubuntu Gnome 17.04

terrycain, 2017-06-16:
https://terrycain.github.io/razer-drivers/#ubuntu

kprinssu, 2017-06-16:
I installed the package via apt-get but running `sudo modprobe razerkbd` returns nothing. 
Running `razer-daemon -F` shows that it detects my keyboard, but fails with the first error since I am not running the latest changes.

terrycain, 2017-06-16:
yeah so now patch, and reboot (modprobe only outputs on error btw)

kprinssu, 2017-06-16:
Can you clarify on what you mean by patching?

terrycain, 2017-06-16:
Take https://github.com/terrycain/razer-drivers/blob/master/daemon/razer_daemon/hardware/device_base.py
and replace the one in /usr/lib/python... somewhere

kprinssu, 2017-06-16:
Replacing the file gives the following error from Python (when running the daemon):
https://gist.github.com/kprinssu/cd08b057a2cda71cce113b02dda4d498

terrycain, 2017-06-16:
You replaced /usr/lib/python3/dist-packages/razer_daemon/hardware/device_base.py
right?

terrycain, 2017-06-16:
hmm, ok, remove the razer-daemon package, and i'll release a proper package again soon

terrycain, 2017-07-01:
Version 1.1.14 is out on launchpad, non Ubuntu packages will be out shortly.

closed 2017-07-01 by terrycain