openrazer/openrazer #311: Bringing kernel drivers upstream

Lyude, 2017-06-17:
Hi! I am a developer from Red Hat who primarily works on driver support for Linux. I recently actually purchased a Razer Blade Stealth for myself and stumbled upon your project. It's really nice to see that there's been a good bit of work reverse engineering these devices done here. However, it would be really nice to eventually see these machines actually supported out of the box on Linux distributions, and doing that would require that you guys eventually merge your drivers upstream with the kernel itself. Do you guys happen to have any plans for doing that in the future?

z3ntu, 2017-06-17:
You are probably interested in #29

terrycain, 2017-06-25:
Hey, Yes and No, #29 is correct to some degree. 

The majority of the keyboard driver part can be done in userspace as the endpoint that deals with lighting does not transmit key events.

From what I understand  you'll need to unbind a USB endpoint from whatever kernel driver is using it before you can interact with it from userspace. That wont work for cases like the mice which use the same endpoint for transmitting movement data as well as receiving packets to control lighting.

It would be nice to get what we have into the kernel but it would require some big changes which is not too bad, its mostly once its in the kernel its going to become increasingly difficult to modify, which in its self isn't a bad thing either.

Currently the aim is just to support as many devices as possible once that's slowed down a bit I don't see any issue with it going into the kernel. Were using dkms at the moment to deal with all the kernel stuff which seems to be doing its job for the most part (except when secure boot makes the odd appearance, which would be solved if it were in-tree).

whompy, 2017-07-04:
I'm with lyude on this one. I'd love it in the kernel as it's the only dkms module on my system. Having the kernel driver in tree would be most excellent.

That said, I understand that it's best to get something in a good, solid state before stabilizing the api in upstream. As a follower of the project, I'd be interested to know if there are specific milestones you'd like before upstreaming. 

Again, I understand the logic and what you're doing, just curious if there's a long-term plan.

ukyg9e5r6k7gubiekd6, 2017-12-20:
I too am interested in fixing the kernel-mode parts and upstreaming them (see #428, which partly duplicates this).
I'm intending to just scratch my itch, support for control over the LEDs on Razer mice (I don't have any other Razer devices) shortly, when I make the time. Especially as I understand from terrycain's explanation that that has to be done in-kernel, due to multiplexing lighting and movement data on the same endpoints.

terrycain, 2017-12-20:
> due to multiplexing lighting and movement data on the same endpoints.

Not exactly, take a look at the `hid` branch were slowly moving stuff to do as much of the work in userspace as possible. For the mice and keyboards, all that will end up in the kernel I hope is some messing around/fixing of key/button events

iKarith, 2021-02-07:
This issue's been dormant for about three years now, and it doesn't look like the hid branch has seen any work to pull in changes from master in about that long. Is the notion of pushing as much as possible into userspace and the approach taken by the hid branch being more or less abandoned?

What does this mean for these drivers becoming mainstream? "Tainted kernels", etc.

z3ntu, 2021-02-07:
I would love for most of the functionality to be implemented in user space (see my experimental project https://github.com/z3ntu/razer_test) but it's not really possible to change the implementation for 100+ already supported devices and keep them working without any regressions.

FFY00, 2021-02-07:
If you know how the devices behave, you can emulate the devices using UHID, run the tests against the current driver and make sure everything is working as expected, and then you can port the whole thing to userspace and make sure everything still works as expected. You'd still want to have some people testing this on real hardware at some point, but this approach would allow you to port the drivers without having the devices physically.

z3ntu, 2021-02-07:
If you could develop this tool then that would be great, but I don't have too much time / motivation to work on such a thing (at the moment).

FFY00, 2021-02-07:
We have a PoC at https://github.com/libratbag/ratbag-emu, but I will be redesigning everything from scratch on top of another backend library. You can make use of that work later if you want. I don't have any razer devices, though I'd be interested in bringing support to libratbag, and I have limited time so I probably won't be contributing to this upstream.

FFY00, 2025-12-11:
FWIW, I have the same opinion.

Lyude, 2026-06-30:
(I got the mouse! So this is officially on my to-do list when I've got the time!)

Lyude, 2026-09-13:
JFYI - what sort of proof for avoiding regressions would actually be needed here, btw? I'm not terribly confident that we could actually go through every device one-by-one to confirm there are no regressions, but I also don't really think it would be necessary if we can just get enough devices to cover the majority of the functions that the driver already supports and then just compare the actual HID traces going across the line between the in-kernel version of openrazer and the (future) userspace version of openrazer. I think this is true as well because `/dev/hidraw*` is really just a character-device front-end to the same kernel interface that the kernel module is already using to communicate with mice/keyboards/anything over hid.

If that's something we'd be nervous about doing, would anyone be against me seeing if I could just make it configurable whether it uses the kernel module or not?

z3ntu, 2026-11-12:
At least a sort of theoretical "proof" that the different code paths that are present in the driver are then also taken - the equivalent through hidraw. And some validation on tens of devices to make sure this actually works, the rest is then hopefully just working as-is.