openrazer/openrazer #312: Cannot see Razer Core on Arch

BryceBeagle, 2017-06-17:
I have the latest version of the `razer-driver-meta` package from the AUR.

I looked through the troubleshooting section, but nothing there helped.

I tried following [this issue thread](https://github.com/terrycain/razer-drivers/issues/244), but I get stuck [here](https://github.com/terrycain/razer-drivers/issues/244#issuecomment-290941735), with the `Failed to start razer-service.service: Unit razer-service.service not found.` error. Neither `systemctl --user daemon-reload` nor her solution of changing the owner of the service worked. 

Looking at `/usr/lib/systemd/user/razer-service.service` I see `ExecStart=/usr/bin/razer-service` but my `/usr/bin` directory only has a `razer-daemon`. I'm not sure if this could be a problem (I don't have much experience with systemd services).

z3ntu, 2017-06-17:
We changed the executable name in the latest release or so. You have to change razer-service with razer-daemon in the unit file.

BryceBeagle, 2017-06-17:
I get the same error trying to start the service: 

> ┌─[griefcake][/usr/lib/systemd/user]
> └──▪ systemctl start razer-service.service 
> Failed to start razer-service.service: Unit razer-service.service not found.

However, `systemctl status` now has a more helpful message

> ┌─[griefcake][/usr/lib/systemd/user]
> └──▪ systemctl status razer-service.service 
> ● razer-service.service
>    Loaded: not-found (Reason: No such file or directory)
>    Active: failed (Result: exit-code) since Sat 2017-06-17 01:20:07 MST; 8min ago
>  Main PID: 22023 (code=exited, status=1/FAILURE)
> 
> Jun 17 01:20:07 griefcake systemd[1]: Started Razer Daemon (User).
> Jun 17 01:20:07 griefcake razer-daemon[22023]: Traceback (most recent call last):
> Jun 17 01:20:07 griefcake razer-daemon[22023]:   File "/usr/bin/razer-daemon", line 16, in <module>
> Jun 17 01:20:07 griefcake razer-daemon[22023]:     BASE_PATH = os.path.join(os.path.abspath(os.environ['HOME']), '.razer-service')
> Jun 17 01:20:07 griefcake razer-daemon[22023]:   File "/usr/lib/python3.6/os.py", line 669, in __getitem__
> Jun 17 01:20:07 griefcake razer-daemon[22023]:     raise KeyError(key) from None
> Jun 17 01:20:07 griefcake razer-daemon[22023]: KeyError: 'HOME'
> Jun 17 01:20:07 griefcake systemd[1]: razer-service.service: Main process exited, code=exited, status=1/FAILURE
> Jun 17 01:20:07 griefcake systemd[1]: razer-service.service: Unit entered failed state.
> Jun 17 01:20:07 griefcake systemd[1]: razer-service.service: Failed with result 'exit-code'.

z3ntu, 2017-06-17:
You have to use `systemctl --user start razer-service`

BryceBeagle, 2017-06-17:
No errors when starting the service this time, but now running `status` gives

> ┌─[griefcake][/usr/lib/systemd/user]
> └──▪ systemctl --user status razer-service.service 
> ● razer-service.service - Razer Daemon (User)
>    Loaded: loaded (/usr/lib/systemd/user/razer-service.service; disabled; vendor preset: enabled)
>    Active: failed (Result: exit-code) since Sat 2017-06-17 01:46:49 MST; 19s ago
>   Process: 1599 ExecStart=/usr/bin/razer-daemon --foreground --verbose --run-dir %h/.razer-service/data --log-dir %h/.razer-ser
>  Main PID: 1599 (code=exited, status=255)
> 
> Jun 17 01:46:49 griefcake systemd[447]: Started Razer Daemon (User).
> Jun 17 01:46:49 griefcake razer-daemon[1599]: Daemon already exists. Please stop that one.
> Jun 17 01:46:49 griefcake systemd[447]: razer-service.service: Main process exited, code=exited, status=255/n/a
> Jun 17 01:46:49 griefcake systemd[447]: razer-service.service: Unit entered failed state.
> Jun 17 01:46:49 griefcake systemd[447]: razer-service.service: Failed with result 'exit-code'.

This was immediately after running `start` and I had previously checked with `status` to verify that it wasn't already.

z3ntu, 2017-06-17:
What Desktop Environment / Window Manager are you using?

BryceBeagle, 2017-06-17:
I'm using budgie.
Thanks for the help by the way

z3ntu, 2017-06-17:
As Budgie is a fork of GNOME 3, xdg-autostart (the current "standard" way of starting the daemon) is used. You don't need to do anything with systemd user units.

* Please post the output of `sudo dkms status razer_chroma_driver/1.0.0`

BryceBeagle, 2017-06-17:
`razer_chroma_driver, 1.0.0, 4.11.5-1-ARCH, x86_64: installed`

z3ntu, 2017-06-17:
Okay good,
`python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"` please

z3ntu, 2017-06-17:
`lsmod | grep razercore; sudo modprobe razercore; modprobe razercore` please

z3ntu, 2017-06-17:
Please see my edit in the previous message :)

BryceBeagle, 2017-06-17:
` `

BryceBeagle, 2017-06-17:
`
razercore              32768  0
hid                   110592  5 i2c_hid,hid_generic,usbhid,hid_multitouch,razercore
usbcore               208896  7 uvcvideo,usbhid,xhci_pci,brcmfmac,btusb,razercore,xhci_hcd
`

z3ntu, 2017-06-17:
`lsusb | grep 1532` and `sudo journalctl -b | grep -i razer_mount` please

BryceBeagle, 2017-06-17:
` `
and 
` `

z3ntu, 2017-06-17:
Are you sure your Razer Core is connected...?

BryceBeagle, 2017-06-17:
Yep, and if I switch to Windows it works

z3ntu, 2017-06-17:
Okay `lsusb` please

BryceBeagle, 2017-06-17:
> ┌─[griefcake][~]
> └──▪ lsusb
> Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
> Bus 001 Device 003: ID 0a5c:6410 Broadcom Corp. BCM20703A1 Bluetooth 4.1 + LE
> Bus 001 Device 004: ID 0c45:6713 Microdia 
> Bus 001 Device 002: ID 062a:4106 Creative Labs 
> Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

z3ntu, 2017-06-17:
Apparently you need to do special things with the Core to get it working. I quote from the original support thread (#169)

> I modded my early 2016 RBS BIOS and got the Core working. It presents a USB device for the lighting controller. I will try to capture some packets this weekend and will try to work on a driver based on the Firefly/Chroma Mug code. I expect it to be similar to the Firefly with 9 LEDs (inside LED strip is all one zone, then 8 individually controlled ones on the front). It definitely supports custom effects through the official SDK. I am not aware of any other funtionality provided by the Core's USB device, as the USB hub is just a hub, the Ethernet port is a separate USB 3.0 ASIX gigabit controller, and the GPU is presented on the PCIe bus.

BryceBeagle, 2017-06-17:
Unfortunately I'm trying to do this on a non-Razer laptop (Dell XPS). Any idea how I should proceed?

z3ntu, 2017-06-17:
Does the Core itself work?

BryceBeagle, 2017-06-17:
The Core works on Windows. I'm honestly not sure how I would verify any functionality on Linux.

z3ntu, 2017-06-17:
Plug in a monitor to the core and see if it works?^^ But I'm not really familiar with what the Razer Core can do and can't do.

And apparently according to https://insider.razerzone.com/index.php?threads/razer-blade-stealth-and-core-2017-without-bios-mods-on-debian.21429/ you don't even need BIOS mods on the Razer Blade Stealth and it should still show the USB interface... @CalcProgrammer1 can you clarify a bit how that stuff works?

BryceBeagle, 2017-06-17:
Monitors have been plugged in the whole time. They're definitely not working :P
xrandr sees nothing but my built-in display.

Thanks again for the help

z3ntu, 2017-06-17:
I think you should first get the Core itself running and then afterwards you can play around with this driver to change the lighting 😉  I'm gonna close this for now, but if you get an usb interface (`lsusb | grep 1532`), feel free to ask here again :)

closed 2017-06-17 by z3ntu

BryceBeagle, 2017-06-17:
Wait.... I was under the impression that this was a set of general purpose drivers and not just for lighting...

z3ntu, 2017-06-17:
No, we just send usb stuff to the device, sorry. So if there's no usb interface, we can't do anything...

BryceBeagle, 2017-06-17:
Darn... the Readme description might wanna mention that haha

> Razer Drivers
> 
> A collection of Linux drivers for the Razer devices, providing kernel drivers, DBus services and python bindings to interact with the DBus interface.

With a name and description like that...

z3ntu, 2017-06-17:
Yeah... It's pretty obvious what the driver does for normal keyboards/mice but not really clear for the Core... eg for Mice is lets you change lighting, polling rate, dpi, etc

terrycain, 2017-06-17:
Yeah the driver just controls the cores lighting but the core needs to be seen by the system before we can actually do anything.

lezed1, 2017-06-17:
@BryceBeagle - I use a Razer Blade Stealth laptop with a Razer Core, so I might be able to help a bit. Did you disable Thunderbolt authentication in the BIOS settings? The Core did not appear as a USB device/GPU unless I did that.

BryceBeagle, 2017-06-17:
I updated the BIOS and disabled the TB authentication and that seem to do the trick. Thanks!

z3ntu, 2017-06-17:
Does everything work now? 😃

BryceBeagle, 2017-06-17:
Well I can see the GPU and the other devices that are part of the Core, but I'm still working on trying to get the graphics card to actually be used. glxgears still shows me as using my internal 960M.

z3ntu, 2017-06-17:
And does our driver work? 
`python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"` should give you RazerCore

BryceBeagle, 2017-06-17:
> ─[griefcake][~]
> └──▪ python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
> Traceback (most recent call last):
>   File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 175, in activate_name_owner
>     return self.get_name_owner(bus_name)
>   File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 361, in get_name_owner
>     's', (bus_name,), **keywords)
>   File "/usr/lib/python3.6/site-packages/dbus/connection.py", line 651, in call_blocking
>     message, timeout)
> dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.razer': no such name
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/usr/lib/python3.6/site-packages/razer/client/__init__.py", line 20, in __init__
>     self._dbus = session_bus.get_object("org.razer", "/org/razer")
>   File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 241, in get_object
>     follow_name_owner_changes=follow_name_owner_changes)
>   File "/usr/lib/python3.6/site-packages/dbus/proxies.py", line 248, in __init__
>     self._named_service = conn.activate_name_owner(bus_name)
>   File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 180, in activate_name_owner
>     self.start_service_by_name(bus_name)
>   File "/usr/lib/python3.6/site-packages/dbus/bus.py", line 278, in start_service_by_name
>     'su', (bus_name, flags)))
>   File "/usr/lib/python3.6/site-packages/dbus/connection.py", line 651, in call_blocking
>     message, timeout)
> dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.razer was not provided by any .service files
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "<string>", line 1, in <module>
>   File "/usr/lib/python3.6/site-packages/razer/client/__init__.py", line 22, in __init__
>     raise DaemonNotFound("Could not connect to daemon")
> razer.client.DaemonNotFound: Could not connect to daemon
>

lezed1, 2017-06-17:
It looks like you aren't running the daemon. Try `systemctl --user start razer-service` and then run that command again.

BryceBeagle, 2017-06-17:
Ok so starting the service appears to initially work. I can see the RazerCore in the `systemctl status`:
```
─[griefcake][/usr/lib/systemd/user]
└──▪ systemctl --user status razer-service.service 
● razer-service.service - Razer Daemon (User)
   Loaded: loaded (/usr/lib/systemd/user/razer-service.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2017-06-17 14:29:16 MST; 14s ago
 Main PID: 5145 (razer-daemon)
   CGroup: /user.slice/user-1000.slice/user@1000.service/razer-service.service
           └─5145 razer-daemon

Jun 17 14:29:17 griefcake razer-daemon[5145]: 2017-06-17 14:29:17 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaver -------------------------- (1532:0113)
Jun 17 14:29:17 griefcake razer-daemon[5145]: 2017-06-17 14:29:17 | razer                          | DEBUG    | Loaded device specification: RazerOrnata ----------------------------- (1532:021F)
Jun 17 14:29:17 griefcake razer-daemon[5145]: 2017-06-17 14:29:17 | razer                          | DEBUG    | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
Jun 17 14:29:17 griefcake razer-daemon[5145]: 2017-06-17 14:29:17 | razer                          | DEBUG    | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
Jun 17 14:29:17 griefcake razer-daemon[5145]: 2017-06-17 14:29:17 | razer                          | DEBUG    | Loaded device specification: RazerOuroboros -------------------------- (1532:0032)
Jun 17 14:29:17 griefcake razer-daemon[5145]: 2017-06-17 14:29:17 | razer                          | DEBUG    | Loaded device specification: RazerTaipan ----------------------------- (1532:0034)
Jun 17 14:29:17 griefcake razer-daemon[5145]: 2017-06-17 14:29:17 | razer                          | DEBUG    | Loaded device specification: RazerTartarus --------------------------- (1532:0208)
Jun 17 14:29:17 griefcake razer-daemon[5145]: 2017-06-17 14:29:17 | razer                          | INFO     | Found device.0: 0003:1532:0215.0002
Jun 17 14:29:17 griefcake razer-daemon[5145]: 2017-06-17 14:29:17 | razer.device0                  | INFO     | Initialising device.0 RazerCore
Jun 17 14:29:27 griefcake razer-daemon[5145]: 2017-06-17 14:29:27 | razer.device0                  | DEBUG    | getting serial:  count:1
```

However, `python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"` spins its wheels for a while with no output and then errors out:
```
┌─[griefcake][/usr/lib/systemd/user]
└──▪ python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
ERROR:dbus.proxies:Introspect error on :1.128:/org/razer: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.6/site-packages/razer/client/__init__.py", line 30, in __init__
    self._device_serials = self._dbus_devices.getDevices()
  File "/usr/lib/python3.6/site-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python3.6/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3.6/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Message recipient disconnected from message bus without replying
 
```

Running `systemctl status` after this gives:
```
┌─[griefcake][/usr/lib/systemd/user]
└──▪ systemctl --user status razer-service.service 
● razer-service.service - Razer Daemon (User)
   Loaded: loaded (/usr/lib/systemd/user/razer-service.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2017-06-17 14:28:10 MST; 38s ago
  Process: 5121 ExecStart=/usr/bin/razer-daemon --foreground --verbose --run-dir %h/.razer-service/data --log-dir %h/.razer-service/logs --pid-file %h/.razer-service/daemon.pid (code=exited, status=1/FAILURE)
 Main PID: 5121 (code=exited, status=1/FAILURE)

Jun 17 14:28:10 griefcake razer-daemon[5121]:     self.serial = self.get_serial()
Jun 17 14:28:10 griefcake razer-daemon[5121]:   File "/usr/lib/python3.6/site-packages/razer_daemon/hardware/device_base.py", line 202, in get_serial
Jun 17 14:28:10 griefcake razer-daemon[5121]:     serial = open(serial_path, 'r').read().strip()
Jun 17 14:28:10 griefcake razer-daemon[5121]:   File "/usr/lib/python3.6/codecs.py", line 318, in decode
Jun 17 14:28:10 griefcake razer-daemon[5121]:     def decode(self, input, final=False):
Jun 17 14:28:10 griefcake razer-daemon[5121]: TypeError: idle_handler() takes 0 positional arguments but 2 were given
Jun 17 14:28:10 griefcake systemd[414]: razer-service.service: Main process exited, code=exited, status=1/FAILURE
Jun 17 14:28:10 griefcake systemd[414]: Stopped Razer Daemon (User).
Jun 17 14:28:10 griefcake systemd[414]: razer-service.service: Unit entered failed state.
Jun 17 14:28:10 griefcake systemd[414]: razer-service.service: Failed with result 'exit-code'.

```