openrazer/openrazer #313: Black Widow Chroma device not detected by driver

idiotninja, 2017-06-18:
Fedora 25

I installed the drivers using the fedora installation instructions here: https://terrycain.github.io/razer-drivers/#download

I added my user to the plugdev group and rebooted. It appears that the daemon came up normally.
[user@saitama ~]$ ps -aux | grep razer
user     2143  0.0  0.4 696188 35936 tty2     Sl+  23:25   0:00 razer-daemon
user     3253 10.6  3.2 1408928 260504 tty2   SLl+ 23:26   0:40 /opt/google/chrome/chrome https://github.com/terrycain/razer-drivers#device-support

LSUSB Shows my early model Black Widow Chroma:
[user@saitama ~]$ lsusb
[...]
Bus 004 Device 003: ID 1532:0203 Razer USA, Ltd 
[...]

but when I open polychromatic I get the error in the attached screenshot. I am also attaching an SOSReport Just in case this helps. This is a relatively new installation with little customization so far.
[image]

[sosreport.zip](https://github.com/terrycain/razer-drivers/files/1082886/sosreport.zip)

z3ntu, 2017-06-18:
Please post the output of `sudo dkms status razer_chroma_driver/1.0.0` and 
`python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"`

z3ntu, 2017-06-22:
Please respond otherwise I'll close this issue.

idiotninja, 2017-06-22:
I apologize. Things got away from me.

[jesse@saitama ~]$ python3 -c "from razer.client import DeviceManager; a =
DeviceManager(); print(a.devices)"
[]
[jesse@saitama ~]$ sudo dkms status razer_chroma_driver/1.0.0
razer_chroma_driver, 1.0.0: added

z3ntu, 2017-06-22:
`lsmod | grep razer; sudo modprobe razerkbd; lsmod | grep razer` please

idiotninja, 2017-06-23:
I'll get that to you tonight

idiotninja, 2017-06-24:
well this is interesting

[root@saitama ~]# lsmod | grep razer; sudo modprobe razerkbd; lsmod | grep
razer
modprobe: FATAL: Module razerkbd not found in directory
/lib/modules/4.8.6-300.fc25.x86_64

Where does that kernel module come from? I'm assuming the drivers?

z3ntu, 2017-06-24:
`sudo dkms install razer_chroma_driver/1.0.0` please

idiotninja, 2017-06-24:
I'm not terribly familiar with this depth of kernel stuff it seems to me I
have some broken symlinks in kernel land. Should I manually re-symlink them?

[root@saitama ~]# sudo dkms install razer_chroma_driver/1.0.0
^[[AError! echo
Your kernel headers for kernel 4.8.6-300.fc25.x86_64 cannot be found at
/lib/modules/4.8.6-300.fc25.x86_64/build or
/lib/modules/4.8.6-300.fc25.x86_64/source.
[root@saitama ~]# ll  /lib/modules/4.8.6-300.fc25.x86_64/source
lrwxrwxrwx. 1 root root 5 Nov  1  2016
/lib/modules/4.8.6-300.fc25.x86_64/source -> build
[root@saitama ~]# ll  /lib/modules/4.8.6-300.fc25.x86_64/build
lrwxrwxrwx. 1 root root 38 Nov  1  2016
/lib/modules/4.8.6-300.fc25.x86_64/build ->
/usr/src/kernels/4.8.6-300.fc25.x86_64
[root@saitama ~]# ll /usr/src/kernels/4.11.5-200.fc25.x86_64/
total 4804
drwxr-xr-x.  33 root root    4096 Jun 16 20:59 arch
drwxr-xr-x.   3 root root    4096 Jun 16 20:59 block
drwxr-xr-x.   2 root root    4096 Jun 16 20:59 certs
drwxr-xr-x.   4 root root    4096 Jun 16 20:59 crypto
drwxr-xr-x. 129 root root    4096 Jun 16 20:59 drivers
drwxr-xr-x.   2 root root    4096 Jun 16 20:59 firmware
drwxr-xr-x.  74 root root    4096 Jun 16 20:59 fs
drwxr-xr-x.  30 root root    4096 Jun 16 20:59 include
drwxr-xr-x.   2 root root    4096 Jun 16 20:59 init
drwxr-xr-x.   2 root root    4096 Jun 16 20:59 ipc
-rw-r--r--.   1 root root     252 Jun 14 13:38 Kconfig
drwxr-xr-x.  16 root root    4096 Jun 16 20:59 kernel
drwxr-xr-x.  12 root root    4096 Jun 16 20:59 lib
-rw-r--r--.   1 root root   59260 Jun 14 13:38 Makefile
drwxr-xr-x.   3 root root    4096 Jun 16 20:59 mm
-rw-r--r--.   1 root root 1208604 Jun 14 13:38 Module.symvers
drwxr-xr-x.  67 root root    4096 Jun 16 20:59 net
drwxr-xr-x.  26 root root    4096 Jun 16 20:59 samples
drwxr-xr-x.  14 root root    4096 Jun 16 20:59 scripts
drwxr-xr-x.  10 root root    4096 Jun 16 20:59 security
drwxr-xr-x.  24 root root    4096 Jun 16 20:59 sound
-rw-r--r--.   1 root root 3550927 Jun 14 13:38 System.map
drwxr-xr-x.  27 root root    4096 Jun 16 20:59 tools
drwxr-xr-x.   2 root root    4096 Jun 16 20:59 usr
drwxr-xr-x.   4 root root    4096 Jun 16 20:59 virt
-rw-r--r--.   1 root root      41 Jun 14 13:38 vmlinux.id

nekrodomus, 2017-06-24:
nekrodomus@HAL9000:~$ sudo dkms install razer_chroma_driver/1.0.0
[sudo] password for nekrodomus: 
Module razer_chroma_driver/1.0.0 already installed on kernel 4.8.0-56-generic/x86_64

idiotninja, 2017-06-24:
Ok ill check it out. Doing an update in the hopes that it fixes my kernel
headers

idiotninja, 2017-06-24:
Yeh DKMS still can't find my kernel headers... weird...

# sudo dkms install razer_chroma_driver/1.0.0
Error! echo
Your kernel headers for kernel 4.8.6-300.fc25.x86_64 cannot be found at
/lib/modules/4.8.6-300.fc25.x86_64/build or
/lib/modules/4.8.6-300.fc25.x86_64/source.

z3ntu, 2017-06-24:
Are you sure, you followed the instructions in the website (https://terrycain.github.io/razer-drivers/#fedora)? I'd guess the package kernel-devel is missing.

Stayfirst, 2017-06-25:
I have same issue and here are my outputs for `sudo dkms status razer_chroma_driver/1.0.0`

```
~$ sudo dkms status razer_chroma_driver/1.0.0
razer_chroma_driver, 1.0.0, 4.8.0-36-generic, x86_64: installed
razer_chroma_driver, 1.0.0, 4.8.0-56-generic, x86_64: installed
```
and `python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)" `
```
~$ sudo  python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 175, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 361, in get_name_owner
    's', (bus_name,), **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.razer': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer/client/__init__.py", line 20, in __init__
    self._dbus = session_bus.get_object("org.razer", "/org/razer")
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.razer was not provided by any .service files

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/razer/client/__init__.py", line 22, in __init__
    raise DaemonNotFound("Could not connect to daemon")
razer.client.DaemonNotFound: Could not connect to daemon
```
`

Getting same exact screen with my Razer BlackWidow Ultimate 2013 	1532:011A in Ubuntu 16.04 Xenial

z3ntu, 2017-06-25:
I'll check later today in my VM. Please post the output of `rpm -q kernel-devel` anyways.

Stayfirst, 2017-06-25:
I'm on Ubuntu so you probably mean:
```

$ uname -r
4.8.0-56-generic

```
Thnxx in advance.

z3ntu, 2017-06-25:
@Stayfirst  `sudo dkms install razer_chroma_driver/1.0.0` ? And did you try rebooting?

Stayfirst, 2017-06-25:
```
~$ sudo dkms install razer_chroma_driver/1.0.0
Module razer_chroma_driver/1.0.0 already installed on kernel 4.8.0-56-generic/x86_64
```
Yes, tried restarting just now again and again had to unplug and plug the keyboard back in so I can at least type.

z3ntu, 2017-06-25:
`sudo modprobe razerkbd`

Stayfirst, 2017-06-25:
nothing

Stayfirst, 2017-06-25:
unity

z3ntu, 2017-06-25:
Please run `razer-daemon -F`

Stayfirst, 2017-06-25:
Daemon already exists. Please stop that one.

z3ntu, 2017-06-25:
`python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"` still gives an error, yes?

Stayfirst, 2017-06-25:
```
$ sudo python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 175, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 361, in get_name_owner
    's', (bus_name,), **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.razer': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer/client/__init__.py", line 20, in __init__
    self._dbus = session_bus.get_object("org.razer", "/org/razer")
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.razer was not provided by any .service files

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/razer/client/__init__.py", line 22, in __init__
    raise DaemonNotFound("Could not connect to daemon")
razer.client.DaemonNotFound: Could not connect to daemon

```

z3ntu, 2017-06-25:
`tail -n50 ~/.razer-service/logs/razer.log`

Stayfirst, 2017-06-25:
```
~$ sudo tail -n50 ~/.razer-service/logs/razer.log
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderChroma ------------------- (1532:0043)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderElite -------------------- (1532:005C)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerImperiator ------------------------- (1532:002F)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wired --------------------- (1532:0024)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wireless ------------------ (1532:0025)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerNaga2014 --------------------------- (1532:0040)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerNagaHex ---------------------------- (1532:0041)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexRed ------------------------- (1532:0036)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexV2 -------------------------- (1532:0050)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerNostromo --------------------------- (1532:0111)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerOrbweaver -------------------------- (1532:0113)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerOrnata ----------------------------- (1532:021F)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerOuroboros -------------------------- (1532:0032)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerTaipan ----------------------------- (1532:0034)
2017-06-25 15:55:43 | razer                          | DEBUG    | Loaded device specification: RazerTartarus --------------------------- (1532:0208)
2017-06-25 15:55:43 | razer                          | INFO     | Adding razer.devices.getDevices method to DBus
[... cut ...]
2017-06-25 15:55:46 | razer                          | DEBUG    | Device event [remove]: /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1:1.2/0003:1532:011A.0003
2017-06-25 15:57:22 | razer                          | DEBUG    | Device event [add]: /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1:1.0/0003:1532:011A.000D
2017-06-25 15:57:22 | razer                          | DEBUG    | Device event [add]: /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1:1.1/0003:1532:011A.000E
2017-06-25 15:57:22 | razer                          | DEBUG    | Device event [add]: /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1:1.2/0003:1532:011A.000F

```

z3ntu, 2017-06-25:
`ps -aux | grep razer-daemon`

Stayfirst, 2017-06-25:
```
~$ sudo ps -aux | grep razer-daemon
i         1956  0.0  0.4 370872 34104 ?        Sl   15:55   0:00 razer-daemon
i         2944  0.0  0.0  16272   932 pts/17   S+   16:12   0:00 grep --color=auto razer-daemon

```

z3ntu, 2017-06-25:
You are user `i` ?

Stayfirst, 2017-06-25:
yes ;)

terrycain, 2017-06-25:
```
pkill razer-daemon
razer-daemon -F

# In another terminal
python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
```

terrycain, 2017-06-25:
Give output of both razer-daemon -F and the python cmd

Stayfirst, 2017-06-25:
pkill razer-daemon
razer-daemon -F
```

~$ sudo razer-daemon -F
2017-06-25 17:46:18 | razer                          | INFO     | Initialising Daemon (v1.1.13). Pid: 17592
2017-06-25 17:46:18 | razer                          | CRITICAL | User is not a member of the plugdev group

```
# In another terminal
python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
```

~$ sudo python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
[sudo] password for i: 
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 175, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 361, in get_name_owner
    's', (bus_name,), **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.razer': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer/client/__init__.py", line 20, in __init__
    self._dbus = session_bus.get_object("org.razer", "/org/razer")
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.razer was not provided by any .service files

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/razer/client/__init__.py", line 22, in __init__
    raise DaemonNotFound("Could not connect to daemon")
razer.client.DaemonNotFound: Could not connect to daemon

```

terrycain, 2017-06-25:
try again without running razer-daemon -F
 as sudo

Stayfirst, 2017-06-25:
first terminal:
`~$ sudo pkill razer-daemon
`

Second terminal:
```
i:~$ sudo python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 175, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 361, in get_name_owner
    's', (bus_name,), **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.razer': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer/client/__init__.py", line 20, in __init__
    self._dbus = session_bus.get_object("org.razer", "/org/razer")
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.razer was not provided by any .service files

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/razer/client/__init__.py", line 22, in __init__
    raise DaemonNotFound("Could not connect to daemon")
razer.client.DaemonNotFound: Could not connect to daemon

```

terrycain, 2017-06-25:
wait so you killed the daemon but didnt run it without sudo?

terrycain, 2017-06-25:
https://github.com/terrycain/razer-drivers/issues/313#issuecomment-310929357

Stayfirst, 2017-06-25:
ok let me see 2 min

Stayfirst, 2017-06-25:
first terminal 
```
i@i:~$ pkill razer-daemon
i@i:~$ razer-daemon -F
2017-06-25 18:01:29 | razer                          | INFO     | Initialising Daemon (v1.1.13). Pid: 18003
2017-06-25 18:01:29 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2012 --------------------- (1532:010D)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderChroma ------------------- (1532:0043)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderElite -------------------- (1532:005C)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerImperiator ------------------------- (1532:002F)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wired --------------------- (1532:0024)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wireless ------------------ (1532:0025)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2017-06-25 18:01:29 | razer                          | DEBUG    | Loaded device specification: RazerNaga2014 --------------------------- (1532:0040)
[... cut ...]
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.razer was not provided by any .service files

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/razer/client/__init__.py", line 22, in __init__
    raise DaemonNotFound("Could not connect to daemon")
razer.client.DaemonNotFound: Could not connect to daemon

```

z3ntu, 2017-06-26:
Please don't run daemon stuff as root. There's something called a dbus session bus and that's user-specific. The empty array from the python command without root is correct.

Now run `ls -al /sys/bus/hid/drivers/razerkbd/0*/`.

Stayfirst, 2017-06-26:
```
~$ ls -al /sys/bus/hid/drivers/razerkbd/0*/
ls: cannot access '/sys/bus/hid/drivers/razerkbd/0*/': No such file or directory

```

z3ntu, 2017-06-26:
`ls -al /lib/udev/rules.d/99-razer.rules`

Stayfirst, 2017-06-26:
```
~$ ls -al /lib/udev/rules.d/99-razer.rules
-rw-r--r-- 1 root root 695 Jun 24 21:54 /lib/udev/rules.d/99-razer.rules
```

terrycain, 2017-06-26:
Whats the contents of that file

Stayfirst, 2017-06-26:
# Razer Keyboards
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0205", ENV{is_razer}="yes"    # Razer Blade Stealth 2016
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="020f", ENV{is_razer}="yes"    # Razer Blade 14 2016
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0203", ENV{is_razer}="yes"    # BlackWidow Chroma

# Run action
SUBSYSTEM=="hid", DRIVER=="hid-generic", ENV{is_razer}=="yes", ATTRS{bInterfaceNumber}=="02", RUN+="razer_mount hid-razer $kernel"

KERNEL=="event[0-9]*", SUBSYSTEM=="input"

terrycain, 2017-06-26:
dpkg --get-selections | grep -i razer

Stayfirst, 2017-06-26:
```
~$ dpkg --get-selections | grep -i razer
python3-razer					install
razer-daemon					install
razer-doc					install
razer-kernel-modules-dkms			install
razergenie					install

```

terrycain, 2017-06-26:
odd, as that 99-razer file is wrong. Replace it with [this](https://github.com/terrycain/razer-drivers/blob/master/install_files/udev/99-razer.rules), also replace `/lib/udev/razer_mount` with [this](https://github.com/terrycain/razer-drivers/blob/master/install_files/udev/razer_mount) (making sure razer_mount is executable).

Then reboot and rerun 
```
ls -al /sys/bus/hid/drivers/razerkbd/0*/
```

Stayfirst, 2017-06-26:
Got a crash report on startup "The application razer service has closed unexpectedly"

Here's the output from 
```
~$ ls -al /sys/bus/hid/drivers/razerkbd/0*/
/sys/bus/hid/drivers/razerkbd/0003:1532:011A.000A/:
total 0
drwxr-xr-x 5 root plugdev    0 Jun 26 17:44 .
drwxr-xr-x 5 root root       0 Jun 26 17:44 ..
-r--r--r-- 1 root plugdev 4096 Jun 26 17:44 country
lrwxrwxrwx 1 root plugdev    0 Jun 26 17:44 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 Jun 26 17:44 hidraw
drwxr-xr-x 3 root plugdev    0 Jun 26 17:44 input
-rw-rw---- 1 root plugdev 4096 Jun 26 17:44 key_alt_f4
-rw-rw---- 1 root plugdev 4096 Jun 26 17:44 key_alt_tab
-rw-rw---- 1 root plugdev 4096 Jun 26 17:44 key_super
-r--r--r-- 1 root plugdev 4096 Jun 26 17:44 modalias
drwxr-xr-x 2 root plugdev    0 Jun 26 17:44 power
-r--r--r-- 1 root plugdev 4096 Jun 26 17:44 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Jun 26 17:44 subsystem -> ../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Jun 26 17:44 uevent

/sys/bus/hid/drivers/razerkbd/0003:1532:011A.000B/:
total 0
drwxr-xr-x 5 root plugdev    0 Jun 26 17:44 .
drwxr-xr-x 5 root root       0 Jun 26 17:44 ..
-r--r--r-- 1 root plugdev 4096 Jun 26 17:44 country
lrwxrwxrwx 1 root plugdev    0 Jun 26 17:44 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 Jun 26 17:44 hidraw
drwxr-xr-x 3 root plugdev    0 Jun 26 17:44 input
-rw-rw---- 1 root plugdev 4096 Jun 26 17:44 key_alt_f4
-rw-rw---- 1 root plugdev 4096 Jun 26 17:44 key_alt_tab
-rw-rw---- 1 root plugdev 4096 Jun 26 17:44 key_super
-r--r--r-- 1 root plugdev 4096 Jun 26 17:44 modalias
drwxr-xr-x 2 root plugdev    0 Jun 26 17:44 power
-r--r--r-- 1 root plugdev 4096 Jun 26 17:44 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Jun 26 17:44 subsystem -> ../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Jun 26 17:44 uevent

/sys/bus/hid/drivers/razerkbd/0003:1532:011A.000C/:
total 0
drwxr-xr-x 5 root plugdev    0 Jun 26 17:44 .
drwxr-xr-x 5 root root       0 Jun 26 17:44 ..
-r--r--r-- 1 root plugdev 4096 Jun 26 17:44 country
-rw-rw---- 1 root plugdev 4096 Jun 26 17:44 device_mode
-r--r----- 1 root plugdev 4096 Jun 26 17:44 device_serial
-r--r----- 1 root plugdev 4096 Jun 26 17:44 device_type
lrwxrwxrwx 1 root plugdev    0 Jun 26 17:44 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
-r--r----- 1 root plugdev 4096 Jun 26 17:44 firmware_version
-rw-rw---- 1 root plugdev 4096 Jun 26 17:44 game_led_state
drwxr-xr-x 3 root plugdev    0 Jun 26 17:44 hidraw
drwxr-xr-x 3 root plugdev    0 Jun 26 17:44 input
-rw-rw---- 1 root plugdev 4096 Jun 26 17:44 macro_led_effect
-rw-rw---- 1 root plugdev 4096 Jun 26 17:44 macro_led_state
-rw-rw---- 1 root plugdev 4096 Jun 26 17:44 matrix_brightness
-rw-rw---- 1 root plugdev 4096 Jun 26 17:44 matrix_effect_pulsate
--w--w---- 1 root plugdev 4096 Jun 26 17:44 matrix_effect_static
-r--r--r-- 1 root plugdev 4096 Jun 26 17:44 modalias
drwxr-xr-x 2 root plugdev    0 Jun 26 17:44 power
-r--r--r-- 1 root plugdev 4096 Jun 26 17:44 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Jun 26 17:44 subsystem -> ../../../../../../../../bus/hid
-rw-rw---- 1 root plugdev 4096 Jun 26 17:44 test
-rw-r--r-- 1 root plugdev 4096 Jun 26 17:44 uevent
-r--r----- 1 root plugdev 4096 Jun 26 17:44 version

```
and again had to unplug and plug in keyboard for it to function.

terrycain, 2017-06-26:
Ok the title is misleading as it's not a chroma it's an ultimate 2013. What happens if you Run that ls again now the kb works

Stayfirst, 2017-06-26:
keyboard works when I unplug it and plug back in, but I can try reboot again and use the last command with onscreen kboard. Should I?

terrycain, 2017-06-26:
Nah, run that ls command now that the keyboard is working, is it any different?

Stayfirst, 2017-06-26:
it was exactly the same but the polychromatic app booted up and worked perfectly fine

I did a reboot now again and no crash but:

```
~$ ls -al /sys/bus/hid/drivers/razerkbd/0*/
ls: cannot access '/sys/bus/hid/drivers/razerkbd/0*/': No such file or directory
```

Stayfirst, 2017-06-26:
after unplug and plug in

```
~$ ls -al /sys/bus/hid/drivers/razerkbd/0*/
/sys/bus/hid/drivers/razerkbd/0003:1532:011A.000A/:
total 0
drwxr-xr-x 5 root plugdev    0 Jun 26 18:02 .
drwxr-xr-x 5 root root       0 Jun 26 18:02 ..
-r--r--r-- 1 root plugdev 4096 Jun 26 18:02 country
lrwxrwxrwx 1 root plugdev    0 Jun 26 18:02 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 Jun 26 18:02 hidraw
drwxr-xr-x 3 root plugdev    0 Jun 26 18:02 input
-rw-rw---- 1 root plugdev 4096 Jun 26 18:02 key_alt_f4
-rw-rw---- 1 root plugdev 4096 Jun 26 18:02 key_alt_tab
-rw-rw---- 1 root plugdev 4096 Jun 26 18:02 key_super
-r--r--r-- 1 root plugdev 4096 Jun 26 18:02 modalias
drwxr-xr-x 2 root plugdev    0 Jun 26 18:02 power
-r--r--r-- 1 root plugdev 4096 Jun 26 18:02 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Jun 26 18:02 subsystem -> ../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Jun 26 18:02 uevent

/sys/bus/hid/drivers/razerkbd/0003:1532:011A.000B/:
total 0
drwxr-xr-x 5 root plugdev    0 Jun 26 18:02 .
drwxr-xr-x 5 root root       0 Jun 26 18:02 ..
-r--r--r-- 1 root plugdev 4096 Jun 26 18:02 country
lrwxrwxrwx 1 root plugdev    0 Jun 26 18:02 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 Jun 26 18:02 hidraw
drwxr-xr-x 3 root plugdev    0 Jun 26 18:02 input
-rw-rw---- 1 root plugdev 4096 Jun 26 18:02 key_alt_f4
-rw-rw---- 1 root plugdev 4096 Jun 26 18:02 key_alt_tab
-rw-rw---- 1 root plugdev 4096 Jun 26 18:02 key_super
-r--r--r-- 1 root plugdev 4096 Jun 26 18:02 modalias
drwxr-xr-x 2 root plugdev    0 Jun 26 18:02 power
-r--r--r-- 1 root plugdev 4096 Jun 26 18:02 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Jun 26 18:02 subsystem -> ../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Jun 26 18:02 uevent

/sys/bus/hid/drivers/razerkbd/0003:1532:011A.000C/:
total 0
drwxr-xr-x 5 root plugdev    0 Jun 26 18:02 .
drwxr-xr-x 5 root root       0 Jun 26 18:02 ..
-r--r--r-- 1 root plugdev 4096 Jun 26 18:02 country
-rw-rw---- 1 root plugdev 4096 Jun 26 18:02 device_mode
-r--r----- 1 root plugdev 4096 Jun 26 18:02 device_serial
-r--r----- 1 root plugdev 4096 Jun 26 18:02 device_type
lrwxrwxrwx 1 root plugdev    0 Jun 26 18:02 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
-r--r----- 1 root plugdev 4096 Jun 26 18:02 firmware_version
-rw-rw---- 1 root plugdev 4096 Jun 26 18:02 game_led_state
drwxr-xr-x 3 root plugdev    0 Jun 26 18:02 hidraw
drwxr-xr-x 3 root plugdev    0 Jun 26 18:02 input
-rw-rw---- 1 root plugdev 4096 Jun 26 18:02 macro_led_effect
-rw-rw---- 1 root plugdev 4096 Jun 26 18:02 macro_led_state
-rw-rw---- 1 root plugdev 4096 Jun 26 18:02 matrix_brightness
-rw-rw---- 1 root plugdev 4096 Jun 26 18:02 matrix_effect_pulsate
--w--w---- 1 root plugdev 4096 Jun 26 18:02 matrix_effect_static
-r--r--r-- 1 root plugdev 4096 Jun 26 18:02 modalias
drwxr-xr-x 2 root plugdev    0 Jun 26 18:02 power
-r--r--r-- 1 root plugdev 4096 Jun 26 18:02 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Jun 26 18:02 subsystem -> ../../../../../../../../bus/hid
-rw-rw---- 1 root plugdev 4096 Jun 26 18:02 test
-rw-r--r-- 1 root plugdev 4096 Jun 26 18:02 uevent
-r--r----- 1 root plugdev 4096 Jun 26 18:02 version

```

terrycain, 2017-06-26:
Wtf. Ok do 
`find / -iname '*razer*''`
Might take a while

Stayfirst, 2017-06-26:
Corrected the ' to be " but here it is 
```

~$ find / -iname "*razer*"
/tmp/tmp_yda4hrig_razer_daemon
find: ‘/tmp/systemd-private-1e1b3e15c2e44f49bfad01da2378c89b-colord.service-YpX3xC’: Permission denied
find: ‘/tmp/systemd-private-1e1b3e15c2e44f49bfad01da2378c89b-systemd-timesyncd.service-Uwbar8’: Permission denied
find: ‘/tmp/systemd-private-1e1b3e15c2e44f49bfad01da2378c89b-rtkit-daemon.service-MrxhyY’: Permission denied
/usr/bin/razergenie
/usr/bin/razer-daemon
/usr/src/razer_chroma_driver-1.0.0
/usr/src/razer_chroma_driver-1.0.0/driver/razermouse_driver.c
/usr/src/razer_chroma_driver-1.0.0/driver/razercommon.c
/usr/src/razer_chroma_driver-1.0.0/driver/razerchromacommon.h
/usr/src/razer_chroma_driver-1.0.0/driver/razerkraken_driver.h
/usr/src/razer_chroma_driver-1.0.0/driver/razercore_driver.c
/usr/src/razer_chroma_driver-1.0.0/driver/razerchromacommon.c
/usr/src/razer_chroma_driver-1.0.0/driver/razerfirefly_driver.c
/usr/src/razer_chroma_driver-1.0.0/driver/razermouse_driver.h
/usr/src/razer_chroma_driver-1.0.0/driver/razermug_driver.c
/usr/src/razer_chroma_driver-1.0.0/driver/razerfirefly_driver.h
/usr/src/razer_chroma_driver-1.0.0/driver/razermug_driver.h
/usr/src/razer_chroma_driver-1.0.0/driver/razerkraken_driver.c
/usr/src/razer_chroma_driver-1.0.0/driver/razercommon.h
/usr/src/razer_chroma_driver-1.0.0/driver/razerkbd_driver.c
/usr/src/razer_chroma_driver-1.0.0/driver/razerkbd_driver.h
/usr/src/razer_chroma_driver-1.0.0/driver/razercore_driver.h
/usr/src/razer-drivers-1.0.0
/usr/src/razer-drivers-1.0.0/src/hid-razer.h
/usr/src/razer-drivers-1.0.0/src/hid-razer-common.c
/usr/src/razer-drivers-1.0.0/src/hid-razer.c
/usr/src/razer-drivers-1.0.0/src/hid-razer-common.h
/usr/lib/x86_64-linux-gnu/librazer.so
/usr/lib/x86_64-linux-gnu/librazer.so.0.0.1
/usr/lib/x86_64-linux-gnu/librazer.so.0
/usr/lib/python3/dist-packages/razer
/usr/lib/python3/dist-packages/razer_daemon
/usr/share/man/man5/razer.conf.5.gz
/usr/share/man/man8/razer-daemon.8.gz
/usr/share/doc/razergenie
/usr/share/doc/razer-daemon
/usr/share/doc/razer-doc
/usr/share/doc/razer-kernel-modules-dkms
/usr/share/doc/python3-razer
/usr/share/icons/hicolor/scalable/apps/razergenie.svg
/usr/share/razer-daemon
/usr/share/razer-daemon/razer.conf.example
/usr/share/applications/razergenie.desktop
/dev/input/by-id/usb-Razer_Razer_BlackWidow_Ultimate-if02-event-mouse
/dev/input/by-id/usb-Razer_Razer_BlackWidow_Ultimate-if02-mouse
/dev/input/by-id/usb-Razer_Razer_BlackWidow_Ultimate-if01-event-kbd
/dev/input/by-id/usb-Razer_Razer_BlackWidow_Ultimate-event-kbd
/dev/input/by-id/usb-Razer_Razer_Firefly-event-mouse
/dev/input/by-id/usb-Razer_Razer_Firefly-if02-event-kbd
/dev/input/by-id/usb-Razer_Razer_Firefly-if01-event-kbd
/dev/input/by-id/usb-Razer_Razer_Firefly-mouse
find: ‘/home/lost+found’: Permission denied
find: ‘/home/i/.dbus/session-bus’: Permission denied
/home/i/.razer-service
/home/i/.razer-service/razer.conf
/home/i/.razer-service/logs/razer.log
/home/i/Downloads/openrazer-drivers-master
/home/i/Downloads/openrazer-drivers-master/scripts/razer-test
/home/i/Downloads/openrazer-drivers-master/scripts/razer-set-colors
/home/i/Downloads/openrazer-drivers-master/udev/razer_mount
/home/i/Downloads/openrazer-drivers-master/udev/99-razer.rules
/home/i/Downloads/openrazer-drivers-master/Documentation/sysfs-driver-hid-razer
/home/i/Downloads/openrazer-drivers-master/src/hid-razer.c
/home/i/Downloads/openrazer-drivers-master/src/hid-razer-common.h
/home/i/Downloads/openrazer-drivers-master/src/hid-razer-common.c
[... 1623 lines cut ...]
/var/lib/dkms/razer_chroma_driver/1.0.0/build/driver/razerkbd_driver.h
/var/lib/dkms/razer_chroma_driver/1.0.0/build/driver/razercore_driver.h
/var/lib/dkms/razer_chroma_driver/1.0.0/4.8.0-56-generic/x86_64/module/razermouse.ko
/var/lib/dkms/razer_chroma_driver/1.0.0/4.8.0-56-generic/x86_64/module/razermug.ko
/var/lib/dkms/razer_chroma_driver/1.0.0/4.8.0-56-generic/x86_64/module/razerkraken.ko
/var/lib/dkms/razer_chroma_driver/1.0.0/4.8.0-56-generic/x86_64/module/razerkbd.ko
/var/lib/dkms/razer_chroma_driver/1.0.0/4.8.0-56-generic/x86_64/module/razercore.ko
/var/lib/dkms/razer_chroma_driver/1.0.0/4.8.0-56-generic/x86_64/module/razerfirefly.ko
/var/lib/dkms/razer_chroma_driver/1.0.0/4.8.0-36-generic/x86_64/module/razermouse.ko
/var/lib/dkms/razer_chroma_driver/1.0.0/4.8.0-36-generic/x86_64/module/razermug.ko
/var/lib/dkms/razer_chroma_driver/1.0.0/4.8.0-36-generic/x86_64/module/razerkraken.ko
/var/lib/dkms/razer_chroma_driver/1.0.0/4.8.0-36-generic/x86_64/module/razerkbd.ko
/var/lib/dkms/razer_chroma_driver/1.0.0/4.8.0-36-generic/x86_64/module/razercore.ko
/var/lib/dkms/razer_chroma_driver/1.0.0/4.8.0-36-generic/x86_64/module/razerfirefly.ko
find: ‘/var/lib/colord/.cache’: Permission denied
find: ‘/var/lib/bluetooth/84:E0:F4:03:28:81’: Permission denied
find: ‘/var/spool/cron/crontabs’: Permission denied
find: ‘/var/spool/cups’: Permission denied
find: ‘/var/spool/rsyslog’: Permission denied
find: ‘/root/.gnupg’: Permission denied
find: ‘/root/.npm’: Permission denied
find: ‘/root/.cache’: Permission denied
find: ‘/root/.config’: Permission denied

```

idiotninja, 2017-06-27:
Kernel-devel is installed:

# rpm -qa | grep kernel-devel
kernel-devel-4.11.5-200.fc25.x86_64
kernel-devel-4.11.6-201.fc25.x86_64

Openrazer-meta is also installed:

# rpm -qa | grep openrazer-meta
openrazer-meta-1.1.13-2.1.noarch

To my knowledge I did follow those instructions and even restarted. What do
we test/check now?

Stayfirst, 2017-06-27:
Ok this a weird but good (I guess) update to my situation.

Keyboard works perfectly normal now and no longer needs a replug on startup. 

Funny thing is I didn't mess with the kbd at all, instead I messed with my sound switching between analog and s/pdif. 

Fixed it by disabling nvidia as an output option for sound completely so now I have default s/pdif output and it switches when analog is plugged in, unplugged - switches back to spdif. 

Integrated keyboard (I assume passthrough) connection works too, however it doesn't work the same as plugging in or out as directly to mobo. Does nothing basically besides extending the sound output. 

I'm not sure what to point fingers at here but I'm leaning towards gtx970 by nvidia.

Thnx again for doing gods work you guys are amazing and the superb support I received, keep it up! I'll be checking in for the sound port solution every now and then and perhaps anything I can contribute.

Have a great day you wonderful people!

terrycain, 2017-06-27:
Yeah I want to blame nvidia ;-), the passthrough on the keyboard is dumb and should work regardless, its just an extention cable afaik

closed 2017-06-27 by terrycain

z3ntu, 2017-06-27:
@idiotninja sorry that your issue got hijacked 😄 

```
uname -r
ls -al /usr/src/kernels/
```