openrazer/openrazer #314: Unable to fix Error 2 using Troubleshooting Method

adi105, 2017-06-19:
From the start, forgive me as I am fairly new to linux and I want to get my razer keyboard to work with it. I have Kali Linux and I am trying to install the razer drivers, however once I clone the repo and try to run make inside of it, it gives me an error 2. The output is as follows:
### :: Compiling Razer kernel modules
========================================
make -C /lib/modules/4.9.0-kali3-amd64/build SUBDIRS=/root/Desktop/razer-drivers/driver modules
make[1]: *** /lib/modules/4.9.0-kali3-amd64/build: No such file or directory.  Stop.
Makefile:23: recipe for target 'driver' failed
make: *** [driver] Error 2

Now, I notice that in the SUBDIRS path, it has .../driver modules. This does not exist inside of the razer-drives repo. Any help would be greatly appreciated, as I have been toying with this for a while and it still will not work.

z3ntu, 2017-06-19:
Please follow the instructions on our website for "Debian Testing": https://terrycain.github.io/razer-drivers/#debian

adi105, 2017-06-20:
I followed the directions on this page exactly pertaining to debian: 
https://software.opensuse.org/download.html?project=hardware%3Arazer&package=openrazer-meta 
The drivers still don't seem to be installed. It ran as I assume it should, but when I run the main.py for RazerCommander, it gives me this message:
"Importing device logic, waiting for daemon
ERROR: the daemon is not responding!
Try running `killall razer-service && razer-service` or rebooting. If this doesn't work, please fill an issue!"
Now, running killall razer-service tells me there is no such process as razer-service, so I am assuming that I do not have the drivers. Is there any way for me to explicitly check for this? Thanks for any and all help; like I said, I'm super new to linux.

z3ntu, 2017-06-20:
Please post the output of `sudo dkms status razer_chroma_driver/1.0.0` and of `lsmod | grep razer`

tildearrow, 2017-06-20:
Did you try `killall razer-daemon && razer-daemon`?

z3ntu, 2017-06-20:
He did try but the bash && only runs the second command when the first is successful and the killall command returned "No such process"

adi105, 2017-06-20:
sudo dkms status razer_chroma_driver/1.0.0:
**razer_chroma_driver, 1.0.0, 4.9.0-kali4-amd64, x86_64: installed
**
lsmod | grep razer did not return any output

z3ntu, 2017-06-20:
Please run `sudo modprobe razerkbd`. I hope you have rebooted since installing btw.

adi105, 2017-06-20:
Yes sir, I have rebooted. sudo modprobe razerkbd outputs:
modprobe: FATAL: Module razerkbd not found in directory /lib/modules/4.9.0-kali3-amd64

z3ntu, 2017-06-20:
`sudo dkms install razer_chroma_driver/1.0.0`

adi105, 2017-06-20:
sudo dkms install razer_chroma_driver/1.0.0 returns:
**Error! echo
Your kernel headers for kernel 4.9.0-kali3-amd64 cannot be found at
/lib/modules/4.9.0-kali3-amd64/build or /lib/modules/4.9.0-kali3-amd64/source.**

z3ntu, 2017-06-20:
You have to install the kernel headers for your kernel. Don't know the package name for Kali though.

z3ntu, 2017-06-20:
You could try `sudo apt-get install linux-headers-$(uname -r)`

adi105, 2017-06-20:
so I updated the kernel version itself since I did have a non-recent version of kali through:
sudo apt-get -y dist-upgrade

Now, when I run sudo dkms install razer_chroma_driver/1.0.0, it says:
**Module razer_chroma_driver/1.0.0 already installed on kernel 4.9.0-kali4-amd64/x86_64
**
Does this mean that the drivers were installed?

adi105, 2017-06-20:
Also, searching lsmod | grep razer yields:
razermouse             61440  0
razerkbd               53248  0
usbcore               249856  8 razermouse,usbhid,razerkbd,usb_storage,usblp,xhci_pci,uas,xhci_hcd
hid                   122880  5 i2c_hid,razermouse,hid_generic,usbhid,razerkbd

z3ntu, 2017-06-20:
That's good. Should work now (otherwise reboot)

adi105, 2017-06-20:
Thanks for all the help last night. When I run polychromatic, it tells me the daemon is not running, and when I run razer-daemon -f it tells me this:
**raceback (most recent call last):
  File "/usr/bin/razer-daemon", line 9, in <module>
    from razer_daemon.daemon import daemonize
ModuleNotFoundError: No module named 'razer_daemon'
**
All of the above is still working, the drivers appear to be installed, but this error still persists even after reboots.

lah7, 2017-06-21:
All seems to be working now, as reported here: https://github.com/lah7/polychromatic/issues/99

closed 2017-06-21 by lah7