openrazer/openrazer #315: Using the driver broke my network card driver

ChristopherBare, 2017-06-22:
As the title suggests, I installed the Razer drivers and rebooted. My network was Ethernet only and the WiFi was unavailable. I uninstalled the Razer drivers and it worked again.

z3ntu, 2017-06-22:
Uhm can you please post `sudo journalctl | grep -i razer_mount`

ChristopherBare, 2017-06-22:
It doesn't return anything when I run that.

z3ntu, 2017-06-22:
Then there is afaik no possibility of our driver interfering. Can you try just installing the kernel driver (package name depends on your distro but it's probably razer-kernel-modules-dkms or razer-driver-dkms) and try that?

ChristopherBare, 2017-06-22:
This was what I ran to install it from the PPA: `sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc`. I'm running Ubuntu 16.04 - Cinnamon if that helps.

z3ntu, 2017-06-22:
Yes, please run `sudo apt install razer-kernel-modules-dkms` and then reboot.

ChristopherBare, 2017-06-22:
Okay, I did that. The WiFi appears to be behaving normally again. Thank you!

z3ntu, 2017-06-22:
Okay now install the rest of the packages.

ChristopherBare, 2017-06-22:
Alright, I did that.

z3ntu, 2017-06-22:
Does wifi still work (after reboot)?

ChristopherBare, 2017-06-22:
It does.

z3ntu, 2017-06-22:
¯\_(ツ)_/¯

closed 2017-06-22 by z3ntu

solofortitude, 2017-07-21:
Hi, 
Did the location change?  I have a razer bland 2014 and i have lost the ability to use my wi-fi.

thx!

:/home/raven# apt install razer-kernel-modules-dkms
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package razer-kernel-modules-dkms