openrazer/openrazer #318: Razer Ornata Chroma device not detected by driver

nekrodomus, 2017-06-23:
I installed the drivers using the Ubuntu installation instructions here: 

https://terrycain.github.io/razer-drivers/#download

I added my user to the plugdev group and rebooted. It appears that the daemon came up normally.

nekrodomus@HAL9000:~$ ps -aux | grep razer
nekrodo+  4956  0.0  0.1 375008 32268 ?        Sl   16:32   0:00 razer-daemon
nekrodo+ 15815  0.0  0.0  18808  1020 pts/0    S+   20:46   0:00 grep --color=auto razer

LSUSB Shows my keyboard:

nekrodomus@HAL9000:~$ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 1532:021e Razer USA, Ltd 
Bus 001 Device 002: ID 046d:c062 Logitech, Inc. M-UAS144 [LS1 Laser Mouse]
Bus 001 Device 004: ID 1b1c:0c03 Corsair 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

But when I open polychromatic I get the error in the attached screenshot.
[image]

Output of: sudo dkms status razer_chroma_driver/1.0.0

nekrodomus@HAL9000:~$ sudo dkms status razer_chroma_driver/1.0.0
razer_chroma_driver, 1.0.0, 4.8.0-54-generic, x86_64: installed
razer_chroma_driver, 1.0.0, 4.8.0-56-generic, x86_64: installed

Output of: 

nekrodomus@HAL9000:~$ python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
[]

Output of: lsmod | grep razer; sudo modprobe razerkbd

nekrodomus@HAL9000:~$ lsmod | grep razer; sudo modprobe razerkbd
modprobe: ERROR: could not insert 'razerkbd': Required key not available

z3ntu, 2017-06-23:
Please see https://github.com/terrycain/razer-drivers/wiki/Troubleshooting#late-razer-blade-stealth

z3ntu, 2017-06-23:
Basically disable secure boot in the UEFI for the easy solution. Hard solution would be signing the modules.

z3ntu, 2017-06-23:
You posted the comment to the wrong issue 😆  Please close if it's really solved.

nekrodomus, 2017-06-24:
Thanks z3ntu ;)

closed 2017-06-24 by nekrodomus