openrazer/openrazer #319: Multi user System: Ubuntu 17.04 - daemon not started for 2nd user

rumplstielz, 2017-06-25:
hi, just posted that on the polychromatic github, but they are saying that this seems to be a daemon issue:
 have a multi user system. 

The problem: when I am logged in as one user and switching to another user, the icon goes red, which seems to show that the daemon is not running.
Restarting the daemon from the context menu of the polychromatic icon also does not work (stays at "killing daemon").

So for me it also looks like a daemon issue.

both users are in the plugdev group. 46(plugdev) --> I only see the problem if both users are logged in. 
The first user sees the icon + daemon ok. 
When you login with the second user you see the problem with this second user.

From checking the running processes with "razer" in it, it seems, that the daemon is only started for the 1st user and not for the 2nd user.

Please help, this is anoying ;-)

br, rumpl

[razer_daemon_ps1stuser.txt](https://github.com/terrycain/razer-drivers/files/1100362/razer_daemon_ps1stuser.txt)
[rzrdaemon_ps_2nduser.txt](https://github.com/terrycain/razer-drivers/files/1100361/rzrdaemon_ps_2nduser.txt)

rumplstielz, 2017-06-25:
Don't know what 

[razer_daemon_ps1stuser.txt](https://github.com/terrycain/razer-drivers/files/1100364/razer_daemon_ps1stuser.txt)
[rzrdaemon_ps_2nduser.txt](https://github.com/terrycain/razer-drivers/files/1100363/rzrdaemon_ps_2nduser.txt)

z3ntu, 2017-06-25:
You still need help? 😄

rumplstielz, 2017-06-25:
yeah, sorry, was a little bit confused by the interface here and had to figure out, why files with the .log ending won't upload...

terrycain, 2017-06-25:
Yeah most likely is the daemon checking that its not already running. Will have a look into adjusting the check to take into account users

rumplstielz, 2017-06-25:
hey, thank you for your fast answer! This sounds great!

rumplstielz, 2017-06-25:
Danke übrigens auch dir @z3ntu   ;-)

terrycain, 2017-07-01:
Latest code should fix this

terrycain, 2017-07-01:
Version 1.1.14 is out on launchpad, non Ubuntu packages will be out shortly.

closed 2017-07-01 by terrycain