openrazer/openrazer #320: Fedora 26

bbugarski, 2017-06-27:
Hey guys,
I am running Fedora 26, though I can't install the drivers.
Both py 3.5 and 3.6 are installed though I am getting the following:

```
$ sudo dnf install razer-daemon-1.1.13-2.1.noarch 
Last metadata expiration check: 0:28:07 ago on Tue 27 Jun 2017 09:38:56 PM CEST.
Error: 
 Problem: conflicting requests
  - nothing provides python(abi) = 3.5 needed by razer-daemon-1.1.13-2.1.noarch

```

```
$ python3.5 -V
Python 3.5.2
```

z3ntu, 2017-06-27:
I'm pretty sure you have selected Fedora 25, right? I've just enabled Fedora 26 (and also Rawhide) so it should be available soon on the download page.

z3ntu, 2017-06-27:
Builds are finished. It'll take a while though to show up on the download page. 
Command to add the new repo is

```
dnf config-manager --add-repo http://download.opensuse.org/repositories/hardware:razer/Fedora_26/hardware:razer.repo
```
Maybe you have to remove the old repository before, I'm not really familiar with dnf; but I'm sure you'll find the solution.

closed 2017-06-27 by z3ntu

aelkz, 2017-11-10:
still not working for me:
Fedora 26
4.13.11-100.fc25.x86_64

Did someone solve this issue? Thanks

z3ntu, 2017-11-10:
@aelkz can you provide some logs? Otherwise I can't tell at all what issue you hit

aelkz, 2017-11-10:
sorry @z3ntu , I was installing the Fedora 26 packages. I'm using Fedora 25.
Now it is working! 

Steps that I've realized:
1. uninstall python
`$ sudo dnf remove python`
2. clean cache
`$ sudo dnf clean all`
3. install openrazer again
```
$ sudo dnf install openrazer-meta
$ sudo dnf install kernel-devel
```
4. check currently installed python version:
$ python -V
>> Python 2.7.13

z3ntu, 2017-11-10:
Great :)