openrazer/openrazer #321: Ubuntu 17.04 Zesty

ViralAphinity, 2017-06-27:
I'm rather novice at Linux and also have Ubuntu 17.04, and could really use some help.

I've used the commands supplied on the website, but have been unable to run any of the other programs
(Razer Commander and polychromatic to be precise). Is there an easy-ish way to install the drivers/ppa for Ubuntu 17.04?

z3ntu, 2017-06-27:
* Please post the output of `sudo dkms status razer_chroma_driver/1.0.0`
* And of `python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"`

ViralAphinity, 2017-06-27:
The first line has no output but the second string sends
`Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named 'razer'`

z3ntu, 2017-06-27:
Output of `sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc` please

ViralAphinity, 2017-06-27:
`DKMS: install completed.
-e 
ADD YOUR USER TO THE PLUGDEV GROUP 'sudo gpasswd -a <USERNAME> plugdev'
Setting up python3-evdev (0.6.4-0ubuntu1) ...
Setting up liblapack3 (3.7.0-1) ...
update-alternatives: using /usr/lib/lapack/liblapack.so.3 to provide /usr/lib/liblapack.so.3 (liblapack.so.3) in auto mode
Setting up razer-daemon (1.1.13-0ubuntu1) ...
Setting up python3-numpy (1:1.12.1-1ubuntu1) ...
Setting up python3-razer (1.1.13-0ubuntu1) ...
`

z3ntu, 2017-06-27:
Now reboot and the output of `python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"`

z3ntu, 2017-06-27:
Don't forget to 

> ADD YOUR USER TO THE PLUGDEV GROUP 'sudo gpasswd -a <USERNAME> plugdev'

ViralAphinity, 2017-06-27:
So, for plugdev, I would run `sudo gpasswd -a <myname>'?

z3ntu, 2017-06-27:
No, `sudo gpasswd -a YOURUSERNAME plugdev`

z3ntu, 2017-06-27:
Also github doesn't like the smaller-than signs as it tries to interpret them as html. use codeblocks when using them or escape them with a backslash \\

ViralAphinity, 2017-06-27:
Output of `python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"` was `[<RazerKeyboard IO1535122712542>]`

z3ntu, 2017-06-27:
Should be alright then.

ViralAphinity, 2017-06-27:
Thanks for the quick help!

closed 2017-06-27 by z3ntu