openrazer/openrazer #322: Support Razer Lancehead / TE

lolrepeatlol, 2017-06-30:
Before I go out and buy myself a Lancehead, I'd like it to work on this. If it's possible, that'd be great! Thanks ✅

z3ntu, 2017-06-30:
We need the hardware connected to a Linux box before we can do anything. So go ahead and buy it and come back here when you have it :)

terrycain, 2017-06-30:
The lancehead is new therefore its highly likely that it wont take much effort to get working.

lolrepeatlol, 2017-06-30:
Perfect. Thanks!

omgimdrunk, 2017-07-24:
I have the lance head connected to a linux box right now, what do you all need?

z3ntu, 2017-07-24:
@omgimdrunk First the output of `lsusb | grep 1532`

Garrappachc, 2017-07-31:
@z3ntu I bought the Lancehead mouse and will do anything to get it working with this project.
```
~ lsusb | grep 1532
Bus 002 Device 014: ID 1532:0060 Razer USA, Ltd
```

z3ntu, 2017-07-31:
* Please post the output of
`lsusb -d 1532:0060 -v`
and
`ls -lah /dev/input/by-id/`
* Read [this](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol) because we probably need packet captures.
* Screenshots of all available options on Razer Synapse are needed.
* And lastly, the product & model number of the device for documentation reasons please.

Garrappachc, 2017-07-31:
[lsusb -d 1532:0060 -v](https://gist.github.com/Garrappachc/b771af7218f41b02fda1d63951947e8c)
[ls -lah /dev/input/by-id/](https://gist.github.com/Garrappachc/848b1a5ac954a205ab2356f1203d635f)
I'm onto screenshots rn, will upload in a few minutes.
Where can I find the product & model number? What do they look like?

Garrappachc, 2017-07-31:
[Synapse screenshots](http://imgur.com/a/3pY9Z)
Tell me whether you need any additional screenshots or not.

terrycain, 2017-07-31:
Packet captures:
* 1 capture of dpi going from min -> max with a few stops in between
* 1 capture of changing poll rate
* 1 capture of 
* 1 capture of 
* 1 capture of

terrycain, 2017-07-31:
Updated ^^

Garrappachc, 2017-07-31:
I've updated the imgur album. You can configure 4 sections separately. I'm on the packet captures.

terrycain, 2017-07-31:
Cool cool 
Also 1 packet capture starting on none (no effect) switching to custom (it'll start spamming packets) back to none again

Garrappachc, 2017-07-31:
Ok, first part of packets - all from the main synapse window, I'll upload chroma configurator packets in a while - a lot of combinations to be captured :)
https://transfer.sh/Hr1rx/lancehead-packet-captures.tar.gz

Garrappachc, 2017-07-31:
Also, you can store up to 4 custom profiles in the mouse and switch between them using one of the buttons. Do you want me to capture all of this profile-related packets as well?

terrycain, 2017-07-31:
Profiles are sort of in software so I'm not too worried about that yet. And as for the custom mode, you only need one configuration I can workout the rest

Garrappachc, 2017-07-31:
Packet capture from the chroma configurator is going to be tricky. You can setup presets for every single led separately, all of them at once, or bunch them. You can apply a number of different layers of effects, mark them as visible and so on. Any particular presets you want we to capture or just make as many combinations as I can?

terrycain, 2017-08-03:
It doesnt matter what you do in the configurator, all it will do is spam rgb values for every single led every time you change something and all i need is one packet of it setting the leds so i think just setting it to custom you'll see it spam packets in wireshark and that'll be enough

Garrappachc, 2017-08-03:
Ok, the custom mode was captured by me as well, it's in the package. Let me know if you need anything else to be captured.

terrycain, 2017-08-03:
Yep just been through and decoded all the pcaps and noticed custom was in there.
I'll need to make some more endpoints as your mouse is the first to have the leds on the sides groups as the left and right leds. But the majority of it should be a copy and paste job. Will have a crack at it tomorrow.

Mohd113, 2017-09-25:
Any update on this?

terrycain, 2017-09-25:
Have moved flat no internet yet. Haven't forgot about it.

Mohd113, 2017-09-25:
Awesome, thanks!

flexiondotorg, 2017-11-06:
I have the Lancehead TE and would be happy to test a branch or pull request with patch that enables support for this mouse :smile:

Ultrakrash, 2017-11-22:
I have one as well and would be more than willing. Just got the whole set of mercury edition stuff.

dustan9546, 2017-12-04:
Anyone still working on this?

LightNeZz, 2017-12-17:
I really want to have support for this mouse.

z3ntu, 2018-01-28:
PR in #460

fiercebrake, 2018-02-05:
In case it helps, attaching the results from teh queries done on CLI.

I also own a lancehead, so support for the thing  will be greatly appreciated.

Thanks!

[usb.txt](https://github.com/openrazer/openrazer/files/1696623/usb.txt)
[usbid.txt](https://github.com/openrazer/openrazer/files/1696624/usbid.txt)

fiercebrake, 2018-07-11:
Greetings to you all!

No production push yet ? thank you for your effort! is greatly appreciated!

nightsky30, 2018-07-19:
I recently bought one of these.  I read through all the linked issues.  Wasn't sure if the lighting was functional or not based on what I read.  But I am willing to test and provide captures if needed.  Thanks for what you guys have done :)

MelonCola, 2018-07-20:
How would I get this for Linux Mint?

MelonCola, 2018-07-20:
I was able to try it, my Lancehead TE shows up in Polychromatic, however choosing an effect does not work at all.

z3ntu, 2018-07-20:
Did you try all of the effects and none worked? Did you reboot after installing the new packages?

MelonCola, 2018-07-20:
Yes to both of those.

MelonCola, 2018-07-21:
How do I get wireshark for Linux?

z3ntu, 2018-07-21:
Ask Dr. Google please

nightsky30, 2018-07-21:
Trying to "git checkout feature_lancehead_te_0x0060" is reporting:

error: pathspec 'feature_lancehead_te_0x0060' did not match any file(s) known to git.

I see it listed in github's active branches, but it won't checkout the lancehead branch, and I don't see it in my list of branches when I "git branch -a".

~/Documents/openrazer$ git branch -a
* master
  remotes/origin/HEAD -> origin/master
  remotes/origin/cyanogen-async
  remotes/origin/feature_abyssus1800
  remotes/origin/feature_custom
  remotes/origin/feature_custom-old
  remotes/origin/feature_deathadder3500
  remotes/origin/feature_docs
  remotes/origin/feature_macro
  remotes/origin/feature_naga_epic_001F
  remotes/origin/feature_naga_epic_001F-old
  remotes/origin/feature_orochi_0x0056
  remotes/origin/feature_orochi_0x0056-old
  remotes/origin/feature_ouroboros0x0033
  remotes/origin/feature_ouroboros0x0033-old
  remotes/origin/fix_bladematrix
  remotes/origin/master
  remotes/origin/stable

I've tried recloning, fetching, updating, etc.  I'd like to test, but not sure what's going on here.

z3ntu, 2018-07-21:
Run `git fetch --all -p` and then `git checkout -b feature_lancehead_te_0x0060 origin/feature_lancehead_te_0x0060`

nightsky30, 2018-07-21:
Will this do??

[Lancehead TE.tar.gz](https://github.com/openrazer/openrazer/files/2216517/Lancehead.TE.tar.gz)

z3ntu, 2018-07-22:
@nightsky30 I meant a packet capture of changing the effects in Linux with openrazer, not of Synapse doing the same.

nightsky30, 2018-07-22:
@z3ntu Sorry, misread that.  Here ya go

[lancehead-te-polychromatic-effect-change-pcap.pcapng.tar.gz](https://github.com/openrazer/openrazer/files/2216993/lancehead-te-polychromatic-effect-change-pcap.pcapng.tar.gz)

z3ntu, 2018-07-22:
thanks! new commit is in the branch, please try again. Synapse always sends the same effect for the scroll wheel, logo, left side and right side LED. Currently there's only stuff for the scroll wheel in place, please tell me if everything's lighting up or only part of the mouse leds.

nightsky30, 2018-07-22:
Fetched, rebuilt, installed, rebooted, and there does not appear to be any change.  The whole mouse (every LED) remains in spectrum cycle :(

MelonCola, 2018-07-22:
I did everything, static, reactive, and spectrum seem to work for the scroll wheel.

nightsky30, 2018-07-22:
@MelonCola Cool Beans!  I'm seeing the same now.  I hadn't pulled properly.  Progress is good :)

dustan9546, 2018-07-22:
I'm kind of salty because you guys started working on the Lancehead compatibility right when I switched to Windows.

z3ntu, 2018-07-22:
¯\\\_(ツ)\_/¯

nightsky30, 2018-07-28:
Been tinkering with the code.

Seems the LOGO_LED and SCROLL_WHEEL_LED make a difference when used with razer_chroma_extended_matrix_effect_*, but BACKLIGHT_LED not so much.

Not sure how the left and right side LEDs are referenced, though I see a few values not used like 0x02, 0x06, 0x09, 0x0A, and 0x0B.  And perhaps the other keyboard specific LED values make a difference on the mouse.  Haven't tried those yet.

And not sure what the deal is with razer_chroma_**mouse**_extended_matrix_effect_* but that doesn't seem to play well with the Lancehead TE.

I have the breath mode working with the scroll wheel and got the logo to turn off.

Is the only way to send commands to multiple LEDs and sync them by creating multiple reports followed by multiple payload calls?

nightsky30, 2018-07-29:
Ahhh, after looking at how RazerGenie handles the different mouse LEDs vs Polychromatic, I understand the issue.  It boils down to the management tool.  Polychromatic doesn't handle the lighting in the same manner.  Maybe I should be using RazerGenie then.

nightsky30, 2018-07-29:
Ok, so this is what I see within RazerGenie :confused: 

[image]

nightsky30, 2018-07-29:
@lah7 Thank you so much!  I just created a quick py test file based on the openrazer/scripts/driver/mamba_chroma_wireless_mouse/test.py file.  I modified the standard section so that it tested scroll_matrix_effect and logo_matrix_effect, and everything seems well with the logo and scroll wheel with my modified driver files.

What I can't figure out is why the options aren't visible in either Polychromatic or RazerGenie.  I thought perhaps those things were determined by the files created when the device is attached.  So I modeled it after the values given to the DeathAdder Elite:

```
        case USB_DEVICE_ID_RAZER_LANCEHEAD_TE_WIRED:
			CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_poll_rate);
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_dpi);
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_logo_led_brightness);
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_logo_matrix_effect_spectrum);
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_logo_matrix_effect_wave);
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_logo_matrix_effect_reactive);
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_logo_matrix_effect_breath);
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_logo_matrix_effect_static);
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_logo_matrix_effect_none);
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_scroll_led_brightness);
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_scroll_matrix_effect_spectrum);
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_scroll_matrix_effect_wave);
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_scroll_matrix_effect_reactive);
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_scroll_matrix_effect_breath);
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_scroll_matrix_effect_static);
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_scroll_matrix_effect_none);
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_matrix_effect_custom);
            CREATE_DEVICE_FILE(&hdev->dev, &dev_attr_matrix_custom_frame);
            break;
```

I can see the files created:

```
'/sys/bus/hid/drivers/razermouse/0003:1532:0060.0002/':
total 0
drwxr-xr-x 5 root plugdev    0 Jul 29 11:57 .
drwxr-xr-x 5 root root       0 Jul 29 11:56 ..
-r--r--r-- 1 root plugdev 4096 Jul 29 11:57 country
-rw-rw---- 1 root plugdev 4096 Jul 29 11:57 device_mode
-r--r----- 1 root plugdev 4096 Jul 29 11:57 device_serial
-r--r----- 1 root plugdev 4096 Jul 29 11:57 device_type
-rw-rw---- 1 root plugdev 4096 Jul 29 12:41 dpi
lrwxrwxrwx 1 root plugdev    0 Jul 29 11:57 driver -> ../../../../../../../bus/hid/drivers/razermouse
-r--r----- 1 root plugdev 4096 Jul 29 11:57 firmware_version
drwxr-xr-x 3 root plugdev    0 Jul 29 11:57 hidraw
drwxr-xr-x 3 root plugdev    0 Jul 29 11:57 input
-rw-rw---- 1 root plugdev 4096 Jul 29 11:57 logo_led_brightness
--w--w---- 1 root plugdev 4096 Jul 29 12:40 logo_matrix_effect_breath
--w--w---- 1 root plugdev 4096 Jul 29 12:39 logo_matrix_effect_none
--w--w---- 1 root plugdev 4096 Jul 29 11:57 logo_matrix_effect_reactive
--w--w---- 1 root plugdev 4096 Jul 29 12:39 logo_matrix_effect_spectrum
--w--w---- 1 root plugdev 4096 Jul 29 12:39 logo_matrix_effect_static
--w--w---- 1 root plugdev 4096 Jul 29 12:40 logo_matrix_effect_wave
--w--w---- 1 root plugdev 4096 Jul 29 11:57 matrix_custom_frame
--w--w---- 1 root plugdev 4096 Jul 29 11:57 matrix_effect_custom
-r--r--r-- 1 root plugdev 4096 Jul 29 11:57 modalias
-rw-rw---- 1 root plugdev 4096 Jul 29 11:57 poll_rate
drwxr-xr-x 2 root plugdev    0 Jul 29 11:57 power
-r--r--r-- 1 root plugdev 4096 Jul 29 11:57 report_descriptor
-rw-rw---- 1 root plugdev 4096 Jul 29 11:57 scroll_led_brightness
--w--w---- 1 root plugdev 4096 Jul 29 12:39 scroll_matrix_effect_breath
--w--w---- 1 root plugdev 4096 Jul 29 12:38 scroll_matrix_effect_none
--w--w---- 1 root plugdev 4096 Jul 29 11:57 scroll_matrix_effect_reactive
--w--w---- 1 root plugdev 4096 Jul 29 12:38 scroll_matrix_effect_spectrum
--w--w---- 1 root plugdev 4096 Jul 29 12:38 scroll_matrix_effect_static
--w--w---- 1 root plugdev 4096 Jul 29 12:38 scroll_matrix_effect_wave
lrwxrwxrwx 1 root plugdev    0 Jul 29 11:57 subsystem -> ../../../../../../../bus/hid
--w--w---- 1 root plugdev 4096 Jul 29 11:57 test
-rw-r--r-- 1 root plugdev 4096 Jul 29 11:57 uevent
-r--r----- 1 root plugdev 4096 Jul 29 11:57 version
```

I must be missing something :confused:

nightsky30, 2018-07-30:
Everything seems to work minus:

- Breath Dual effect
- Side LEDs

[image]

I also found that the wave effect works for the scroll wheel and logo when I ran the test script.  Tried adding what was needed to show a wave effect for scroll and logo in GUI, but no luck yet.

I've never contributed code via github before, and I don't consider myself a coder in C/C++ or Python, but I'm willing to contribute what I've done through pull request if you would like :)

sankakucomplex, 2018-08-02:
Awesome nightsky30  ! 

Thank you all for  your contributions ! Much appreciated !

adwil13, 2018-08-27:
Are those changes going to master anytime soon?

adwil13, 2018-09-03:
Sorry for double post, but I have this mouse and I'm willing to do any kind of testing in order to get it to master branch - not being able to turn off those lights under linux is pretty annoying

z3ntu, 2018-09-03:
@adwil13 Actually no, because the left side and right side LEDs are not easily done with the architecture of openrazer and would require much copy-pasting in both openrazer and the frontends.
But I have something coming up, to test it, please message me on Telegram (@z3ntu), Matrix (@z3ntu:matrix.org), IRC (z3ntu), whatever you prefer.

MelonCola, 2018-09-16:
So is this sort of on pause? Or is it still being worked on?

DanyGee, 2018-10-04:
Guys, can someone confirm if onboard memory profiles together with simple color settings (not chroma advanced ones) can be stored with openrazer soft? I'm currently struggling with razer support about whether it officially can, or can't store lighting settings. All official information is just misleading.
[image]
(not a word about that it can't store lighting settings)
[image]
Support claims, it only stores DPI settings, not lighting. And I don't even wish for chroma config, those standard color settings would be enough.
I remember the same sh** with Blackwidow TE X Chroma (keyboard) which also, officially, didn't support storing lighting on the device itself, but with openrazer, the config stays on the device even after power off! 🥇

zakanybalazs, 2018-10-05:
Hi I'd likot to try out the razer_test with the razergenie razer_test branch, but I don't understand what I'm supposed to do. If I try to install the razergenie test branch the ninja install says that the <razer_test.h> is missing. Do I need to install the razer_test project somehow?
Thanks

[27 more replies]

commit 4ffe4147 (2019-07-29): Add Lancehead Tournament Edition support

Fixes #322
closed 2019-07-29 by z3ntu