openrazer/openrazer #324: Razer Daemon won't respond to any Razer controller program

gabagabagaba132, 2017-07-01:
After trying to use Polychromatic and Razercommander, both gave me the same results. They both reported that the Daemon is not responding. I tried rebooting and saw no results. I am using a Razer Anasi

z3ntu, 2017-07-01:
* What distro are you running?  
* Please post the output of `sudo dkms status razer_chroma_driver/1.0.0`
* And of `python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"`

gabagabagaba132, 2017-07-01:
Ubuntu 17.04
--razer_chroma_driver, 1.0.0, 4.10.0-26-generic, x86_64: installed
--"from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 175, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 361, in get_name_owner
    's', (bus_name,), **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.razer': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer/client/__init__.py", line 20, in __init__
    self._dbus = session_bus.get_object("org.razer", "/org/razer")
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.razer was not provided by any .service files

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/razer/client/__init__.py", line 22, in __init__
    raise DaemonNotFound("Could not connect to daemon")
razer.client.DaemonNotFound: Could not connect to daemon

z3ntu, 2017-07-01:
What's the output of `razer-daemon -Fv` ?

gabagabagaba132, 2017-07-01:
2017-07-01 12:18:10 | razer                          | INFO     | Initialising Daemon (v1.1.13). Pid: 3220
2017-07-01 12:18:10 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2012 --------------------- (1532:010D)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassic ------------------ (1532:011B)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowClassicAlternate --------- (1532:010E)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXUltimate ---------------- (1532:0217)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerBladeLate2016 ---------------------- (1532:0224)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerCore ------------------------------- (1532:0215)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerDeathStalkerChroma ----------------- (1532:0204)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderChroma ------------------- (1532:0043)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderElite -------------------- (1532:005C)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerDiamondbackChroma ------------------ (1532:004C)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerImperiator ------------------------- (1532:002F)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wired --------------------- (1532:0024)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerMamba2012Wireless ------------------ (1532:0025)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerNaga2014 --------------------------- (1532:0040)
[... cut ...]
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerTaipan ----------------------------- (1532:0034)
2017-07-01 12:18:10 | razer                          | DEBUG    | Loaded device specification: RazerTartarus --------------------------- (1532:0208)
2017-07-01 12:18:10 | razer                          | INFO     | Found device.0: 0003:1532:010F.0003
2017-07-01 12:18:10 | razer.device0                  | INFO     | Initialising device.0 RazerAnansi

z3ntu, 2017-07-01:
Which desktop environment are you using? Anyways, please upgrade to the latest version (released a few minutes ago). And run `systemctl --user enable --now razer-daemon && systemctl --user status razer-daemon`

gabagabagaba132, 2017-07-01:
I am using Cinnamon. I updated and using the command gave me this
Failed to enable unit: File razer-daemon.service: No such file or directory

z3ntu, 2017-07-01:
`ls -al /lib/systemd/user/razer-daemon.service`?

gabagabagaba132, 2017-07-01:
-al /lib/systemd/user/razer-daemon.service
ls: cannot access '/lib/systemd/user/razer-daemon.service': No such file or directory
I know I have it installed so I am pretty confused.

z3ntu, 2017-07-01:
Let me check.

z3ntu, 2017-07-01:
`dpkg -l | grep razer` ?

gabagabagaba132, 2017-07-01:
ii  python3-razer                                   1.1.13-0ubuntu1                             all          Razer Python Library
ii  razer-daemon                                    1.1.13-0ubuntu1                             amd64        Razer Device Daemon
ii  razer-doc                                       1.1.13-0ubuntu1                             all          Documentation and examples for the Razer Tools
ii  razer-kernel-modules-dkms                       1.1.13-0ubuntu1                             amd64        Razer Peripheral Drivers

z3ntu, 2017-07-01:
Not latest version :P 1.1.14 is the newest one. Try `apt update && apt upgrade`.

gabagabagaba132, 2017-07-01:
After that I am still stuck on 1.1.13
ii  python3-razer                                   1.1.13-0ubuntu1                             all          Razer Python Library
ii  razer-daemon                                    1.1.13-0ubuntu1                             amd64        Razer Device Daemon
ii  razer-doc                                       1.1.13-0ubuntu1                             all          Documentation and examples for the Razer Tools
ii  razer-kernel-modules-dkms                       1.1.13-0ubuntu1                             amd64        Razer Peripheral Drivers

gabagabagaba132, 2017-07-01:
I just pluggef in my Black Widow Ultimate 2016 and it works fine with the drivers so I know it's installed properly

z3ntu, 2017-07-01:
`lsusb | grep 1532` please

gabagabagaba132, 2017-07-01:
Bus 003 Device 007: ID 1532:010f Razer USA, Ltd

z3ntu, 2017-07-01:
What does `python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"` give now?

gabagabagaba132, 2017-07-01:
Nothing

z3ntu, 2017-07-01:
Weird. When you ran the daemon manually the line

> 2017-07-01 12:18:10 | razer.device0 | INFO | Initialising device.0 RazerAnansi

indicates, the daemon found the device... @terrycain care to jump in?

z3ntu, 2017-07-01:
`ls -al /sys/bus/hid/drivers/razerkbd/0*/`

gabagabagaba132, 2017-07-01:
total 0
drwxr-xr-x 5 root plugdev    0 Jul  1 14:40 .
drwxr-xr-x 5 root root       0 Jul  1 14:40 ..
-r--r--r-- 1 root plugdev 4096 Jul  1 14:40 country
lrwxrwxrwx 1 root plugdev    0 Jul  1 14:40 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 Jul  1 14:40 hidraw
drwxr-xr-x 3 root plugdev    0 Jul  1 14:40 input
-rw-rw---- 1 root plugdev 4096 Jul  1 14:40 key_alt_f4
-rw-rw---- 1 root plugdev 4096 Jul  1 14:40 key_alt_tab
-rw-rw---- 1 root plugdev 4096 Jul  1 14:40 key_super
-r--r--r-- 1 root plugdev 4096 Jul  1 14:40 modalias
drwxr-xr-x 2 root plugdev    0 Jul  1 14:40 power
-r--r--r-- 1 root plugdev 4096 Jul  1 14:40 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Jul  1 14:40 subsystem -> ../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Jul  1 14:40 uevent

/sys/bus/hid/drivers/razerkbd/0003:1532:010F.0017/:
total 0
drwxr-xr-x 5 root plugdev    0 Jul  1 14:40 .
drwxr-xr-x 5 root root       0 Jul  1 14:40 ..
-r--r--r-- 1 root plugdev 4096 Jul  1 14:40 country
lrwxrwxrwx 1 root plugdev    0 Jul  1 14:40 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 Jul  1 14:40 hidraw
drwxr-xr-x 3 root plugdev    0 Jul  1 14:40 input
-rw-rw---- 1 root plugdev 4096 Jul  1 14:40 key_alt_f4
-rw-rw---- 1 root plugdev 4096 Jul  1 14:40 key_alt_tab
-rw-rw---- 1 root plugdev 4096 Jul  1 14:40 key_super
-r--r--r-- 1 root plugdev 4096 Jul  1 14:40 modalias
drwxr-xr-x 2 root plugdev    0 Jul  1 14:40 power
-r--r--r-- 1 root plugdev 4096 Jul  1 14:40 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Jul  1 14:40 subsystem -> ../../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Jul  1 14:40 uevent

/sys/bus/hid/drivers/razerkbd/0003:1532:010F.0018/:
total 0
drwxr-xr-x 5 root plugdev    0 Jul  1 14:40 .
drwxr-xr-x 5 root root       0 Jul  1 14:40 ..
-r--r--r-- 1 root plugdev 4096 Jul  1 14:40 country
-rw-rw---- 1 root plugdev 4096 Jul  1 14:40 device_mode
-r--r----- 1 root plugdev 4096 Jul  1 14:40 device_serial
-r--r----- 1 root plugdev 4096 Jul  1 14:40 device_type
lrwxrwxrwx 1 root plugdev    0 Jul  1 14:40 driver -> ../../../../../../../../bus/hid/drivers/razerkbd
-r--r----- 1 root plugdev 4096 Jul  1 14:40 firmware_version
-rw-rw---- 1 root plugdev 4096 Jul  1 14:40 game_led_state
drwxr-xr-x 3 root plugdev    0 Jul  1 14:40 hidraw
drwxr-xr-x 3 root plugdev    0 Jul  1 14:40 input
-rw-rw---- 1 root plugdev 4096 Jul  1 14:40 macro_led_effect
-rw-rw---- 1 root plugdev 4096 Jul  1 14:40 macro_led_state
-rw-rw---- 1 root plugdev 4096 Jul  1 14:40 matrix_brightness
--w--w---- 1 root plugdev 4096 Jul  1 14:40 matrix_effect_none
--w--w---- 1 root plugdev 4096 Jul  1 14:40 matrix_effect_spectrum
--w--w---- 1 root plugdev 4096 Jul  1 14:40 matrix_effect_static
-r--r--r-- 1 root plugdev 4096 Jul  1 14:40 modalias
drwxr-xr-x 2 root plugdev    0 Jul  1 14:40 power
-r--r--r-- 1 root plugdev 4096 Jul  1 14:40 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Jul  1 14:40 subsystem -> ../../../../../../../../bus/hid
-rw-rw---- 1 root plugdev 4096 Jul  1 14:40 test
-rw-r--r-- 1 root plugdev 4096 Jul  1 14:40 uevent
-r--r----- 1 root plugdev 4096 Jul  1 14:40 version

closed 2018-02-28 by stale[bot]