openrazer/openrazer #327: 14.04 LTS

Acterax, 2017-07-03:
I can t start the program 
It say Unable to find Razer Copperhead Laser Mouse

z3ntu, 2017-07-03:
We don't support Ubuntu 14.04 anymore. Please upgrade to a more recent version (eg Ubuntu 16.04 LTS).

Acterax, 2017-07-03:
I updated my system and I have the same error

z3ntu, 2017-07-03:
Are you running 16.04 now?
Please post the output of `sudo dkms status razer_chroma_driver/1.0.0` and of `razer-daemon -Fv`

Acterax, 2017-07-03:
tom@Tom-HP:~$ sudo dkms status razer_chroma_driver/1.0.0
razer_chroma_driver, 1.0.0, 3.19.0-58-generic, x86_64: installed
razer_chroma_driver, 1.0.0, 4.4.0-83-generic, x86_64: installed

tom@Tom-HP:~$ razer-daemon -Fv
razer-daemon : commande introuvable

Acterax, 2017-07-03:
And yes I'am running on 16.04 I just updated my system

z3ntu, 2017-07-03:
Did you install all packages as listed on the website? http://terrycain.github.io/razer-drivers/#ubuntu

Acterax, 2017-07-03:
Yes

z3ntu, 2017-07-03:
Your Ubuntu installation is still using the ppa from 14.04. You should be able to edit a file (probably in /etc/apt/sources.list.d) and change trusty to xenial. Then run `sudo apt update && sudo apt upgrade`

Acterax, 2017-07-03:
What file?

z3ntu, 2017-07-03:
Basically follow https://askubuntu.com/a/293210 but change our ppa (terrz/razerutils) and change it from trusty to xenial.

terrycain, 2017-07-03:
@Acterax I believe there is a file /etc/apt/sources.list.d/terrz-ubuntu-razerutils-trusty.list, you probably want to delete that, re-install ppa, update

Acterax, 2017-07-03:
Yes i will do that but after I' am following this because all my ppa are ...
https://askubuntu.com/questions/124017/how-do-i-restore-the-default-repositories

Acterax, 2017-07-03:
I have an username in 2 words
ADD YOUR USER TO THE PLUGDEV GROUP 'sudo gpasswd -a <USERNAME> plugdev'

z3ntu, 2017-07-03:
Output of `id -u -n`?

Acterax, 2017-07-03:
Yes ....

Acterax, 2017-07-03:
and know I print 

"razertool" ?

Acterax, 2017-07-03:
Doesn't work but I see this 

URL : http://ppa.launchpad.net/upubuntu-com/devel/ubuntu
xenial
main
Disabled for upgrade to xenial

Acterax, 2017-07-03:
Doesn't work

tom@Tom-HP:/etc/apt/sources.list.d$ razer-daemon -Fv
Daemon already exists. Please stop that one.

terrycain, 2017-07-03:
output of `ls -lah /etc/apt/sources.list.d`

Acterax, 2017-07-03:
tom@Tom-HP:/etc/apt/sources.list.d$ output of ls -lah /etc/apt/sources.list.d

Yagi-Uda antenna analysi program output, version 1.19
Written by Dr. David Kirkby Ph.D. G8WRB (email:david.kirkby@onetel.net)

USAGE: output [-cehps] [-E E_max -H H_max -r min -R max -Z Zo] filename 

Where options are:

 -c  Calculate sidelobe levels (slows program considerably).
 -e  Suppress calculation of 3dB E-plane BW.
 -h  Suppress calculation of 3dB H-plane BW.
 -p  Put data into filename.freq, filename.glog and filename.glin for gnuplot
 -s  Suppress diagnostic output.
 -E  Max angle to find the 3dB point. Min=90, max=180  (default = 179 degrees)
 -H  Max angle to find the 3dB point. Min=0, max=90    (default = 60 degrees)
 -r  Set minimum range on the radial gnuplot log graph (default = -50 dB)
 -R  Set maximum range on the radial gnuplot lin graph (default = 20 dB)
 -Z  Set characteristic impedance                      (default = 50 Ohms)

Acterax, 2017-07-03:
tom@Tom-HP:/etc/apt/sources.list.d$ ls -lah /etc/apt/sources.list.d
total 124K
drwxr-xr-x 2 root root 4,0K juil.  3 16:56 .
drwxr-xr-x 6 root root 4,0K juil.  3 16:44 ..
-rw-r--r-- 1 root root  240 juil.  3 17:01 google-chrome.list
-rw-r--r-- 1 root root  189 juil.  3 14:34 google-chrome.list.distUpgrade
-rw-r--r-- 1 root root  240 juil.  3 17:01 google-chrome.list.save
-rw-r--r-- 1 root root  183 juil.  3 17:01 mumble-release-trusty.list
-rw-r--r-- 1 root root  132 juil.  3 14:34 mumble-release-trusty.list.distUpgrade
-rw-r--r-- 1 root root  183 juil.  3 17:01 mumble-release-trusty.list.save
-rw-r--r-- 1 root root  101 juil.  3 17:01 spotify.list
-rw-r--r-- 1 root root   50 juil.  3 14:34 spotify.list.distUpgrade
-rw-r--r-- 1 root root  101 juil.  3 17:01 spotify.list.save
-rw-r--r-- 1 root root  187 juil.  3 17:01 synapse-core-ppa-trusty.list
-rw-r--r-- 1 root root  136 juil.  3 14:34 synapse-core-ppa-trusty.list.distUpgrade
-rw-r--r-- 1 root root  187 juil.  3 17:01 synapse-core-ppa-trusty.list.save
-rw-r--r-- 1 root root  181 juil.  3 17:01 team-xbmc-ppa-trusty.list
-rw-r--r-- 1 root root  130 juil.  3 14:34 team-xbmc-ppa-trusty.list.distUpgrade
-rw-r--r-- 1 root root  181 juil.  3 17:01 team-xbmc-ppa-trusty.list.save
-rw-r--r-- 1 root root  138 juil.  3 17:01 terrz-razerutils-trusty.list
-rw-r--r-- 1 root root  136 juil.  3 14:34 terrz-razerutils-trusty.list.distUpgrade
-rw-r--r-- 1 root root  187 juil.  3 17:01 terrz-razerutils-trusty.list.save
-rw-r--r-- 1 root root   71 juil.  3 17:01 terrz-ubuntu-razerutils-xenial.list
-rw-r--r-- 1 root root   71 juil.  3 17:01 terrz-ubuntu-razerutils-xenial.list.save
-rw-r--r-- 1 root root  215 juil.  3 17:01 upubuntu-com-devel-trusty.list
-rw-r--r-- 1 root root  213 juil.  3 14:34 upubuntu-com-devel-trusty.list.distUpgrade
-rw-r--r-- 1 root root  215 juil.  3 17:01 upubuntu-com-devel-trusty.list.save
-rw-r--r-- 1 root root  270 juil.  3 17:01 webupd8team-brackets-trusty.list
-rw-r--r-- 1 root root  219 juil.  3 14:34 webupd8team-brackets-trusty.list.distUpgrade
-rw-r--r-- 1 root root  270 juil.  3 17:01 webupd8team-brackets-trusty.list.save
-rw-r--r-- 1 root root  329 juil.  3 17:01 webupd8team-java-trusty.list
-rw-r--r-- 1 root root  278 juil.  3 14:34 webupd8team-java-trusty.list.distUpgrade
-rw-r--r-- 1 root root  329 juil.  3 17:01 webupd8team-java-trusty.list.save

terrycain, 2017-07-03:
```
sudo rm -f /etc/apt/sources.list.d/terrz-razerutils-trusty.list /etc/apt/sources.list.d/terrz-razerutils-trusty.list.distUpgrade /etc/apt/sources.list.d/terrz-ubuntu-razerutils-xenial.list.save
apt-get update
apt-get upgrade -y
dpkg -l razer-kernel-modules-dkms
```
Post output of all that

Acterax, 2017-07-03:
tom@Tom-HP:/etc/apt/sources.list.d$ sudo rm -f /etc/apt/sources.list.d/terrz-razerutils-trusty.list /etc/apt/sources.list.d/terrz-razerutils-trusty.list.distUpgrade /etc/apt/sources.list.d/terrz-ubuntu-razerutils-xenial.list.save
tom@Tom-HP:/etc/apt/sources.list.d$ sudo apt-get update
Atteint:1 http://fr.archive.ubuntu.com/ubuntu xenial InRelease
Réception de:2 http://security.ubuntu.com/ubuntu xenial-security InRelease [102 kB]             
Atteint:3 http://fr.archive.ubuntu.com/ubuntu xenial-updates InRelease                          
102 ko réceptionnés en 0s (142 ko/s)                                                          
Lecture des listes de paquets... Fait
tom@Tom-HP:/etc/apt/sources.list.d$ sudo apt-get upgrade -y
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
Calcul de la mise à jour... Fait
Les paquets suivants ont été installés automatiquement et ne sont plus nécessaires :
  python-aptdaemon python-aptdaemon.gtk3widgets python-debian python-defer
Veuillez utiliser « sudo apt autoremove » pour les supprimer.
0 mis à jour, 0 nouvellement installés, 0 à enlever et 0 non mis à jour.
tom@Tom-HP:/etc/apt/sources.list.d$ sudo dpkg -l razer-kernel-modules-dkms
Souhait=inconnU/Installé/suppRimé/Purgé/H=à garder
| État=Non/Installé/fichier-Config/dépaqUeté/échec-conFig/H=semi-installé/W=attend-traitement-déclenchements
|/ Err?=(aucune)/besoin Réinstallation (État,Err: majuscule=mauvais)
||/ Nom                                           Version                     Architecture                Description
+++-=============================================-===========================-===========================-===============================================================================================
ii  razer-kernel-modules-dkms                     1.1.14-0ubuntu1             amd64                       Razer Peripheral Drivers
tom@Tom-HP:/etc/apt/sources.list.d$

terrycain, 2017-07-03:
reboot, then paste output of `id`

Acterax, 2017-07-03:
reboot my computer ?

terrycain, 2017-07-03:
@Acterax yep

Acterax, 2017-07-03:
tom@Tom-HP:~$ id
uid=1000(tom) gid=1000(tom) groupes=1000(tom),4(adm),24(cdrom),27(sudo),30(dip),46(plugdev),108(lpadmin),124(sambashare)

z3ntu, 2017-07-03:
`systemctl --user enable --now razer-daemon` and `systemctl --user status razer-daemon`

Acterax, 2017-07-03:
tom@Tom-HP:~$ systemctl --user enable --now razer-daemon and systemctl --user status razer-daemon
Failed to execute operation: No such file or directory

Acterax, 2017-07-03:
I'am really idiot ...

Acterax, 2017-07-03:
tom@Tom-HP:~$ systemctl --user enable --now razer-daemon
Created symlink from /home/tom/.config/systemd/user/default.target.wants/razer-daemon.service to /usr/lib/systemd/user/razer-daemon.service.
Failed to start razer-daemon.service: Unit dbus.socket not found.

tom@Tom-HP:~$ systemctl --user status razer-daemon
● razer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/razer-daemon.service; enabled; vendor p
   Active: inactive (dead)
     Docs: man:razer-daemon(8)

terrycain, 2017-07-03:
output of `dpkg -l dbus`

Acterax, 2017-07-03:
Souhait=inconnU/Installé/suppRimé/Purgé/H=à garder
| État=Non/Installé/fichier-Config/dépaqUeté/échec-conFig/H=semi-installé/W=attend-traitement-déclenchements
|/ Err?=(aucune)/besoin Réinstallation (État,Err: majuscule=mauvais)
||/ Nom            Version      Architecture Description
+++-==============-============-============-=================================
ii  dbus           1.10.6-1ubun amd64        simple interprocess messaging sys

terrycain, 2017-07-03:
Try
```
sudo systemctl daemon-reload
systemctl --user start razer-daemon
systemctl --user status razer-daemon
```

Acterax, 2017-07-03:
tom@Tom-HP:~$ systemctl --user start razer-daemon
Failed to start razer-daemon.service: Unit dbus.socket not found.
tom@Tom-HP:~$ systemctl --user start razer-daemon
Failed to start razer-daemon.service: Unit dbus.socket not found.
tom@Tom-HP:~$ systemctl --user status razer-daemon
● razer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/razer-daemon.service; enabled; vendor p
   Active: inactive (dead)
     Docs: man:razer-daemon(8)
lines 1-4/4 (END)

terrycain, 2017-07-03:
`sudo systemctl status dbus.socket`

Acterax, 2017-07-03:
tom@Tom-HP:~$ sudo systemctl status dbus.socket
● dbus.socket - D-Bus System Message Bus Socket
   Loaded: loaded (/lib/systemd/system/dbus.socket; static; vendor preset: enabled)
   Active: active (running) since lun. 2017-07-03 17:19:36 CEST; 26min ago
   Listen: /var/run/dbus/system_bus_socket (Stream)

juil. 03 17:19:36 Tom-HP systemd[1]: Listening on D-Bus System Message Bus Socket.

Acterax, 2017-07-03:
Up ?

z3ntu, 2017-07-03:
`systemctl --user status dbus.socket`

Acterax, 2017-07-03:
tom@Tom-HP:~$ systemctl --user status dbus.socket
● dbus.socket
   Loaded: not-found (Reason: No such file or directory)
   Active: inactive (dead)

z3ntu, 2017-07-03:
```
sudo apt install dbus-user-session
systemctl --user daemon-reload
systemctl --user start razer-daemon
systemctl --user status razer-daemon
```

Acterax, 2017-07-03:
tom@Tom-HP:~$ sudo apt install dbus-user-session
[sudo] Mot de passe de tom : 
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
Les paquets suivants ont été installés automatiquement et ne sont plus nécessaires :
  python-aptdaemon python-aptdaemon.gtk3widgets python-debian python-defer
Veuillez utiliser « sudo apt autoremove » pour les supprimer.
Les NOUVEAUX paquets suivants seront installés :
  dbus-user-session
0 mis à jour, 1 nouvellement installés, 0 à enlever et 0 non mis à jour.
Il est nécessaire de prendre 9 294 o dans les archives.
Après cette opération, 115 ko d'espace disque supplémentaires seront utilisés.
Réception de:1 http://fr.archive.ubuntu.com/ubuntu xenial-updates/universe amd64 dbus-user-session all 1.10.6-1ubuntu3.3 [9 294 B]
9 294 o réceptionnés en 0s (58,6 ko/s)       
Sélection du paquet dbus-user-session précédemment désélectionné.
(Lecture de la base de données... 249754 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de .../dbus-user-session_1.10.6-1ubuntu3.3_all.deb ...
Dépaquetage de dbus-user-session (1.10.6-1ubuntu3.3) ...
Paramétrage de dbus-user-session (1.10.6-1ubuntu3.3) ...
tom@Tom-HP:~$ systemctl --user daemon-reload
tom@Tom-HP:~$ systemctl --user start razer-daemon
tom@Tom-HP:~$ systemctl --user status razer-daemon
● razer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/razer-daemon.service; enabled; vendor preset: 
   Active: active (running) since lun. 2017-07-03 19:23:14 CEST; 4s ago
     Docs: man:razer-daemon(8)
 Main PID: 4829 (razer-daemon)
   CGroup: /user.slice/user-1000.slice/user@1000.service/razer-daemon.service
           └─4829 razer-daemon                     

juil. 03 19:23:14 Tom-HP razer-daemon[4829]: 2017-07-03 19:23:14 | razer.device1.batter
juil. 03 19:23:14 Tom-HP razer-daemon[4829]: 2017-07-03 19:23:14 | razer               
juil. 03 19:23:14 Tom-HP razer-daemon[4829]: 2017-07-03 19:23:14 | razer               
juil. 03 19:23:14 Tom-HP razer-daemon[4829]: 2017-07-03 19:23:14 | razer               
juil. 03 19:23:14 Tom-HP razer-daemon[4829]: 2017-07-03 19:23:14 | razer               
juil. 03 19:23:14 Tom-HP razer-daemon[4829]: 2017-07-03 19:23:14 | razer               
juil. 03 19:23:14 Tom-HP razer-daemon[4829]: 2017-07-03 19:23:14 | razer               
juil. 03 19:23:14 Tom-HP razer-daemon[4829]: 2017-07-03 19:23:14 | razer               
juil. 03 19:23:14 Tom-HP razer-daemon[4829]: 2017-07-03 19:23:14 | razer               
juil. 03 19:23:14 Tom-HP razer-daemon[4829]: 2017-07-03 19:23:14 | razer.device1       
lines 1-18/18 (END)

z3ntu, 2017-07-03:
`python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"`

Acterax, 2017-07-03:
tom@Tom-HP:~$ python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 175, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 361, in get_name_owner
    's', (bus_name,), **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.razer': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer/client/__init__.py", line 20, in __init__
    self._dbus = session_bus.get_object("org.razer", "/org/razer")
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/razer/client/__init__.py", line 22, in __init__
    raise DaemonNotFound("Could not connect to daemon")
razer.client.DaemonNotFound: Could not connect to daemon

z3ntu, 2017-07-03:
Can you reboot and run that command again?

z3ntu, 2017-07-03:
It seems Ubuntu has changed some things between Ubuntu 16.04 and 17.04. On 17.04 dbus-user-session is pre-installed (which it isn't on 16.04) and more things changed internally probably... Right now I can't give you a solution (except using 17.04 :stuck_out_tongue_winking_eye: )

Acterax, 2017-07-03:
If use 17.04 that will work ?

z3ntu, 2017-07-03:
In my VM it works and at least on the computer of @terrycain . I've tried the dbus stuff in a 16.04 VM and hit the same issue (and didn't find a solution yet) but on 17.04 it works as expected.

Acterax, 2017-07-03:
And you advise me to go in 17.04 or wait for an update ?

Acterax, 2017-07-03:
@z3ntu And you advise me to go in 17.04 or wait for an update ?

z3ntu, 2017-07-03:
I can't promise I'll find a solution for 16.04 anytime soon so if you need the driver right now, yes, installing 17.04 should solve it.

Acterax, 2017-07-03:
@z3ntu I'am downloading ubuntu 17.04 iso

Acterax, 2017-07-03:
I have no razertool

Instalation log:

tom@Tom:~$ sudo add-apt-repository ppa:terrz/razerutils
 This PPA contains open source drivers that allow many Razer peripherals to utilise their full feature set and allow customisation via the daemon.

Supported devices include:

 - BlackWidow (Chroma, Classic, Ultimate)
 - BlackWidow Mamba
 - Blade Stealth
 - Firefly
 - Mamba
 - Abyssus
 - Tartarus

For more information, see here: https://github.com/terrycain/razer-drivers

Licensed under the GPL. These drivers are not affiliated with Razer, Inc.
 Plus d’info : https://launchpad.net/~terrz/+archive/ubuntu/razerutils
Appuyez sur [ENTRÉE] pour continuer ou Ctrl-C pour annuler l’ajout

gpg: le trousseau local « /tmp/tmpjpyzv7gv/pubring.gpg » a été créé
gpg: /tmp/tmpjpyzv7gv/trustdb.gpg : base de confiance créée
gpg: clef FDB22788DC9CC596 : clef publique « Launchpad PPA for Terry Cain » importée
gpg: Quantité totale traitée : 1
gpg:               importées : 1
OK
tom@Tom:~$ sudo apt-get update
Atteint:1 http://fr.archive.ubuntu.com/ubuntu zesty InRelease
Réception de:2 http://security.ubuntu.com/ubuntu zesty-security InRelease [89,2 kB]
Atteint:3 http://fr.archive.ubuntu.com/ubuntu zesty-updates InRelease          
Atteint:4 http://fr.archive.ubuntu.com/ubuntu zesty-backports InRelease        
Réception de:5 http://ppa.launchpad.net/terrz/razerutils/ubuntu zesty InRelease [15,9 kB]
Réception de:6 http://ppa.launchpad.net/terrz/razerutils/ubuntu zesty/main i386 Packages [1 384 B]
Réception de:7 http://ppa.launchpad.net/terrz/razerutils/ubuntu zesty/main amd64 Packages [1 392 B]
Réception de:8 http://ppa.launchpad.net/terrz/razerutils/ubuntu zesty/main Translation-en [844 B]
109 ko réceptionnés en 1s (55,9 ko/s)   
Lecture des listes de paquets... Fait
tom@Tom:~$ sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances       
Lecture des informations d'état... Fait
The following additional packages will be installed:
  dkms libblas-common libblas3 libgfortran3 liblapack3 libxdo3 python3-evdev
  python3-notify2 python3-numpy python3-pyudev python3-setproctitle
  xautomation xdotool
Paquets suggérés :
  menu gfortran python-numpy-doc python3-dev python3-nose python3-numpy-dbg
Les NOUVEAUX paquets suivants seront installés :
  dkms libblas-common libblas3 libgfortran3 liblapack3 libxdo3 python3-evdev
  python3-notify2 python3-numpy python3-pyudev python3-razer
  python3-setproctitle razer-daemon razer-doc razer-kernel-modules-dkms
  xautomation xdotool
0 mis à jour, 17 nouvellement installés, 0 à enlever et 138 non mis à jour.
Il est nécessaire de prendre 4 843 ko dans les archives.
Après cette opération, 21,7 Mo d'espace disque supplémentaires seront utilisés.
Souhaitez-vous continuer ? [O/n] y
Réception de:1 http://fr.archive.ubuntu.com/ubuntu zesty-updates/main amd64 dkms all 2.3-3ubuntu1 [68,4 kB]
Réception de:2 http://ppa.launchpad.net/terrz/razerutils/ubuntu zesty/main amd64 python3-evdev amd64 0.6.4-0ubuntu1 [30,2 kB]
Réception de:3 http://fr.archive.ubuntu.com/ubuntu zesty/main amd64 libblas-common amd64 3.7.0-1 [5 696 B]
Réception de:4 http://fr.archive.ubuntu.com/ubuntu zesty/main amd64 libblas3 amd64 3.7.0-1 [144 kB]
Réception de:5 http://ppa.launchpad.net/terrz/razerutils/ubuntu zesty/main amd64 python3-razer all 1.1.14-0ubuntu1 [18,2 kB]
Réception de:6 http://fr.archive.ubuntu.com/ubuntu zesty/main amd64 libgfortran3 amd64 6.3.0-12ubuntu2 [270 kB]
Réception de:7 http://ppa.launchpad.net/terrz/razerutils/ubuntu zesty/main amd64 razer-kernel-modules-dkms amd64 1.1.14-0ubuntu1 [40,4 kB]
Réception de:8 http://fr.archive.ubuntu.com/ubuntu zesty/main amd64 liblapack3 amd64 3.7.0-1 [2 129 kB]
Réception de:9 http://ppa.launchpad.net/terrz/razerutils/ubuntu zesty/main amd64 razer-daemon amd64 1.1.14-0ubuntu1 [46,9 kB]
Réception de:10 http://ppa.launchpad.net/terrz/razerutils/ubuntu zesty/main amd64 razer-doc all 1.1.14-0ubuntu1 [5 856 B]
Réception de:11 http://fr.archive.ubuntu.com/ubuntu zesty/universe amd64 python3-notify2 all 0.3-3 [10,6 kB]
[... 114 lines cut ...]
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.10.0-19-generic/updates/dkms/

depmod.....

DKMS: install completed.
-e 
ADD YOUR USER TO THE PLUGDEV GROUP 'sudo gpasswd -a <USERNAME> plugdev'
Paramétrage de python3-evdev (0.6.4-0ubuntu1) ...
Paramétrage de liblapack3 (3.7.0-1) ...
update-alternatives: utilisation de « /usr/lib/lapack/liblapack.so.3 » pour fournir « /usr/lib/liblapack.so.3 » (liblapack.so.3) en mode automatique
Paramétrage de xdotool (1:3.20160805.1-3) ...
Paramétrage de razer-daemon (1.1.14-0ubuntu1) ...
Paramétrage de python3-numpy (1:1.12.1-1ubuntu1) ...
Paramétrage de python3-razer (1.1.14-0ubuntu1) ...
Traitement des actions différées (« triggers ») pour libc-bin (2.24-9ubuntu2) ...
tom@Tom:~$ sudo gpasswd -a tom plugdev
Ajout de l'utilisateur tom au groupe plugdev
tom@Tom:~$ razertool
razertool : commande introuvable
tom@Tom:~$ razer-daemon 
tom@Tom:~$

Acterax, 2017-07-03:
Please @z3ntu or @terrycain

z3ntu, 2017-07-04:
I don't know why you always want to run razertool as it's not an executable we provide. 
Please run `systemctl --user start razer-daemon.service; systemctl --user status razer-daemon.service`

Acterax, 2017-07-04:
tom@Tom:~$ systemctl --user start razer-daemon.service; systemctl --user status razer-daemon.service
● razer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/usr/lib/systemd/user/razer-daemon.service; disabled; vendor 
   Active: active (running) since Tue 2017-07-04 10:59:09 CEST; 10ms ago
     Docs: man:razer-daemon(8)
 Main PID: 2713 (razer-daemon)
   CGroup: /user.slice/user-1000.slice/user@1000.service/razer-daemon.service
           └─2713 razer-daemon

juil. 04 10:59:09 Tom systemd[1796]: Starting Daemon to manage razer devices in 
juil. 04 10:59:09 Tom systemd[1796]: Started Daemon to manage razer devices in u
lines 1-10/10 (END)

Acterax, 2017-07-04:
And a notification appeared indicating the battery of my mouse

z3ntu, 2017-07-04:
`python3 -c "from razer.client import DeviceManager; a = DeviceManager(); print(a.devices)"`
If that commands returns your device, everything is working great :)

Acterax, 2017-07-04:
[<RazerKeyboard PM1423080400224>, <RazerMouse PM002153868345>]
And next ? to launch the frame ?

z3ntu, 2017-07-04:
What do you mean with "launch the frame"?

[5 more replies]

closed 2017-07-04 by z3ntu