openrazer/openrazer #328: Firefly reactive mode not working

wico, 2017-07-03:
Hi,

I really hope I have not overseen anything - but:

Does the reactive mode for razer's firefly work on linux? I tried to enable it but the firefly stays off as soon as I activate it via polychromatic (and of course by clicking the mouse, a razer mamba tournament edition, which normally should trigger some reaction on the firefly). All other modes work fine btw.

My setup:
ii  python3-razer                                   1.1.14-0ubuntu1                             all          Razer Python Library
ii  razer-daemon                                    1.1.14-0ubuntu1                             amd64        Razer Device Daemon
ii  razer-kernel-modules-dkms                       1.1.14-0ubuntu1                             amd64        Razer Peripheral Drivers
ii  polychromatic                                   0.3.9.3-1-zesty                             all          Graphical front-end for managing Razer peripherals.

Linux phoenix 4.11.8-SYSTEM4 #1 SMP PREEMPT Mon Jul 3 10:44:56 CEST 2017 x86_64 x86_64 x86_64 GNU/Linux

Base distro: Ubuntu 17.04

Update:
Some more investigations: Booting a (live-USB) windows 10 with razer-stuff installed: reactive mode works fine. So technically it can work. It just stays black on linux.

Does anyone has a clue why? :)

Thank you for an answer (if any),
Martin

z3ntu, 2017-07-03:
Is anything in the dmseg? (command: `dmesg | grep razer`). I have a Firefly at home so I can test it later today.

wico, 2017-07-03:
Checked right now:

`[    3.315362] razermouse 0003:1532:0046.0004: input,hidraw0: USB HID v1.11 Mouse [Razer Razer Mamba Tournament Edition] on usb-0000:00:14.0-2/input0
[    3.419271] razerkbd 0003:1532:021E.0001: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer Ornata Chroma] on usb-0000:00:14.0-1/input0
[    3.471546] razermouse 0003:1532:0046.0005: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer Mamba Tournament Edition] on usb-0000:00:14.0-2/input1
[    3.471614] razerkbd 0003:1532:021E.0003: input,hidraw3: USB HID v1.11 Mouse [Razer Razer Ornata Chroma] on usb-0000:00:14.0-1/input2
[    3.575497] razerkbd 0003:1532:021E.0002: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer Ornata Chroma] on usb-0000:00:14.0-1/input1
[    3.627566] razermouse 0003:1532:0046.0006: input,hidraw5: USB HID v1.11 Keyboard [Razer Razer Mamba Tournament Edition] on usb-0000:00:14.0-2/input2
[    4.140636] razerfirefly 0003:1532:0C00.000A: input,hidraw7: USB HID v1.11 Keyboard [Razer Razer Firefly] on usb-0000:00:14.0-10.2/input2
[    4.192564] razerfirefly 0003:1532:0C00.0009: input,hidraw8: USB HID v1.11 Keyboard [Razer Razer Firefly] on usb-0000:00:14.0-10.2/input1
[    4.192744] razerfirefly 0003:1532:0C00.0008: input,hidraw9: USB HID v1.11 Mouse [Razer Razer Firefly] on usb-0000:00:14.0-10.2/input0
[ 9767.567060] Modules linked in: arc4 ecb md4 hmac nls_utf8 cifs ccm fscache ufs qnx4 hfsplus hfs minix ntfs jfs xfs dm_mod bnep ipt_MASQUERADE nf_nat_masquerade_ipv4 nf_conntrack_netlink nfnetlink xfrm_user xfrm_algo iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 xt_addrtype xt_conntrack nf_nat nf_conntrack br_netfilter bridge overlay xt_multiport iptable_filter vmnet(O) fuse vmw_vsock_vmci_transport vsock vmw_vmci vmmon(O) 8021q garp mrp stp llc razerfirefly(O) joydev input_leds mousedev pci_stub vboxpci(O) vboxnetadp(O) vboxnetflt(O) razermouse(O) razerkbd(O) snd_emu10k1_synth snd_emux_synth snd_seq_midi_emul snd_seq_virmidi snd_emu10k1 snd_util_mem snd_ac97_codec snd_hwdep ac97_bus snd_pcm_oss snd_mixer_oss snd_pcm cpufreq_userspace cpufreq_powersave cpufreq_conservative vboxdrv(O) snd_seq_dummy`

Nothing happens (gets logged) when enabling/disabling reactive mode.

z3ntu, 2017-07-03:
Then it's probably an issue in the daemon. Will get to you later today hopefully.

terrycain, 2017-07-03:
technically yes the reactive mode on the firefly works hence it goes off. The part I havent implemented is poking the firefly to do the reactive flash when mouse clicks.

wico, 2017-07-03:
Oha, so, when it goes off in reactive mode - what causes a "wake up" then?

terrycain, 2017-07-03:
nothing yet, eventually I'll code in the on mouse click to flash

wico, 2017-07-03:
Got it. :) Otherwise I'll connect the firefly's color to some cpu temp. But if you could add the click->flash (same like the windows-drivers do), that would be awesome!

terrycain, 2017-07-03:
Dont worry, its on my todo list

wico, 2017-07-03:
My finger is over the donate button :)

terrycain, 2017-07-03:
Haha

wico, 2017-07-03:
I know its unrelated to the issue - but I really want to donate. How to do it (without paypal)?

z3ntu, 2017-07-03:
I take Bitcoin at `1FEgkNbqNdjyrCk4FQqMup9mBj1hPkA9RL` :smile:

z3ntu, 2017-07-03:
Also, what method of payment do you suggest? I am not aware of popular alternatives to PayPal.

wico, 2017-07-03:
There is an oldschool way, bank-transfer. :) At least withing the EU its just a matter of having the IBAN.

terrycain, 2017-07-03:
you have an email I can send you the iban?

wico, 2017-07-03:
martin@system4.org

wico, 2017-07-03:
done ;)

wico, 2018-01-30:
Mhhh ... I actually tried openrazer 2.2.0~ubuntu18.04.1 and the issue (aka missing feature ;) still exists.

lezed1, 2018-01-30:
@wico can you try setting your Firefly to reactive again. It should not light up on mouse presses since it only lights up when a trigger is sent. You can manually send that trigger by using the dbus interface of the daemon (I used d-feet for this, session bus tab, look for razer, find the firefly device and call the reactive trigger function).

I had a Python script that uses pynput to send this trigger on every mouse click, but I can't find it on my phone right now. If you can verify that the dbus trigger works, I can look for that code (or rewrite it because it was pretty easy) and maybe get it merged into the daemon!

wico, 2018-01-30:
@lezed1 Oha! Thank you! I will check that ... :)

wico, 2018-01-30:
Yes, works. If I send a "triggerReactive" it blinks. If you find your script and merge it into the daemon, that would be awesome.

lezed1, 2018-01-30:
Luckily I found this old gist: https://gist.github.com/lezed1/aa4918d6b5e6bd638e7c325c0ed44e7a. `reactive.py` will listen for mouse clicks while `reactive-move-scroll.py` is a bit more... aggressive. I'll have to test if `pynput` can receive X events when running as the daemon, but that's a project for tomorrow! Hopefully I'll get something working soon, and I wish I had noticed this issue before the stale bot revived it.

wico, 2018-01-30:
Noooo worries! :) The reactive-mode is a "bonus" anyway ...