openrazer/openrazer #329: Razer Wildcat / controllers

tripkin, 2017-07-04:
I was wondering if it would be possible to add controller support to the drivers, specifically the Razer Wildcat? I am willing to help, if someone could tell me what information I could provide. I am also willing to test as I have the controller...

z3ntu, 2017-07-05:
Well packet captures from Windows surely are needed for a driver. But honestly I don't think support will be added in this driver but that would be a project to write a driver and bring it upstream into Linux (instead of "our" driver). 
If possible, please send us screenshot of Synapse (all pages) while the controller is plugged in and maybe we can do something.

terrycain, 2017-08-05:
Ahh I completely missed this. Yeah get me screenshots from synapse and I'll see if there's anything that can be done.

closed 2018-01-28 by z3ntu