openrazer/openrazer #330: Setting custom matrix effect in matrix_effect_custom doesn't work correctly with start index > 0

WindohsCrasher, 2017-07-07:
I tried writing a shell script which would change the color of just one single key without affecting the other keys in the same row.

If I try to set only the first key in row 1 (for my BlackWidow Chroma V2 this is the M1 key), everything works correctly when sending the following input: \x1\x0\x0\xFF\xFF\xFF (i.e. the M1 key turns white).

However, if I want to set the key right next to the M1 key (the ` key) to a different color using this input: \x1\x1\x1\xFF\xFF\xFF nothing happens. If I modify it to \x1\x2\x1\xFF\xFF\xFF\xFF\xFF\xFF, only the third key (the "1" key) in the row will be set.

If I understand it correctly, the second byte in the command stream is the start position and the third one is the stop position, so if I want to set the first key on the row, it needs to be 0 and 0 and for the first key it needs to be 1 and 1, and so on - am I correct here or did I get something wrong?

Thank you in advance :)

z3ntu, 2017-07-07:
Try sending the whole row at once and not just a part of the row.

z3ntu, 2017-07-07:
And you should also probably use the D-Bus api instead of directly interfacing with the kernel driver.

terrycain, 2017-07-07:
@z3ntu The driver will send an entire row at once anyway, it'll be easier if you also do the same. But as that interface may change whenever I feel like it, it would be better to use the dbus api

WindohsCrasher, 2017-07-07:
Thank you for your answers. I will try sending entire rows only then!

By the way, is there any documentation for the dbus interface?

z3ntu, 2017-07-07:
For the python library there are some docs in the branch `feature_docs` but otherwise not yet. It's on our todo list.

WindohsCrasher, 2017-07-07:
Alright then, thank you for the quick answers :)

closed 2018-02-28 by stale[bot]