openrazer/openrazer #331: Nostrom not recognised

RomainLG, 2017-07-07:
(redirected from https://github.com/lah7/polychromatic/issues/102)

I get a Device Not Recognized error under Polychromatic.

I'm on Kali which is a rolling Debian.

```
sessionR@kali:~$  sudo journalctl -b | grep -i razer_mount
Jul 07 19:47:49 kali razer_mount[387]: Driver razerkbd
Jul 07 19:47:49 kali razer_mount[386]: Driver razerkbd
Jul 07 19:47:49 kali razer_mount[398]: Device_ID 0003:1532:0111.0002
Jul 07 19:47:49 kali razer_mount[397]: Device_ID 0003:1532:0111.0003
Jul 07 19:47:49 kali razer_mount[401]: Modprobing razerkbd
Jul 07 19:47:49 kali razer_mount[400]: Modprobing razerkbd
Jul 07 19:47:49 kali systemd-udevd[322]: Process 'razer_mount razerkbd 0003:1532:0111.0003' failed with exit code 1.
Jul 07 19:47:49 kali systemd-udevd[312]: Process 'razer_mount razerkbd 0003:1532:0111.0002' failed with exit code 1.

sessionR@kali:~$ lsusb | grep '1532:'
Bus 004 Device 006: ID 1532:0111 Razer USA, Ltd 

sessionR@kali:~$ sudo modprobe razerkbd
modprobe: FATAL: Module razerkbd not found in directory /lib/modules/4.9.0-kali3-amd64
```

Any help appreciated ! :)

terrycain, 2017-07-07:
What version of the driver are you running? has DKMS build the driver for your kernel verison

RomainLG, 2017-07-07:
[image]

z3ntu, 2017-07-07:
Please run `sudo apt-get install linux-headers-$(uname -r)` and post the output.

RomainLG, 2017-07-07:
```
sessionR@kali:~$ sudo apt-get install linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-headers-4.9.0-kali3-amd64
E: Couldn't find any package by glob 'linux-headers-4.9.0-kali3-amd64'
E: Couldn't find any package by regex 'linux-headers-4.9.0-kali3-amd64'
```

I guess I should update my system? Can't get the nostromo to work if I don't have the latest kernal ?

z3ntu, 2017-07-07:
Well you need the kernel headers for the installed kernel. Do all upgrades (`sudo apt-get dist-upgrade`) and then run the previous command again.

RomainLG, 2017-07-07:
I will try that tomorrow morning, I don't have propper internet right now.

RomainLG, 2017-07-08:
@z3ntu @terrycain 
OK with the headers it works! I can only control brightness but I guess it already something.

I intend to use the nostromo as an application/shortcut launcher or to control playback during movies. I hope I'll succeed in this endeavor! 

Thanks!

closed 2017-07-08 by RomainLG