openrazer/openrazer #332: Device not recognized by daemon/driver

SnowCatis, 2017-07-09:
Hello, around a week ago my keyboard lights stopped turning on, I also use a razer deathadder chroma mouse which seems to be working fine. The driver seems to be detecting the keyboard but the daemon doesn't initialize it. I've attached the logs of the commands from the Troubleshooting page.

[dmesg.txt](https://github.com/terrycain/razer-drivers/files/1133992/dmesg.txt)
[daemon.txt](https://github.com/terrycain/razer-drivers/files/1133993/daemon.txt)
[journalctl.txt](https://github.com/terrycain/razer-drivers/files/1133994/journalctl.txt)

terrycain, 2017-07-09:
What ubuntu version you on? Also downgrade to v1.1.13. reboot and see if it works again

SnowCatis, 2017-07-09:
I'm on 16.10 (kubuntu) and I'm not sure how to install the older versions.

terrycain, 2017-07-09:
https://askubuntu.com/questions/138284/how-to-downgrade-a-package-via-apt-get

SnowCatis, 2017-07-09:
apt-cache only shows 1.1.14 being available, and 1.1.13 can't be found.

terrycain, 2017-07-09:
Ahh sigh, the ppa doesnt retain old versions. Grab debs from [here](https://github.com/terrycain/razer-drivers/files/1134023/openrazer-1.1.13_amd64.zip), remove current ones, install old ones, reboot

SnowCatis, 2017-07-09:
I've restarted and it still does not work on version 1.1.13

this is what doing razer-daemon -f does:
[razer-daemon.txt](https://github.com/terrycain/razer-drivers/files/1134038/razer-daemon.txt)

terrycain, 2017-07-09:
`ls -lah /sys/bus/hid/drivers/razer*/`

SnowCatis, 2017-07-09:
output of "ls -lah /sys/bus/hid/drivers/razer*/"
[1.txt](https://github.com/terrycain/razer-drivers/files/1134044/1.txt)

terrycain, 2017-07-09:
`ls -lah /sys/bus/hid/drivers/hid-generic*/`

SnowCatis, 2017-07-09:
output of "ls -lah /sys/bus/hid/drivers/hid-generic*/"
[2.txt](https://github.com/terrycain/razer-drivers/files/1134046/2.txt)

terrycain, 2017-07-09:
ls -lah /sys/bus/hid/devices/

SnowCatis, 2017-07-09:
output of "ls -lah /sys/bus/hid/devices/"
[3.txt](https://github.com/terrycain/razer-drivers/files/1134056/3.txt)

terrycain, 2017-07-09:
Hmm very odd. An entry is missing there should be 0003:1532:021E listed 3 times. that would be why. Dont suppose you know if synapse did a firmware update on the ornata?

Output of `lsusb -vvv -d 1532:021E`

SnowCatis, 2017-07-09:
Not sure, but it could be possible.

[4.txt](https://github.com/terrycain/razer-drivers/files/1134059/4.txt)

terrycain, 2017-07-09:
Ok it doesnt seem to be a firmware update.
```
ls -lah /sys/bus/hid/drivers/
lsmod | grep -i raz
```

SnowCatis, 2017-07-09:
ls -lah /sys/bus/hid/drivers/ =
[5.txt](https://github.com/terrycain/razer-drivers/files/1134065/5.txt)
lsmod | grep -i raz =
[6.txt](https://github.com/terrycain/razer-drivers/files/1134066/6.txt)

terrycain, 2017-07-09:
Ok, very weird, udev seems to successfully unbind from hid-generic and rebind to razerkbd but it seems entries are missing from /sys/bus/hid/drivers/razerkbd (and razermouse too)

Ok update to latest drivers, reboot, get me dmesg and journalctl logs again.

Also you aint doing anything dodgy with the usb's like passing through to virtualbox? Am trying to think what would mess with things in /sys

SnowCatis, 2017-07-09:
I'll be a minute till I'm able to reboot but I'll send you the logs for those commands asap.
I'm not sure about virtualbox however but I was trying to setup a program that apparently works on linux for the keyboard, though I only tried that after the keyboard stopped working, is there a way to check?

https://github.com/CalcProgrammer1/KeyboardVisualizer Thats the link to what I was messing with.

terrycain, 2017-07-09:
If your not sure of usb passthrough then you probably havent done that. And that project should in theory if used on linux plug into the driver so shoudlnt be affecting anything

SnowCatis, 2017-07-09:
dmesg 
[dmesg.txt](https://github.com/terrycain/razer-drivers/files/1134118/dmesg.txt)
journalctl
[journalctl.txt](https://github.com/terrycain/razer-drivers/files/1134119/journalctl.txt)

terrycain, 2017-07-09:
```
[   26.030940] usbhid 1-5:1.1: Razer Device disconnected
[   26.083168] usbhid 1-7:1.2: Razer Device disconnected
```

thats not good.
I have a feeling said devices will then be missing from here
`ls -lah /sys/bus/hid/drivers/razer*/`

terrycain, 2017-07-09:
Your not using an unpowered usb dock are you?

terrycain, 2017-07-09:
Also you able to try this on a different ubuntu 16.10 machine/install?

SnowCatis, 2017-07-09:
I hope not, I was switching to different usb ports as I tried fixing it, also I'm not sure if a laptop would work for that.

ls -lah /sys/bus/hid/drivers/razer*/
[7.txt](https://github.com/terrycain/razer-drivers/files/1134125/7.txt)

terrycain, 2017-07-10:
Laptop should work, i run about 4 razer devices off my work laptop. Trying a different install of ubuntu, if that worked would limit it to something on your main system (which at the moment i have no clue what its doing to break things)

SnowCatis, 2017-07-11:
I'm not really able to try a different install on the laptop just yet, my main computer and laptop both use the same version of kubuntu, maybe I can find another laptop to put the different version on? Until then I'll keep you updated asap, also I am able to try the keyboard on the laptop still if you need me to.

terrycain, 2017-07-11:
@SnowCatis Yeah you can try the keyboard on the laptop that should help

SnowCatis, 2017-07-11:
Okay so I have a little update, the laptop was actually on 16.04.2 of kubuntu, I did test the keyboard on there and it was recognized, but the keyboard still failed to light up.

SnowCatis, 2017-07-15:
still won't light up.. I'm not sure what to try anymore.

terrycain, 2017-07-15:
`ls -lah /sys/bus/hid/drivers/razer*/` on laptop and output of dmesg

SnowCatis, 2017-08-10:
sorry for the long wait, heres the outputs of both commands on the laptop, also I decided to refresh my computer to be 17.04 which didn't seem to change anything yet, my computer really needed it though..

ls -lah /sys/bus/hid/drivers/razer*/
[1.txt](https://github.com/terrycain/razer-drivers/files/1216639/1.txt)
dmesg
[2.txt](https://github.com/terrycain/razer-drivers/files/1216643/2.txt)

SnowCatis, 2017-08-14:
UPDATE:
I was curious to see what would happen if I connected the keyboard to a Windows laptop I had, so i used the razer synapse program on there to see if it turned on, and it actually worked! 
after that I unplugged it and plugged it back into my Linux computer the lights turned on and it works with polychromatic, I'm not exactly sure what happened but I think the keyboard had an issue of some sort with it. Anyway I'm just glad it works now and I hope this helps anyone if something similar happens to them!

closed 2017-08-14 by SnowCatis