openrazer/openrazer #336: Keyboard lightning off when logout

xlinbsd, 2017-07-14:
Hello,

Since the support of my Deathstalker keyboard with the daemon, the keyboard lightning is off when the PC is sleeping or simply on screensaver (Ubuntu Mate 17.04)
Is it possible to setup this behavior, since it is often hard to type password in a black on black keyboard ?

lah7, 2017-07-15:
There is a way to turn off the "turn off devices on screensaver" feature, set the option to `False` in `~/.config/openrazer/razer.conf`:

````
devices_off_on_screensaver = False
````

Be sure to restart the daemon afterwards.

````
razer-daemon -r
````

xlinbsd, 2017-07-15:
Happy to learn there is a personal config file.
Sadly that does not work for me. Tried to restart daemon, then all the system. Not sure if login lights were off but now it is on, good thing, but in locked mode (screensaver), black is black !

I tried to remove log verbosity from the same config file without success, DEBUG info is always written. The file is not taken into account. Another prevalent other config file perhaps ?

terrycain, 2017-07-16:
Yeah the config file was an afterthought and doesnt do too much atm

terrycain, 2017-07-17:
Ite, have patched daemon to respect screensaver option in config. Will be out next release

xlinbsd, 2017-07-17:
Thank you very much Mr Terrycain

closed 2017-07-17 by xlinbsd