openrazer/openrazer #337: Will the Abyssus V1 (1532:001c) ever be supported?

zeus86, 2017-07-14:
...or does this make sense? i dont know if the light is controllable on the first version (currently only lightging blue), and it has hardware-dpi switches.

```
$ lsusb | grep Abyssus
Bus 006 Device 002: ID 1532:001c Razer USA, Ltd RZ01-0036 Optical Gaming Mouse [Abyssus]
```
Model No: RZ01-0036
Product No: RZ01-00360100
2 Switches on the bottom Side: Frequency (125Hz/1000Hz) and Resolution (400dpi/1800dpi/3500dpi)
```
$ lsusb -d 1532:001c -v
Bus 006 Device 002: ID 1532:001c Razer USA, Ltd RZ01-0036 Optical Gaming Mouse [Abyssus]
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x001c RZ01-0036 Optical Gaming Mouse [Abyssus]
  bcdDevice            1.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      93
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```
I really would like to know if it is possible to have this rather old Version of Abyssus to be supported.

z3ntu, 2017-07-18:
* Please post the output of
`lsusb -d 1532:001c -v`
and
`ls -lah /dev/input/by-id/`
* Read [this](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol) because we probably need packet captures.
* Screenshots of all available options on Razer Synapse are needed.
* And lastly, the product & model number of the device for documentation reasons please.

zeus86, 2017-07-18:
screenshots of synapse i can currently not provide, because i'm currently not using a windows on any machine, but for the rest; there you go:

```
zeus@linux:~/$  sudo lsusb -d 1532:001c -v
Bus 006 Device 002: ID 1532:001c Razer USA, Ltd RZ01-0036 Optical Gaming Mouse [Abyssus]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x001c RZ01-0036 Optical Gaming Mouse [Abyssus]
  bcdDevice            1.00
  iManufacturer           1 Razer 
  iProduct                2 Razer Abyssus
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      93
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

zeus@linux:~/$   ls -lah /dev/input/by-id/ | grep -v Abyssus
lrwxrwxrwx 1 root root   9 Jul 19 00:15 usb-Razer_Razer_Abyssus-event-mouse -> ../event3
lrwxrwxrwx 1 root root   9 Jul 19 00:15 usb-Razer_Razer_Abyssus-mouse -> ../mouse0
zeus@linux:~/$ 
```

...dont mind the "grep", the rest of the output was related to my chroma v2 keyboard.

z3ntu, 2017-07-19:
We need to know which lighting functions the hardware supports otherwise we can't do much unfortunately. Please install Windows in a virtual machine and get us the screenshots if you want your device supported (or find screenshots of the "Lighting" tab of Synapse online, that's good too)

terrycain, 2017-07-19:
We'll need packet captures, so you'll probably need a windows vm in the end anyway

zeus86, 2017-07-20:
hm, okay, so i'll set up one, but this may take a while

z3ntu, 2017-07-20:
No rush :)

nirkons, 2019-01-16:
Hi guys, is this software still being worked on?

If you can tell me how, I can capture packets for you on windows,
but the problem with getting screenshots is is that razer synapse does not detect Abyssus V1... which i why I thought OpenRazer will be able to help in this case, I need it for the configuration of the mouse, less for the lighting part... is this something that still can be done after we get the packets?

z3ntu, 2019-02-06:
@nirkons The wiki entry for the packet captures is https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol (the first couple of steps before it's about analysing the captures).
This device is quite an old one so there might be a different software released by Razer back when the device was released that supports this device. Then we need screenshots/captures made when using that software.

zeus86, 2021-01-31:
I think this one can be closed. Today I opened up the mouse for cleaning and maintenance, and it is pretty obvious, that the LED is just a blue LED. There is nothing one could change in software.

z3ntu, 2021-01-31:
Surely it's possible to configure whether the light is on, the DPI and/or poll rate, so I'd keep it open.