openrazer/openrazer #338: Dameon unable to recognize Razer Tartarus Chroma	1532:0208

gazny, 2017-07-15:
I am running on kernel 4.12.1 and am unable to use it since I updated

terrycain, 2017-07-15:
os?

gazny, 2017-07-15:
Solus os
I had it working before I think maybe I dont have a kernel module added.

`Bus 001 Device 002: ID 1532:0208 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0208 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
`
`ls -lah /dev/input/by-id/

lrwxrwxrwx 1 root root   9 Jul 15 11:32 usb-Razer_Razer_Tartarus_Chroma-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 Jul 15 11:32 usb-Razer_Razer_Tartarus_Chroma-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 Jul 15 11:32 usb-Razer_Razer_Tartarus_Chroma-if02-event-mouse -> ../event4
lrwxrwxrwx 1 root root   9 Jul 15 11:32 usb-Razer_Razer_Tartarus_Chroma-if02-mouse -> ../mouse1
`
``

terrycain, 2017-07-15:
lsmod | grep razer

gazny, 2017-07-15:
No output

terrycain, 2017-07-15:
@gazny ok driver isnt loaded. Try
sudo modprobe razerkbd

gazny, 2017-07-15:
`modprobe: FATAL: Module razerkbd not found.
`

terrycain, 2017-07-15:
Must be an issue with solus's packaging, DKMS hasnt ran, it was in a postinstall script in our packaging, you'll have to speak to them.

gazny, 2017-07-15:
thanks for the response, Ill take it to them.

z3ntu, 2017-07-16:
Solus doesn't use DKMS afaik

Justinzobel, 2017-07-16:
4.12.x isn't meant for use just yet @gazny, we need to build the drivers against it, it's a work in progress.

z3ntu, 2018-01-28:
Closing as there was no answer.

closed 2018-01-28 by z3ntu