openrazer/openrazer #339: Support for Razer ManO'War 7.1 (1532:050f)

FlamingCupcake, 2017-07-15:
Hello,
Thank you guys for the wonderful job so far. I bought a ManO'War 7.1 yesterday and would love to help with whatever I can. I have a dual-boot system with Ubuntu 17.04 and Windows 10.

Output of <code> lsusb -d 1532:050f -v </code>:
[manowar_lsusb.txt](https://github.com/terrycain/razer-drivers/files/1150250/manowar_lsusb.txt)

Output of <code> ls -lah /dev/input/by-id/ </code>:
[manowar_ls.txt](https://github.com/terrycain/razer-drivers/files/1150251/manowar_ls.txt)

Pics of Razer Synapse:
[synapse.tar.gz](https://github.com/terrycain/razer-drivers/files/1150252/synapse.tar.gz)

Model number: RZ04-0192
Product number: RZ04-01920300

terrycain, 2017-07-15:
Hmm, I have a feeling there not much to support for this. At least for the krakens all the sound part is done within whatever sound service in linux and this driver controlled the lighting

terrycain, 2017-07-15:
As I know on windows, if you changes some stuff in synapse, then dig deep into the windows sound options you'll see synapse has just changed them instead of doing anything fancy

FlamingCupcake, 2017-07-15:
Ah, fair enough. That's good to know :smiley:

terrycain, 2017-07-15:
Yeah on the plus side, it should work fine on linux quite happily ;-)

closed 2017-07-15 by terrycain