openrazer/openrazer #340: New Razer Blade Stealth 13.3" (1532:022d)

atvilelas, 2017-07-18:
Is there any plans to add support to the new release RBS 13.3"?

z3ntu, 2017-07-18:
* Please post the output of
`lsusb -d 1532:022d -v`
and
`ls -lah /dev/input/by-id/`
* Read [this](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol) because we probably need packet captures.
* Screenshots of all available options on Razer Synapse are needed.
* And lastly, the product & model number of the device for documentation reasons please.

atvilelas, 2017-07-18:
`lsubs -d 1532:022d -v` output:

```

Bus 001 Device 004: ID 1532:022d Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x022d 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

atvilelas, 2017-07-18:
`ls -lah /dev/input/by-id/` output:

```
total 0
drwxr-xr-x. 2 root root 160 Jul 18 16:14 .
drwxr-xr-x. 4 root root 540 Jul 18 16:14 ..
lrwxrwxrwx. 1 root root  10 Jul 18 16:14 usb-11918920-0074GA13_USB_Camera-event-if00 -> ../event10
lrwxrwxrwx. 1 root root   9 Jul 18 16:14 usb-ELAN_Touchscreen-event-if00 -> ../event9
lrwxrwxrwx. 1 root root   9 Jul 18 16:14 usb-Razer_Razer_Blade_Stealth-event-kbd -> ../event6
lrwxrwxrwx. 1 root root   9 Jul 18 16:14 usb-Razer_Razer_Blade_Stealth-if01-event-kbd -> ../event7
lrwxrwxrwx. 1 root root   9 Jul 18 16:14 usb-Razer_Razer_Blade_Stealth-if02-event-mouse -> ../event8
lrwxrwxrwx. 1 root root   9 Jul 18 16:14 usb-Razer_Razer_Blade_Stealth-if02-mouse -> ../mouse0
```

atvilelas, 2017-07-18:
Product #: `RZ09-01963E32`
Model #: `RZ09-0196`

atvilelas, 2017-07-18:
I will work on the capture and screenshots. I blew up all my partitions and I will install Windows shortly.

dbrazzor, 2017-07-29:
Hello ! I have the same laptop and it would be awesome to see those drivers working on it :) Do you need any informations I can give ?

z3ntu, 2017-07-30:
@dbrazzor Screenshots of all options in Synapse and potentially packet captures of the different effects (we'll instruct you later if neccessary)

dbrazzor, 2017-07-30:
Thanks for the answer, do I need Windows to screen synapse's options? If so I'm sorry but I'm currently not able to install Windows on it :/ I would like to show you the packet captures, can you tell me what I need to do?

z3ntu, 2017-07-30:
Windows in a virtual machine is also good (and recommended for packet captures). Note, that you need to pass through your keyboard so you should plug in another keyboard for getting out of the virtual machine 😉

jonmanes, 2017-08-02:
Can I get instructions for the packet caps you need? Should be able to get those tomorrow and send over if that helps. I can run either via a windows 10 VM on top of ubuntu or with windows directly loaded, which ever you think is best for what you need.

jonmanes, 2017-08-02:
[stealth.zip](https://github.com/terrycain/razer-drivers/files/1193824/stealth.zip)
screen caps attached, let me know if you need anything more detailed.

jonmanes, 2017-08-02:
Found [this](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol) above. I will work through the packet captures over the next couple of days.

jonmanes, 2017-08-03:
Interesting info... I ran the screen caps through a windows 10 VM, passing through the keyboard... I made some setting changes to how I wanted the keyboard to function while in windows... These changes persisted after I shut down the VM and after reboot. My keyboard is now working exactly as I want it set up in Ubuntu 17.04.

z3ntu, 2017-08-03:
That's the nice part about Razer storing their settings on the device

terrycain, 2017-08-03:
Ok need packet captures of
* 1 capture of sliding the battery level slider on power section from 10 -> 50 and back a few times
* 1 capture of toggling gaming mode on and off a few times
* 1 capture of brightness from off to bright and back a few times
* 1 capture of no effect -> wave left and right a few times -> no effect -> wave etc...
* 1 capture of same as above but with static
* 1 capture of same as above but with starlight messing with all the options and sliders
* 1 capture of same as above but with spectrum cycling
* 1 capture of same as above but with reactive messing with all the options and sliders
* 1 capture of same as above but with breathing messing with all the options and sliders
* 1 capture of same as above but with ripple

jonmanes, 2017-08-03:
[Stealth_1532_022d.zip](https://github.com/terrycain/razer-drivers/files/1198852/Stealth_1532_022d.zip)

Pcaps attached. Let me know if you need anymore. I tried naming them with the order of what was hit. everything was done a few times. the brightness bars got dropped on slide a few times but I tried to be sure and get a 0 to 100 on all a few times. all colors were run as red green blue yellow, repeat and end on red. 

Thank you for all that you do!

terrycain, 2017-08-04:
@jonmanes what distro you on?

jonmanes, 2017-08-04:
Ubuntu 17.04

specifically :)
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=17.04
DISTRIB_CODENAME=zesty
DISTRIB_DESCRIPTION="Ubuntu 17.04"
NAME="Ubuntu"
VERSION="17.04 (Zesty Zapus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 17.04"
VERSION_ID="17.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=zesty
UBUNTU_CODENAME=zesty

terrycain, 2017-08-04:
Ok i built [these](https://drive.google.com/open?id=0BydYgD2LzDUCeHR4SnJFN25BeFU) on 16.04 but in theory they should work, install them, reboot then as root run 

```
ls -l /sys/bus/hid/drivers
ls -l /sys/bus/hid/drivers/razerkbd
ls -l /sys/bus/hid/drivers/razerkbd/0*/
```

jonmanes, 2017-08-04:
In like Flynn. working great so far. I will run through all the different settings tonight but for now everything I touched is working.

jonmanes, 2017-08-04:
ls -l /sys/bus/hid/drivers
```total 0
drwxr-xr-x 2 root root 0 Aug  4 18:42 hid-generic
drwxr-xr-x 2 root root 0 Aug  4 18:42 hid-multitouch
drwxr-xr-x 2 root root 0 Aug  4 18:42 razerkbd```
ls -l /sys/bus/hid/drivers/razerkbd
```total 0
lrwxrwxrwx 1 root root    0 Aug  4 16:18 0003:1532:022D.0001 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1532:022D.0001
lrwxrwxrwx 1 root root    0 Aug  4 16:18 0003:1532:022D.0002 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:022D.0002
lrwxrwxrwx 1 root root    0 Aug  4 16:18 0003:1532:022D.0003 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:022D.0003
--w------- 1 root root 4096 Aug  4 16:18 bind
lrwxrwxrwx 1 root root    0 Aug  4 18:44 module -> ../../../../module/razerkbd
--w------- 1 root root 4096 Aug  4 18:44 new_id
--w------- 1 root root 4096 Aug  4 16:18 uevent
--w------- 1 root root 4096 Aug  4 18:44 unbind```
ls -l /sys/bus/hid/drivers/razerkbd/0*/
```/sys/bus/hid/drivers/razerkbd/0003:1532:022D.0001/:
total 0
-r--r--r-- 1 root plugdev 4096 Aug  4 16:18 country
lrwxrwxrwx 1 root plugdev    0 Aug  4 16:18 driver -> ../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 Aug  4 16:18 hidraw
drwxr-xr-x 3 root plugdev    0 Aug  4 16:18 input
-rw-rw---- 1 root plugdev 4096 Aug  4 16:18 key_alt_f4
-rw-rw---- 1 root plugdev 4096 Aug  4 16:18 key_alt_tab
-rw-rw---- 1 root plugdev 4096 Aug  4 16:18 key_super
-r--r--r-- 1 root plugdev 4096 Aug  4 16:18 modalias
drwxr-xr-x 2 root plugdev    0 Aug  4 16:18 power
-r--r--r-- 1 root plugdev 4096 Aug  4 16:18 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Aug  4 16:18 subsystem -> ../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Aug  4 16:18 uevent

/sys/bus/hid/drivers/razerkbd/0003:1532:022D.0002/:
total 0
-r--r--r-- 1 root plugdev 4096 Aug  4 16:18 country
lrwxrwxrwx 1 root plugdev    0 Aug  4 16:18 driver -> ../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 Aug  4 16:18 hidraw
drwxr-xr-x 3 root plugdev    0 Aug  4 16:18 input
-rw-rw---- 1 root plugdev 4096 Aug  4 16:18 key_alt_f4
-rw-rw---- 1 root plugdev 4096 Aug  4 16:18 key_alt_tab
-rw-rw---- 1 root plugdev 4096 Aug  4 16:18 key_super
-r--r--r-- 1 root plugdev 4096 Aug  4 16:18 modalias
drwxr-xr-x 2 root plugdev    0 Aug  4 16:18 power
-r--r--r-- 1 root plugdev 4096 Aug  4 16:18 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Aug  4 16:18 subsystem -> ../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Aug  4 16:18 uevent

/sys/bus/hid/drivers/razerkbd/0003:1532:022D.0003/:
total 0
-r--r--r-- 1 root plugdev 4096 Aug  4 16:18 country
-rw-rw---- 1 root plugdev 4096 Aug  4 16:18 device_mode
-r--r----- 1 root plugdev 4096 Aug  4 16:18 device_serial
-r--r----- 1 root plugdev 4096 Aug  4 16:18 device_type
lrwxrwxrwx 1 root plugdev    0 Aug  4 16:18 driver -> ../../../../../../../bus/hid/drivers/razerkbd
-r--r----- 1 root plugdev 4096 Aug  4 16:18 firmware_version
--w--w---- 1 root plugdev 4096 Aug  4 16:18 fn_toggle
drwxr-xr-x 3 root plugdev    0 Aug  4 16:18 hidraw
drwxr-xr-x 3 root plugdev    0 Aug  4 16:18 input
-rw-rw---- 1 root plugdev 4096 Aug  4 16:18 logo_led_state
-rw-rw---- 1 root plugdev 4096 Aug  4 16:18 matrix_brightness
--w--w---- 1 root plugdev 4096 Aug  4 16:18 matrix_custom_frame
--w--w---- 1 root plugdev 4096 Aug  4 16:18 matrix_effect_breath
--w--w---- 1 root plugdev 4096 Aug  4 16:18 matrix_effect_custom
--w--w---- 1 root plugdev 4096 Aug  4 16:18 matrix_effect_none
--w--w---- 1 root plugdev 4096 Aug  4 16:19 matrix_effect_reactive
--w--w---- 1 root plugdev 4096 Aug  4 16:18 matrix_effect_spectrum
--w--w---- 1 root plugdev 4096 Aug  4 16:18 matrix_effect_static
--w--w---- 1 root plugdev 4096 Aug  4 16:19 matrix_effect_wave
-r--r--r-- 1 root plugdev 4096 Aug  4 16:18 modalias
drwxr-xr-x 2 root plugdev    0 Aug  4 16:18 power
-r--r--r-- 1 root plugdev 4096 Aug  4 16:18 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Aug  4 16:18 subsystem -> ../../../../../../../bus/hid
-rw-rw---- 1 root plugdev 4096 Aug  4 16:18 test
-rw-r--r-- 1 root plugdev 4096 Aug  4 16:18 uevent
-r--r----- 1 root plugdev 4096 Aug  4 16:18 version```

jonmanes, 2017-08-05:
Ok, I have gone over all the settings with multiple colors. The only issue I see is with the ripple effect. Both the random and single versions look like the fire effect instead of ripple but a static fire with out flickering and no reaction to key presses.

terrycain, 2017-08-05:
the ripple effect is purely "software" so it sounds like, you last had the fire effect saved to the internal frame buffer on the kb, then i sent it some incorrect rgb frames (which get ignored), then told it to display what it has stored which would still be the fire effect

terrycain, 2017-08-05:
I'll merge it for now, then mind making a new issue just for that?

terrycain, 2017-08-05:
Ahh well its already merged as thats the one I forgot to put in a branch. Yeah make a new issue so we can debug this

jonmanes, 2017-08-05:
Sounds good, I'll put up a new issue for it over the next couple of days and include a pcap for just that.

atvilelas, 2017-08-10:
Sorry about updating this. My unit ended up having a serious issue with its TB3 port and I had to return it before I could capture that info. Thanks everybody for the help.

terrycain, 2017-08-17:
Np. 99% of the functionality is working :smile:

AlexCline, 2017-08-27:
I can confirm that installing the new packages in Ubuntu 17.04 with the 4.12.9 kernel works with a Mid 2017 Razer Blade Stealth.

terrycain, 2017-08-27:
:smile:

terrycain, 2017-08-27:
am sure i still havent released it yet though

samr28, 2017-08-29:
I'm on Debian 9 not Ubuntu with the same laptop. Is there a build for Debian? Or some instructions for newbies?

terrycain, 2017-09-01:
Instructions on https://terrycain.github.io/razer-drivers/

1.1.16 is being built, will be out soon

closed 2017-09-01 by terrycain

samr28, 2017-09-01:
@terrycain the github pages site seems to be down. Not sure if it's Github or something with the repo: http://imgur.com/a/cHX1e

lah7, 2017-09-01:
The project had a little rename, it's now **OpenRazer** and the new URL is https://openrazer.github.io - just depending a redirect on the old website.

terrycain, 2017-09-01:
Yeah I've put off the rename for a while (one of the reasons being that) but then did it earlier otherwise it would never have been renamed

samr28, 2017-09-02:
Ok. Maybe I'm looking on the wrong pages but this is what I found:
https://github.com/openrazer/openrazer/wiki/Building-a-package - Shows "???" for building on Debian
https://software.opensuse.org/download.html?project=hardware%3Arazer&package=openrazer-meta - Just shows how to install the built and stable (from what I understand)

terrycain, 2017-09-03:
Ideally I would just use the openrazer packages from [here](https://software.opensuse.org/download.html?project=hardware%3Arazer&package=openrazer-meta)

samr28, 2017-09-03:
Those are the ones I have installed but they don't support the Blade Stealth 2017 yet.

terrycain, 2017-09-03:
Ahh @z3ntu ^^?

z3ntu, 2017-09-03:
I currently have problems with my laptop (BTRFS isn't as stable as I hoped it would be) so I'm in the process of changing the filesystem back to ext4 and that will take a bit of time

terrycain, 2017-09-03:
BTRFS FTW, np

z3ntu, 2017-09-03:
It's updated. Should be available in the repo soon (whenever OBS publishes it)

samr28, 2017-09-03:
Hmm. I just got the update and installed it but polychromatic still says "Device not recognized"

terrycain, 2017-09-04:
As silly as it sounds, you tried completely uninstalling it then reinstall?

samr28, 2017-09-05:
Oh hey it worked! Thanks for the help :)

terrycain, 2017-09-05:
Yeah we think we know what the issue is, but we've got some renaming work to do so that'll probably break it and or fix it magically lol