openrazer/openrazer #342: Razer Blade Stealth - lights off after reboot / lid close

ahmadnassri, 2017-07-22:
no matter what configuration I have on, after a reboot, or lid close, the keyboard is off, and I have to manually enable the configuration again to enable it, any thoughts how to debug this?

gravelg, 2017-07-31:
I have the same issue, usually just re-switching the color to the current ones enables the backlight again. This is however a minor annoyance.

terrycain, 2017-08-05:
next time it happens, before changing the effect, get the brightness of the keyboard, is it 0?

gravelg, 2017-08-07:
After sleep, the controller widget indicates the settings that were present before closing the lid (ex: static, white, 25%) but the keyboard backlight is off. If I change any of the settings, the backlight comes on with the params as specified.

terrycain, 2017-08-07:
@gravelg I'm assuming if you start typing on the keyboard it doesn't "wake up"

gravelg, 2017-08-07:
It does not, no. However I just opened the lid and the widget indicates the color as "Green" even if it was white before I suspended.

terrycain, 2017-08-08:
I'm pretty sure unless the app is remembering the last effect the daemon itself and the hardware too doesn't offer yet a way to get the last effect so I'm going to assume that's unrelated.

I'll have a think but am not too sure what's going on. I'll charge my blade this weekend and see if it does the same

ahmadnassri, 2017-08-08:
my observed behavior is exactly how @gravelg described

terrycain, 2017-08-10:
Output of lsusb please. I have a feeling I know what it is

gravelg, 2017-08-13:
lsusb
```Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 04f3:223c Elan Microelectronics Corp. 
Bus 001 Device 004: ID 1532:0220 Razer USA, Ltd 
Bus 001 Device 003: ID 0bda:579f Realtek Semiconductor Corp. 
Bus 001 Device 002: ID 0cf3:e300 Atheros Communications, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub```

lsusb -vv
[lsusb_vv.txt](https://github.com/terrycain/razer-drivers/files/1220478/lsusb_vv.txt)

gravelg, 2018-01-29:
Hi, this is still happening. Any news?

sjlouder, 2018-08-19:
This may be a separate issue, but I have similar behavior, so perhaps I can help debug.
Model: Razer Blade Stealth (Late 2016)
OS: Fedora 28
Kernel: 4.17.14-202.fc28.x86_64
OpenRazer: 2.3.1-2.1

In my case, if I hibernate/reboot while the brightness is set to 0, when I resume, the backlight does not work even when I change the brightness (the fn keys will light up while the the fn key is pressed, but otherwise there is no backlight). To remedy/workaround, I have to reset the effect while the brightness is not set to 0. I have been able to recreate this while the machine is on using the python api.

    import openrazer.client
    kbd = openrazer.client.DeviceManager().devices[0]
    
    kbd.brightness = 0
    kbd.brightness                      # Returns dbus.Double(0.0)
    kbd.fx.static(100,100,100)    # Returns True
    kbd.brightness = 50              # No backlight, even if changed via fn keys
    kbd.brightness                      # Returns dbus.Double(49.8)
    kbd.fx.static(100,100,100)    # backlight works

The syslog doesn't show any errors, just the dbus call messages

So, my theory is that changing/setting the effect while brightness=0 will cause the backlight not to work. If the driver is setting an effect on boot/resume without checking/updating the brightness value, that may be a specific instance of this issue.

gravelg, 2019-06-16:
I just saw ^, that might be the issue. I have noticed that, even if the backlight was off after a resume, it will be back to "regular" settings after a reboot. Unsure if this is expected or nor.