openrazer/openrazer #344: Missing public key for apt-get update

binarykitchen, 2017-07-27:
```
W: GPG error: http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_17.04  Release: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 062F9FD4D6D11CE4
W: The repository 'http://download.opensuse.org/repositories/hardware:/razer/xUbuntu_17.04  Release' is not signed.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
```

Was following the instructions on https://software.opensuse.org//download.html?project=hardware%3Arazer&package=razergenie, still seeing the above errors when doing an `apt-get update` on Ubuntu 17.04

binarykitchen, 2017-07-27:
ah, missed that part

```
wget -nv http://download.opensuse.org/repositories/hardware:razer/xUbuntu_17.04/Release.key -O Release.key
sudo apt-key add - < Release.key
```

closed 2017-07-27 by binarykitchen

z3ntu, 2017-07-27:
If you think the instructions can be improved, you should look at https://github.com/openSUSE/software-o-o :wink: