openrazer/openrazer #345: Razor Death Adder 2013 not recognized

binarykitchen, 2017-07-27:
Guess probably because it's not supported yet.

```
└─( 2 ) ❱❱❱ lsusb | grep '1532:'
Bus 005 Device 002: ID 1532:0037 Razer USA, Ltd 
```

Not listed in https://github.com/terrycain/razer-drivers#mouse

Any chance to add support for it for Ubuntu 17.04?

z3ntu, 2017-07-27:
Duplicate of #195. You can help by getting pcaps as mentioned in that thread.

closed 2017-07-27 by z3ntu

binarykitchen, 2017-07-27:
@z3ntu will do ... but ummm what is pcaps? not mentioned in that thread ...

z3ntu, 2017-07-27:
Packet captures (it's described in the wiki). But actually we might not need captures. You can subscribe to #195 and get notified of further development. We are quite busy atm so you have to wait a bit.