openrazer/openrazer #346: Razer Orochi 2015

synetax, 2017-07-27:
Hi, the Razer Orochi 2015 seems to work with the daemon, however, Polychromatic doesn't seem to detect it. I was pointed here from there and told to provide the following information...

Model number: RZ01-01550100-R3G1

Wireless (Bluetooth)
```
$ lsusb -d 1532:0056 -v
*lemons*
```
```
$ ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root  80 Jul 27 19:16 .
drwxr-xr-x 4 root root 460 Jul 27 19:18 ..
lrwxrwxrwx 1 root root   9 Jul 27 19:16 usb-Azurewave_USB2.0_UVC_HD_Webcam_NULL-event-if00 -> ../event8
lrwxrwxrwx 1 root root   9 Jul 27 19:16 usb-ITE_Tech._Inc._ITE_Device_8910_-event-kbd -> ../event7
```
```
$ ls -l /sys/bus/hid/devices/ | grep '1532:'
lrwxrwxrwx 1 root root 0 Jul 27 19:18 0005:1532:0056.0003 -> ../../../devices/virtual/misc/uhid/0005:1532:0056.0003 (looks like a razer device)
```

When connected with the wire it works just fine, but as 1532:0048 (Orochi Wired)

I'm wondering if there's a way to bind the driver to the device over Bluetooth and how I would go about doing that with `modprobe`?

I'm willing to provide screenshots of Synapse as well as packet captures if necessary.

z3ntu, 2017-07-27:
Ah so the 0056 is the wireless mode of 0048? Can you confirm this mouse is called "RAZER Orochi 8200" (sticker behind the battery cover)?

synetax, 2017-07-27:
Yeah, it is.

terrycain, 2017-08-03:
Wait just to confirm, if you plug your mouse in wired it will appear as 0x0048 right?

synetax, 2017-08-04:
That's correct, yes.

terrycain, 2017-08-04:
@synetax what os you on

synetax, 2017-08-04:
I'm on Ubuntu 16.04.2 LTS

terrycain, 2017-08-04:
Get debs from here https://drive.google.com/open?id=0BydYgD2LzDUCZnplQU43Ti1Ibms
Install, reboot, then run the following as root

```
ls -l /sys/bus/hid/drivers
ls -l /sys/bus/hid/drivers/razermouse
ls -l /sys/bus/hid/drivers/razermouse/0*/
```

synetax, 2017-09-28:
@terrycain 
Hi, sorry for the later reply I haven't been using the mouse on Ubuntu as much as I expected to...
I installed the debs as requested. Here's the console output:
`$ ls -l /sys/bus/hid/drivers
total 0
drwxr-xr-x 2 root root 0 Sep 28 08:40 hid-generic
drwxr-xr-x 2 root root 0 Sep 28 08:40 hid-multitouch`

`~$ ls -l /sys/bus/hid/drivers/razermouse
ls: cannot access '/sys/bus/hid/drivers/razermouse': No such file or directory
`
`$ ls -l /sys/bus/hid/drivers/razermouse/0*/
ls: cannot access '/sys/bus/hid/drivers/razermouse/0*/': No such file or directory
`
A bit strange, also having installed the drivers doing things like opening the terminal with `ctrl + alt + T` (and opening Chrome) for some reason takes forever.

z3ntu, 2017-09-28:
For Chrome see #358 

Could you run `sudo modprobe razermouse`?

terrycain, 2017-09-28:
@synetax Pretty sure its a bug with dbus, run gnome-keyring-daemon once you've logged in

z3ntu, 2017-09-28:
Please plug your mouse out and back in and run the last command again.

synetax, 2017-09-28:
This is for the wireless mode, so I tried just switching the mouse off and back on and the results of running `$ ls -l /sys/bus/hid/drivers/razermouse/0*/` were the same as before.

However navigating into the razermouse folder it is clearly not empty:
```
/sys/bus/hid/drivers/razermouse/module$ la
coresize  drivers  holders  initsize  initstate  notes  refcnt  sections  srcversion  taint  uevent
```
Am I missing something?

Brushtail-Highton, 2018-01-24:
Hi,
I raised issue #509 which has been closed as a duplicate of this one. 

I have the same issue but my circumstances are slightly different. I have the issue under Ubuntu 17.10 and Fedora 27, with Fedora 27 being the main system I use. My computer doesn't have bluetooth support so I have installed a Bluetooth 4.0 dongle. My experience potentially indicates that Polychromatic and Razergenie have been written to interact with usb devices and not bluetooth devices.
Like the person above said, the mouse works fine with the Daemon, but not the utilities. Like above, when the mouse is connected to usb via the supplied cable, lsusb shows the device with the wired device id mentioned above and both utilities can see it. 
If the mouse is connected via bluetooth, lsusb doesn't see it at all, it only sees the bluetooth dongle, which obviously isn't supported.
Bluetoothctl sees the mouse as its mac address E3:E0:15:FF:A3:2B.

I also have another question about the mouse usage. If I am using rechargable batteries in the mouse and the mouse is being used in wired mode, it does not recharge the batteries. Is this an intended design?

z3ntu, 2018-01-24:
About Polychromatic and RazerGenie: They both (more or less direct) use the D-Bus interface provided by the daemon so if the daemon doesn't support a device, the so-called frontends can't do anything with that device.

That said, I don't personally own a wireless Razer device so I myself can't really do much about it. I don't know if @terrycain has one .

The charging stuff, no idea.

I expected the ls of the razermouse driver directory to be like that otherwise the daemon would see the device too 😉

Brushtail-Highton, 2018-01-24:
Sorry, I'm a little confused by your last comment about the daemon. 

The daemon sees the device irrespective of whether the device is running in wired mode or bluetooth mode.

Also just one other comment, the utilities work fine with the mouse in both wired mode and bluetooth mode under Windows 10. I boot my machine between Windows 10, Fedora 27 and Ubuntu 17.10 depending on what I want to do.

z3ntu, 2018-01-24:
> The daemon sees the device irrespective of whether the device is running in wired mode or bluetooth mode.

This is apparently not true according to what you wrote (?).

> Also just one other comment, the utilities work fine with the mouse in both wired mode and bluetooth mode under Windows 10.

Yeah I can imagine that ^^

Here's a diagram of kind-of how the different components work together and via which "protocol" (the blue components are provided by this project and the green components are the frontends / applications):

[image]

Brushtail-Highton, 2018-01-24:
Sorry, it looks like I confused things. I was incorrectly classifying the openrazer-daemon.service and the kernel driver combined as the daemon. The kernel driver can definitely see the device in both mode, but openrazer-daemon.service appears to see the device if it is wired but not when it is connected via the bluetooth usb dongle.

If I stop the service and restart it while in bluetooth mode and issue the command systemctl --user status openrazer-daemon I get the following output:

● openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/home/steve/.config/systemd/user/openrazer-daemon.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-01-25 08:33:41 AEDT; 5s ago
     Docs: man:openrazer-daemon(8)
 Main PID: 10576 (openrazer-daemo)
   CGroup: /user.slice/user-1000.slice/user@1000.service/openrazer-daemon.service
           └─10576 openrazer-daemon

Jan 25 08:33:41 localhost.localdomain openrazer-daemon[10576]: 2018-01-25 08:33:41 | razer                          | DEBUG    | Loaded device specification: RazerTartarusChroma -------------------
Jan 25 08:33:41 localhost.localdomain openrazer-daemon[10576]: 2018-01-25 08:33:41 | razer                          | DEBUG    | Adding razer.devices.supportedDevices method to DBus
Jan 25 08:33:41 localhost.localdomain openrazer-daemon[10576]: 2018-01-25 08:33:41 | razer                          | DEBUG    | Adding razer.devices.syncEffects method to DBus
Jan 25 08:33:41 localhost.localdomain openrazer-daemon[10576]: 2018-01-25 08:33:41 | razer                          | DEBUG    | Adding razer.devices.getSyncEffects method to DBus
Jan 25 08:33:41 localhost.localdomain openrazer-daemon[10576]: 2018-01-25 08:33:41 | razer                          | DEBUG    | Adding razer.daemon.version method to DBus
Jan 25 08:33:41 localhost.localdomain openrazer-daemon[10576]: 2018-01-25 08:33:41 | razer                          | DEBUG    | Adding razer.devices.getOffOnScreensaver method to DBus
Jan 25 08:33:41 localhost.localdomain openrazer-daemon[10576]: 2018-01-25 08:33:41 | razer                          | DEBUG    | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
Jan 25 08:33:41 localhost.localdomain openrazer-daemon[10576]: 2018-01-25 08:33:41 | razer                          | DEBUG    | Adding razer.daemon.stop method to DBus
Jan 25 08:33:41 localhost.localdomain openrazer-daemon[10576]: 2018-01-25 08:33:41 | razer                          | DEBUG    | Adding razer.devices.getDevices method to DBus
Jan 25 08:33:41 localhost.localdomain openrazer-daemon[10576]: 2018-01-25 08:33:41 | razer                          | INFO     | Serving DBus

If I stop the service and restart it while in wired mode and issue the command systemctl --user status openrazer-daemon I get the following output:

● openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/home/steve/.config/systemd/user/openrazer-daemon.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-01-25 08:52:30 AEDT; 5s ago
     Docs: man:openrazer-daemon(8)
 Main PID: 10763 (openrazer-daemo)
   CGroup: /user.slice/user-1000.slice/user@1000.service/openrazer-daemon.service
           └─10763 openrazer-daemon

Jan 25 08:52:30 localhost.localdomain openrazer-daemon[10763]: 2018-01-25 08:52:30 | razer.device0                  | DEBUG    | Adding razer.device.lighting.scroll.getScrollActive method to DBus
Jan 25 08:52:30 localhost.localdomain openrazer-daemon[10763]: 2018-01-25 08:52:30 | razer                          | DEBUG    | Adding razer.devices.getSyncEffects method to DBus
Jan 25 08:52:30 localhost.localdomain openrazer-daemon[10763]: 2018-01-25 08:52:30 | razer                          | DEBUG    | Adding razer.daemon.stop method to DBus
Jan 25 08:52:30 localhost.localdomain openrazer-daemon[10763]: 2018-01-25 08:52:30 | razer                          | DEBUG    | Adding razer.devices.getDevices method to DBus
Jan 25 08:52:30 localhost.localdomain openrazer-daemon[10763]: 2018-01-25 08:52:30 | razer                          | DEBUG    | Adding razer.devices.syncEffects method to DBus
Jan 25 08:52:30 localhost.localdomain openrazer-daemon[10763]: 2018-01-25 08:52:30 | razer                          | DEBUG    | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
Jan 25 08:52:30 localhost.localdomain openrazer-daemon[10763]: 2018-01-25 08:52:30 | razer                          | DEBUG    | Adding razer.daemon.version method to DBus
Jan 25 08:52:30 localhost.localdomain openrazer-daemon[10763]: 2018-01-25 08:52:30 | razer                          | DEBUG    | Adding razer.devices.supportedDevices method to DBus
Jan 25 08:52:30 localhost.localdomain openrazer-daemon[10763]: 2018-01-25 08:52:30 | razer                          | DEBUG    | Adding razer.devices.getOffOnScreensaver method to DBus
Jan 25 08:52:30 localhost.localdomain openrazer-daemon[10763]: 2018-01-25 08:52:30 | razer                          | INFO     | Serving DBus

If I issue the command systemctl --user status openrazer-daemon after disconnecting the cable from the mouse and usb socket, I get the following output:

● openrazer-daemon.service - Daemon to manage razer devices in userspace
   Loaded: loaded (/home/steve/.config/systemd/user/openrazer-daemon.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-01-25 08:52:30 AEDT; 7min ago
[... cut ...]
Jan 25 08:52:45 localhost.localdomain openrazer-daemon[10763]: 2018-01-25 08:52:45 | razer                          | DEBUG    | Device event [remove]: /devices/pci0000:00/0000:00:13.0/usb5/5-2/5-2
Jan 25 09:02:47 localhost.localdomain openrazer-daemon[10763]: 2018-01-25 09:02:47 | razer                          | DEBUG    | Device event [remove]: /devices/virtual/misc/uhid/0005:1532:0056.000
Jan 25 09:03:06 localhost.localdomain openrazer-daemon[10763]: 2018-01-25 09:03:06 | razer                          | DEBUG    | Device event [add]: /devices/virtual/misc/uhid/0005:1532:0056.000C

perillamint, 2018-03-05:
I examined feature_orochi_0x0056 branch. It seems it is trying to detect Bluetooth device using USB specific way.

Since Bluetooth is not USB, Razer Orochi 2015 in BT mode does not reach razer_attr_read_device_type(), it will not be detected by the driver. so razer daemon could not detect that.

Brushtail-Highton, 2018-03-05:
>
> I examined feature_orochi_0x0056 branch. It seems it is trying to 
> detect Bluetooth device using USB specific way.
>
> Since Bluetooth is not USB, Razer Orochi 2015 in BT mode does not 
> reach razer_attr_read_device_type(), it will not be detected by the 
> driver. so razer daemon could not detect that.
Just one question on this, does it make any difference to its 
functionality if bluetooth is being provided through USB?

regards,

Steve

z3ntu, 2018-03-05:
I personally didn't do much with the kernel driver and nothing with bluetooth Razer devices as I don't own any. So I can't really help here unfortunately. PRs are welcome though :)

perillamint, 2018-03-06:
@z3ntu I'm currently hacking on kernel module. If I achieve some progress about it, I'll report back here.

jxosborne, 2018-03-11:
How's the kernel module modification going?  I'd love to see this working!

Brushtail-Highton, 2018-03-11:
Just one question on the kernel hack, given that the kernel module can 
see the mouse irrespective of whether it is running in wired mode or 
bluetooth mode, otherwise the mouse wouldn't be usable at all in 
bluetooth mode, why is a kernel module change required for the utilities 
to function with the mouse in bluetooth mode?

regards,

Steve

Rmdhn1, 2020-05-09:
is this ded