openrazer/openrazer #347: Razer Anansi :  Invalid USB repsonse spamming

Linuxine, 2017-07-28:
I have installed the razer-drivers on Fedora 25 ; my Razer Anansi seems recognized, I can see it in interfaces such as RazerGenie or PolyChromatic Controller. But I am not able to set the colors or the brightness. When I try with the GUI, they show no error, but nothing happens, and when restarting them, the settings are gone. When I try to modify some file, for example :

`echo -n "255" > matrix_brightness`

The file is not modified, my brightness stays to 0. 
The driver logs indicate that the request is sent :

`2017-07-28 13:49:51 | razer.device0                  | DEBUG    | DBus call get_brightness`

But nothing seems to be received by the keyboard. The logs in /var/log/messages are indicating in a loop the following error :

kernel: razer driver: Device data transfer failed.
kernel: razer driver: Invalid USB repsonse. USB Report length: -32
kernel: razerkbd: Invalid Report Length. Start Marker: 00 id: 00 Num Params: 00 Reserved: 00 Command: 00 Params: 0000000000000000000000
0000000000 .

This seems to be related to  razer_send_control_msg and razer_get_usb_response in razercommon.c. But for razer_get_usb_response, the comment indicate that this function is compatible with the  Chroma, Mamba,  BlackWidow Ultimate 2013, and Firefly*, so is it normal to call in for my Anansi ?

Thanks a lot for your help !

z3ntu, 2017-07-28:
Can you please run
```
cd $(dirname /sys/bus/hid/drivers/razerkbd/0*/device_serial)
cat device_serial
cat device_mode
```

Linuxine, 2017-07-28:
Hello, yes I forgot to add, strangely, the device_serial is empty... the mode and type are set, though.
~~~
/sys/bus/hid/drivers/razerkbd/0003:1532:010F.0016$ cat device_serial

/sys/bus/hid/drivers/razerkbd/0003:1532:010F.0016$ cat device_mode
0:0
/sys/bus/hid/drivers/razerkbd/0003:1532:010F.0016$ cat device_type 
Razer Anansi
~~~

z3ntu, 2017-07-28:
Are new messages appearing in dmesg when you run the `cat` command?

Linuxine, 2017-07-28:
For both commands, (cat device_serial and cat device_mode), I get the same error :

~~~
[11719.617564] razer driver: Device data transfer failed.
[11719.618486] razer driver: Invalid USB repsonse. USB Report length: -32
[11719.618488] razerkbd: Invalid Report Length. Start Marker: 00 id: 00 Num Params: 00 Reserved: 00 Command: 00 Params: 00000000000000000000000000000000 .
~~~

Linuxine, 2017-07-28:
I compiled the driver from the source, and added some print in razercommon.c, if this could help : I added in razer_send_control_msg the following line:
~~~
printk("send_control_msg len=%d size=%u\n");
~~~

and in razer_get_usb_response:
~~~
printk("get_usb_response len=%d size=%u\n");  
~~~

The log are displaying:
~~~
[20490.626426] send_control_msg len=500812384 size=488300
[20490.626428] razer driver: Device data transfer failed.
[20490.626596] get_usb_response len=0 size=393548
[20490.627433] razer driver: Invalid USB repsonse. USB Report length: -32
[20490.627442] razerkbd: Invalid Report Length. Start Marker: 00 id: 00 Num Params: 00 Reserved: 00 Command: 00 Params: 00000000000000000000000000000000 .
[20490.628197] input: Razer Razer Anansi as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:010F.001F/input/input84
[20490.629295] razerkbd 0003:1532:010F.001F: input,hidraw3: USB HID v1.11 Mouse [Razer Razer Anansi] on usb-0000:00:14.0-8/input2
~~~

Zerophase, 2017-07-30:
@Linuxine I just started poking around with the issue. Nothing directly jumps out at me that has changed with the driver to break the support for this keyboard. Could you try changing `.001F` to `.0003`? That's all I can think of off of the top of my head. I'll be looking into this over the next couple days.

Zerophase, 2017-08-03:
Think I found the issue.

response and report index had their values changed. I'm guessing this could affect other keyboards as well. 
```c
    return razer_get_usb_response(usb_dev, 0x02, request_report, 0x02, response_report, RAZER_BLACKWIDOW_CHROMA_WAIT_MIN_US, RAZER_BLACKWIDOW_CHROMA_WAIT_MAX_US);
```
The twos were changed to ones.
```c   
return razer_get_usb_response(usb_dev, 0x01, request_report, 0x01, response_report, RAZER_BLACKWIDOW_CHROMA_WAIT_MIN_US, RAZER_BLACKWIDOW_CHROMA_WAIT_MAX_US);```

I fixed the issue. This could possibly apply to other keyboards as well.

Zerophase, 2017-08-03:
#349

Linuxine, 2017-08-03:
Thanks a lot ! I tested your patch and now it works \o/ I am able to choose and apply the colors on my Anansi *-*. Thanks again for you responsiveness !

closed 2017-08-03 by Linuxine

z3ntu, 2017-08-03:
Thanks @Zerophase

vikasjayaswal, 2021-04-01:
vic@vic-MS-7B92:/sys/bus/hid/drivers/razerkbd/0003:1532:021E.0003$ cat device_serial
PM1836F11802681

vic@vic-MS-7B92:/sys/bus/hid/drivers/razerkbd/0003:1532:021E.0003$ cat device_mode
3:0

[    9.180700] kernel: razer driver: Invalid USB response. USB Report length: 26
[    9.180704] kernel: razerkbd: Invalid Report Length. Start Marker: 00 id: 00 Num Params: 00 Reserved: 00 Command: 00 Params: 00000000000000000000000000000000 .
[    9.823816] kernel: iommu ivhd0: AMD-Vi: Event logged [INVALID_DEVICE_REQUEST device=00:00.0 pasid=0x00000 address=0xfffffffdf8000000 flags=0x0a00]

z3ntu, 2021-04-02:
Continuation in #1191