openrazer/openrazer #348: Perform no-change rebuild of python3-evdev in terrz/razerutils PPA for Python 3.6 (and Ubuntu Artful) compatibility

wk, 2017-07-30:
Terry's Ubuntu terrz/razerutils PPA contains a build of python3-evdev (0.6.4) that is newer than Ubuntu's own build (0.4.1 in Trusty, 0.4.7 in Xenial, Yakkety, Zesty, Artful). It is a dependency of razer-daemon. As a result, users which follow the instructions on https://terrycain.github.io/razer-drivers/#ubuntu and add the PPA will find themselves installing this build.

The current build of this package in terrz/razerutils is marked incompatible with python 3.6 ```Depends: libc6 (>= 2.4), python3 (<< 3.6), python3 (>= 3.5~)```, while Ubuntu Artful recently incorporated the python3 3.6.2 packages, resulting in a conflict.

A no-change rebuild of ``python3-evdev`` in an environment with newer python3 (and an optional version bump to 0.7.0, since upstream has meanwhile moved on) is required to resolve this in order to prevent Ubuntu Artful users who make use of the terrz/razerutils PPA from being prevented from upgrading their python3.

terrycain, 2017-08-05:
does artful still not have anything greater than 0.6.4?

wk, 2017-08-05:
Indeed, at this time Ubuntu [Artful's python3-evdev](https://packages.ubuntu.com/artful/python3-evdev ) is still on 0.4.7. This package is a little special since its only downstream dependencies in Ubuntu are Ubuntu's own [autopilot](https://launchpad.net/autopilot) and related packages, it does not exist in Debian:

```autopilot-desktop
  ubuntu-system-settings-autopilot
  ubuntu-keyboard-autopilot
  qtmir-tests
  python3-mir-perf-framework
  python3-autopilot-tests
  python3-autopilot
  autopilot-touch
```

The version has been pinned to a release that's nearly 2.5 years old by the maintainer, who appears to do regular no-change rebuilds to keep up with Ubuntu (and even a backport for kernel 4.4.0 compatibility), but is not keeping up with upstream.

Perhaps the Ubuntu maintainer of the package, @sil2100 could help here? Unless there are latent forward compatibility issues with autopilot (none of the rdepends appear to rely on this specific version of python-evdev), a bump of the package in Ubuntu universe to the current upstream (0.7.0) would presumably eliminate the need for @terrycain to maintain his own build in the PPA?

terrycain, 2017-08-05:
Yeah even if they released 0.6.4 i could remove the package entirely from my PPA, will have a look this week at rebuilding it, but if launchpad is too much of a pita i might just make 0.6.5

z3ntu, 2017-09-29:
Debian Buster and Sid have packages for python3-evdev :) https://packages.debian.org/search?keywords=python3-evdev

z3ntu, 2017-09-29:
terrz/razerutils is deprecated. See https://github.com/openrazer/openrazer/releases/tag/v2.0.0

closed 2017-09-29 by z3ntu