openrazer/openrazer #350: Unable to set custom colors

joonatoona, 2017-08-04:
Device: Blade Stealth 1.04

Razer-Daemon version: 1.1.15-0ubuntu1

Trying to set custom colors using the python bindings fails

```
ERROR:dbus.connection:Unable to set arguments (b'\x00...(cut)...\xff',) according to signature None: <class 'ValueError'>: embedded null byte
```

terrycain, 2017-08-04:
I need more context than that. Also could do with daemon logs

joonatoona, 2017-08-04:
What other context can I give?

```
2017-08-04 14:31:25 | razer                          | INFO     | Adding razer.devices.getDevices method to DBus
2017-08-04 14:31:25 | razer                          | INFO     | Adding razer.daemon.supportedDevices method to DBus
2017-08-04 14:31:25 | razer                          | INFO     | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2017-08-04 14:31:25 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2017-08-04 14:31:25 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2017-08-04 14:31:25 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2017-08-04 14:31:25 | razer                          | INFO     | Adding razer.daemon.version method to DBus
2017-08-04 14:31:25 | razer                          | INFO     | Adding razer.daemon.stop method to DBus
2017-08-04 14:31:25 | razer                          | INFO     | Serving DBus
2017-08-04 14:31:33 | razer                          | DEBUG    | DBus called get_serial_list
2017-08-04 14:31:33 | razer.device0                  | DEBUG    | DBus call get_device_type
2017-08-04 14:31:33 | razer.device0                  | DEBUG    | DBus call get_device_name
2017-08-04 14:31:33 | razer.device0                  | DEBUG    | DBus call get_device_type
2017-08-04 14:31:33 | razer.device0                  | DEBUG    | DBus call get_firmware
2017-08-04 14:31:33 | razer.device0                  | DEBUG    | DBus call version
2017-08-04 14:31:33 | razer.device0                  | DEBUG    | DBus call has_matrix
2017-08-04 14:31:33 | razer.device0                  | DEBUG    | DBus call has_matrix
2017-08-04 14:31:36 | razer                          | DEBUG    | DBus called get_serial_list
2017-08-04 14:31:36 | razer.device0                  | DEBUG    | DBus call get_device_type
2017-08-04 14:31:36 | razer.device0                  | DEBUG    | DBus call get_device_name
2017-08-04 14:31:36 | razer.device0                  | DEBUG    | DBus call get_device_type
2017-08-04 14:31:36 | razer.device0                  | DEBUG    | DBus call get_firmware
2017-08-04 14:31:36 | razer.device0                  | DEBUG    | DBus call version
2017-08-04 14:31:36 | razer.device0                  | DEBUG    | DBus call has_matrix
2017-08-04 14:31:36 | razer.device0                  | DEBUG    | DBus call has_matrix
2017-08-04 14:31:40 | razer                          | DEBUG    | DBus called get_serial_list
2017-08-04 14:31:40 | razer.device0                  | DEBUG    | DBus call get_device_type
2017-08-04 14:31:40 | razer.device0                  | DEBUG    | DBus call get_device_name
2017-08-04 14:31:40 | razer.device0                  | DEBUG    | DBus call get_device_type
2017-08-04 14:31:40 | razer.device0                  | DEBUG    | DBus call get_firmware
2017-08-04 14:31:40 | razer.device0                  | DEBUG    | DBus call version
2017-08-04 14:31:40 | razer.device0                  | DEBUG    | DBus call has_matrix
2017-08-04 14:31:40 | razer.device0                  | DEBUG    | DBus call has_matrix
```

terrycain, 2017-08-04:
Hmm, odd, you should use device.fx.advanced.rows and device.fx.advanced.cols to iterate and you can use `device.fx.advanced.matrix[col,row] = (255, 0, 0)` instead of calling set.

Your on a blade stealth, am pretty sure they use less keys which should mean it might be an easy fix, send me a pic of the laptop keyboard

joonatoona, 2017-08-04:
Same issue. Is there docs for the python bindings? I'm just copying from razercommander.

[image]

terrycain, 2017-08-05:
Yeah I said the above with a feeling that it still wouldn't work. I reckon cols is saying something like 22, when it should say 15. You on Ubuntu?

joonatoona, 2017-08-05:
Yeah, that laptop is on Ubuntu 17.04

terrycain, 2017-08-05:
Hmm I was going to give you packages to install but all it would be is change 1 number.

There should be a file like `/usr/lib/python3/dist-packages/razer_daemon/hardware/mouse.py`
Look for `class RazerBladeStealth(_MacroKeyboard):`, it will define the matrix dimensions (which is what the python library uses to create its internal numpy array and resultantly the binary that gets sent to the kb)

Change `[6, 22]` to `[6, 15]`
Then reboot, then try that code again.

Here is the diff
```diff
--- a/daemon/razer_daemon/hardware/keyboards.py
+++ b/daemon/razer_daemon/hardware/keyboards.py
@@ -528,7 +528,7 @@ class RazerBladeStealth(_MacroKeyboard):
     USB_PID = 0x0205
     HAS_MATRIX = True
     DEDICATED_MACRO_KEYS = False
-    MATRIX_DIMS = [6, 22]  # 6 Rows, 22 Cols
+    MATRIX_DIMS = [6, 15]  # 6 Rows, 22 Cols
     METHODS = ['get_firmware', 'get_matrix_dims', 'has_matrix', 'get_device_name', 'get_device_type_keyboard', 'get_brightness', 'set_brightness', 'set_wave_effect', 'set_static_effect', 'set_spectrum_eff
                'set_reactive_effect', 'set_none_effect', 'set_breath_random_effect', 'set_breath_single_effect', 'set_breath_dual_effect',
                'set_custom_effect', 'set_key_row'
```

joonatoona, 2017-08-05:
Same issue. (Except I had to shirnk the matrix)

terrycain, 2017-08-05:
What about 

```
import razer.client 
a = razer.client.DeviceManager() 
b = a.devices[0]          // Assuming it's the only device 
b.name 
b.brightness = 100.0 
b.fx.advanced.matrix[4,5] = (255, 0, 0) 
b.fx.advanced.draw() 
```

Does that work?

joonatoona, 2017-08-05:
Exact same issue :/

terrycain, 2017-08-05:
nothing in dmesg?

joonatoona, 2017-08-05:
```
[  229.926539] razerkbd: Command failed. Start Marker: 03 id: ff Num Params: 02 Reserved: 0e Command: 04 Params: 01ff0000000000000000000000000000 .
```

terrycain, 2017-08-05:
WTF thats the command to change the brightness

joonatoona, 2017-08-05:
Changing the brightness works though? That makes no sense ;-;

terrycain, 2017-08-05:
Odd, very odd

terrycain, 2017-08-05:
ahh, it will say that sometimes when its busy doing other stuff so if brightness is working then it can be ignored

joonatoona, 2017-08-05:
Here's the full output of `dmesg | grep razer`

```
[   17.064578] razerkbd: loading out-of-tree module taints kernel.
[   17.064629] razerkbd: module verification failed: signature and/or required key missing - tainting kernel
[   17.375971] razerkbd 0003:1532:0205.0001: input,hidraw0: USB HID v1.11 Keyboard [Razer Razer Blade Stealth] on usb-0000:00:14.0-8/input0
[   17.376159] razerkbd 0003:1532:0205.0003: input,hidraw1: USB HID v1.11 Mouse [Razer Razer Blade Stealth] on usb-0000:00:14.0-8/input2
[   17.439809] razerkbd 0003:1532:0205.0002: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer Blade Stealth] on usb-0000:00:14.0-8/input1
[  229.926539] razerkbd: Command failed. Start Marker: 03 id: ff Num Params: 02 Reserved: 0e Command: 04 Params: 01ff0000000000000000000000000000 .
```

terrycain, 2017-08-05:
Yeah that looks pretty standard

terrycain, 2017-08-05:
Ok to eliminate the daemon doing anything stupid. Grab [this](https://github.com/terrycain/razer-drivers/blob/master/scripts/driver/keyboard_keys.py)
Edit the first 2 file paths in it to point to the right place (the numbers after razerkbd/ will change)

Then run it as root and hit enter a few times, keys "should" light up.

joonatoona, 2017-08-05:
Nope. That set it to the god awful default profile.

terrycain, 2017-08-05:
LOL

joonatoona, 2017-08-05:
And it's refusing to switch to any other profile... :(

joonatoona, 2017-08-05:
So I should just follow the doc for a new device? Maybe there was a firmware update, because it worked perfectly a few months ago.

terrycain, 2017-08-05:
for the packet captures?

joonatoona, 2017-08-05:
Yeah. I'll just pretend it's a new device, because it looks like the protocol changed somehow.

terrycain, 2017-08-05:
Quite possible, yeah that would be good thanks :smile:

joonatoona, 2017-08-05:
Awesome, I'll do that then open a new issue. Thanks!

closed 2017-08-05 by joonatoona

terrycain, 2017-08-05:
Wicked :)

joonatoona, 2017-08-05:
Is there anything weird you need to do to get synapse running in a VM? It's a garbage piece of software and refuses to open in my VM :(

terrycain, 2017-08-05:
did you pass through usb?

terrycain, 2017-08-05:
as i think it only likes to appear if it sees a razer device

joonatoona, 2017-08-05:
It finally appeared, but it claims there's no device connected. Device manager says there's a "Razer Blade Stealth" connected though :/

terrycain, 2017-08-05:
Very odd, try a reboot?

joonatoona, 2017-08-05:
Rebooted both the host and guest, still doesn't work?

Maybe I should use Win7 and VirtualBox instead of WIn10 and QEMU?

[image]

terrycain, 2017-08-05:
Its worth a try, Ive seen more people have success with virtualbox, also i've heard people having success getting captures with wireshark on normal windows instead of linux but i think thats even more difficult

joonatoona, 2017-08-05:
Yeah, I just prefer QEMU because VirtualBox requires the proprietary addons for USB passthrough. I'll try win7 first, then maybe VirtualBox too.

My disk is encrypted, so installing Windows would mean needing to decrypt the disk, which I don't even think is possible.

terrycain, 2017-08-05:
its dooable i think but tons of hassle ;-) plus who wants windows installed to disk :wink:

terrycain, 2017-08-05:
Also if you have it, we have a telegram chat going (add me `@terrycain`) if you wanna join the group.

terrycain, 2017-08-05:
Also there is a docs branch with some basic docs but its not finished yet

joonatoona, 2017-08-05:
Sure, I don't have telegram now, gimme a minute :D

joonatoona, 2017-08-05:
How do I do telegram? I have no clue how to use this ;-;

terrycain, 2017-08-05:
somehow you can add people as friends,

commit 78e59de0 (2018-09-24): Fix stupid error and add logging so it gets noticed the next time

Fixes #683, fixes #350