openrazer/openrazer #351: Razer Blade Stealth New Firmare Or Something

joonatoona, 2017-08-05:
Capped all the modes (Except the custom, no clue what to do about that) and a screenshot of all options in synapse.

[Blade.zip](https://github.com/terrycain/razer-drivers/files/1202582/Blade.zip)

terrycain, 2017-08-05:
for custom, just turn it on, it'll spam packets, then stop capture,

terrycain, 2017-08-05:
whats ouput of lsusb? just for my records

joonatoona, 2017-08-05:
Cool, I'll start up the VM again.

```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 04f3:223c Elan Microelectronics Corp. 
Bus 001 Device 004: ID 1532:0205 Razer USA, Ltd 
Bus 001 Device 003: ID 0bda:579f Realtek Semiconductor Corp. 
Bus 001 Device 002: ID 0cf3:e300 Atheros Communications, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

joonatoona, 2017-08-05:
[Custom.zip](https://github.com/terrycain/razer-drivers/files/1202590/Custom.zip)

terrycain, 2017-08-05:
try
```
import razer.client 
a = razer.client.DeviceManager() 
b = a.devices[0]          // Assuming it's the only device 
b.name 
b.brightness = 100.0 
b.fx.advanced.matrix[4,5] = (255, 0, 0) 
b.fx.advanced.matrix.row_binary(4)
```
the last line should print out a long string what is it

joonatoona, 2017-08-05:
```
b'\x04\x00\x0e\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
```

terrycain, 2017-08-05:
`ls -lah /sys/bus/hid/drivers/razerkbd/0*/`

joonatoona, 2017-08-05:
https://ptpb.pw/XEZp

terrycain, 2017-08-05:
```
sudo -i
cd /sys/bus/hid/drivers/razerkbd/0003:1532:0205.000B

cat matrix_brightness
echo -n '255' > matrix_brightness
cat matrix_brightness
# By this point brightness should be 255

echo -ne \x04\x00\x0f\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00' > matrix_custom_frame
echo -n '1' > matrix_effect_custom
```
In theory should light up a key

terrycain, 2017-08-05:
updated ^^

joonatoona, 2017-08-05:
Nope, still whatever the effect was before :/

z3ntu, 2017-11-26:
Closing as there wasn't a response in the last month.

closed 2017-11-26 by z3ntu