openrazer/openrazer #352: Macro Keys function not working with BlackWidow 2012 keyboard

r2rX, 2017-08-07:
Hey all,

I'm currently running Ubuntu 17.04 x64 (w/ Kernel v4.12) using a Razer BlackWidow 2012 keyboard (firmware v1.8). Currently running is the latest Razer driver (via ppa:terrz/razerutils) and Polychromatic (via ppa:lah7/polychromatic).

Everything seems to be working fine, except for the Macro Keys function. Pressing Fn + M (Fn + Right-Alt) doesn't seem to do anything.

The following is the output of the key combination via xev:
```
KeyRelease event, serial 37, synthetic NO, window 0x5400001,
    root 0x298, subw 0x0, time 817278, (94,120), root:(1836,168),
    state 0x18, keycode 108 (keysym 0xffea, Alt_R), same_screen YES,
    XLookupString gives 0 bytes: 
    XFilterEvent returns: False
```

Please let me know if there's any additional info required.

terrycain, 2017-08-10:
Yeah currently onthefly macros only work if you have m1->m5 as people kept binding keys over normal keyboard keys. New macro stuff is on the way. Until then it will stay that way

lah7, 2017-08-10:
@r2rX Does the macro key light up at all? (I presumed it doesn't)

But if it's blinking, as @terrycain hints, you can edit (as root) `/usr/lib/python3/dist-packages/razer_daemon/misc/key_event_management.py` to ignore that M1-M5 limitation by commenting out (`#`) the logic and restarting the daemon:

```diff '/tmp/test/key_event_management.py.old' '/tmp/test/key_event_management.py' 
486,490c486,490
<                         if key_name not in ('M1', 'M2', 'M3', 'M4', 'M5'):
<                             self._logger.warning("Macros are only for M1-M5 for now.")
<                             self._recording_macro = False
<                             self._parent.setMacroMode(False)
<                         else:
---
>                         #if key_name not in ('M1', 'M2', 'M3', 'M4', 'M5'):
>                             #self._logger.warning("Macros are only for M1-M5 for now.")
>                             #self._recording_macro = False
>                             #self._parent.setMacroMode(False)
>                         #else:

```

r2rX, 2017-08-11:
@terrycain @lah7 Thanks for the feedback.

The keyboard has the M1-M5 keys. The issue is that entering Macro Mode does not work (thus the Macro key does not light up at all).

ExpLog, 2017-10-25:
I'm using a Blackwidow Chroma V2 and while the macro recording is working fine, entering macro recording mode seems to be a bit fiddly. 

Pressing Fn+F9 (the macro key) can have one of the following results:
1. Nothing happens. No blinking or solid macro light.
2. Macro light starts blinking but pressing M1-5 doesn't start recording. Ne
3. Works as intended.

Otherwise, everything works great. (:

GrantFleming, 2018-02-21:
Yeah I'm using Blackwidow Chroma V2 on ubuntu 17.10 and running in device mode. When I press Fn+F9 nothing happens. The macro led doesn't flash and it doesn't seem to record a macro.

myelsukov, 2018-05-06:
I think this may happen because of the wrong key mapping in the driver: https://github.com/openrazer/openrazer/issues/577

Try Fn-F9 and check if Macro LED starts blinking

closed 2018-10-03 by stale[bot]